espressif/esp-matter #593: ESP-Matter example zigbee_bridge compile error using ESP32C6

mrezzaaa, 2023-08-27:
I want to bridging Zigbee Gateway to Matter end device using esp32c6 devkit C.

Im trying to testing matter examples zigbee_bridge and when compiling got this error message:

```
[build] [1119/1159  95% :: 145.601] Linking C static library esp-idf/chip/libchip.a
[build] [1120/1159  95% :: 145.869] Building CXX object esp-idf/esp_matter/CMakeFiles/__idf_esp_matter.dir/Users/icewalker/esp-idf/esp-matter/connectedhomeip/connectedhomeip/src/app/clusters/wifi-network-diagnostics-server/wifi-network-diagnostics-server.cpp.obj
[build] [1121/1159  95% :: 145.902] Building CXX object esp-idf/esp_matter/CMakeFiles/__idf_esp_matter.dir/zap_common/app/callback-stub.cpp.obj
[build] [1122/1159  96% :: 145.962] Building CXX object esp-idf/esp_matter/CMakeFiles/__idf_esp_matter.dir/Users/icewalker/esp-idf/esp-matter/connectedhomeip/connectedhomeip/src/app/clusters/time-synchronization-server/time-synchronization-server.cpp.obj
[build] [1123/1159  96% :: 145.975] Building CXX object esp-idf/esp_matter_console/CMakeFiles/__idf_esp_matter_console.dir/esp_matter_console_diagnostics.cpp.obj
[build] [1124/1159  96% :: 146.028] Building CXX object esp-idf/esp_matter_console/CMakeFiles/__idf_esp_matter_console.dir/esp_matter_console_wifi.cpp.obj
[build] [1125/1159  96% :: 146.269] Building CXX object esp-idf/main/CMakeFiles/__idf_main.dir/app_zboss.cpp.obj
[build] FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/app_zboss.cpp.obj 
[build] /Users/icewalker/esp-idf/components/esptool_py/esptool/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -DESP_PLATFORM -DIDF_VER=\"v5.1.1-dirty\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DRCP_COMPILE_DEFINE="\"zigbee-e088c3766b-7bbafac-0.3.6-2023-08-27 14:00:28 UTC\"" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DUNITY_INCLUDE_CONFIG_H -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/Users/icewalker/esp-idf/esp-matter/examples/zigbee_bridge/build/config -I/Users/icewalker/esp-idf/esp-matter/examples/zigbee_bridge/main -I/Users/icewalker/esp-idf/esp-idf/components/newlib/platform_include -I/Users/icewalker/esp-idf/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/Users/icewalker/esp-idf/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/Users/icewalker/esp-idf/esp-idf/components/freertos/esp_additions/include/freertos -I/Users/icewalker/esp-idf/esp-idf/components/freertos/esp_additions/include -I/Users/icewalker/esp-idf/esp-idf/components/freertos/esp_additions/arch/xtensa/include -I/Users/icewalker/esp-idf/esp-idf/components/esp_hw_support/include -I/Users/icewalker/esp-idf/esp-idf/components/esp_hw_support/include/soc -I/Users/icewalker/esp-idf/esp-idf/components/esp_hw_support/include/soc/esp32 -I/Users/icewalker/esp-idf/esp-idf/components/esp_hw_support/port/esp32/. -I/Users/icewalker/esp-idf/esp-idf/components/esp_hw_support/port/esp32/private_include -I/Users/icewalker/esp-idf/esp-idf/components/heap/include -I/Users/icewalker/esp-idf/esp-idf/components/log/include -I/Users/icewalker/esp-idf/esp-idf/components/soc/include -I/Users/icewalker/esp-idf/esp-idf/components/soc/esp32 -I/Users/icewalker/esp-idf/esp-idf/components/soc/esp32/include -I/Users/icewalker/esp-idf/esp-idf/components/hal/esp32/include -I/Users/icewalker/esp-idf/esp-idf/components/hal/include -I/Users/icewalker/esp-idf/esp-idf/components/hal/platform_port/include -I/Users/icewalker/esp-idf/esp-idf/components/esp_rom/include -I/Users/icewalker/esp-idf/esp-idf/components/esp_rom/include/esp32 -I/Users/icewalker/esp-idf/esp-idf/components/esp_rom/esp32 -I/Users/icewalker/esp-idf/esp-idf/components/esp_common/include -I/Users/icewalker/esp-idf/esp-idf/components/esp_system/include -I/Users/icewalker/esp-idf/esp-idf/components/esp_system/port/soc -I/Users/icewalker/esp-idf/esp-idf/components/esp_system/port/include/private -I/Users/icewalker/esp-idf/esp-idf/components/xtensa/include -I/Users/icewalker/esp-idf/esp-idf/components/xtensa/esp32/include -I/Users/icewalker/esp-idf/esp-idf/components/lwip/include -I/Users/icewalker/esp-idf/esp-idf/components/lwip/include/apps -I/Users/icewalker/esp-idf/esp-idf/components/lwip/include/apps/sntp -I/Users/icewalker/esp-idf/esp-idf/components/lwip/lwip/src/include -I/Users/icewalker/esp-idf/esp-idf/components/lwip/port/include -I/Users/icewalker/esp-idf/esp-idf/components/lwip/port/freertos/include -I/Users/icewalker/esp-idf/esp-idf/components/lwip/port/esp32xx/include -I/Users/icewalker/esp-idf/esp-idf/components/lwip/port/esp32xx/include/arch -I/Users/icewalker/esp-idf/esp-matter/examples/zigbee_bridge/managed_components/espressif__esp-zboss-lib/include -I/Users/icewalker/esp-idf/esp-matter/examples/zigbee_bridge/managed_components/espressif__esp-zigbee-lib/include -I/Users/icewalker/esp-idf/esp-idf/components/esp_ringbuf/include -I/Users/icewalker/esp-idf/esp-idf/components/efuse/include -I/Users/icewalker/esp-idf/esp-idf/components/efuse/esp32/include -I/Users/icewalker/esp-idf/esp-idf/components/driver/include -I/Users/icewalker/esp-idf/esp-idf/components/driver/deprecated -I/Users/icewalker/esp-idf/esp-idf/components/driver/analog_comparator/include -I/Users/icewalker/esp-idf/esp-idf/components/driver/dac/include -I/Users/icewalker/esp-idf/esp-idf/components/driver/gpio/include -I/Users/icewalker/esp-idf/esp-idf/components/driver/gptimer/include -I/Users/icewalker/esp-idf/esp-idf/components/driver/i2c/include -I/Users/icewalker/esp-idf/esp-idf/components/driver/i2s/include -I/Users/icewalker/esp-idf/esp-idf/components/driver/ledc/include -I/Users/icewalker/esp-idf/esp-idf/components/driver/mcpwm/include -I/Users/icewalker/esp-idf/esp-idf/components/driver/parlio/include -I/Users/icewalker/esp-idf/esp-idf/components/driver/pcnt/include -I/Users/icewalker/esp-idf/esp-idf/components/driver/rmt/include -I/Users/icewalker/esp-idf/esp-idf/components/driver/sdio_slave/include -I/Users/icewalker/esp-idf/esp-idf/components/driver/sdmmc/include -I/Users/icewalker/esp-idf/esp-idf/components/driver/sigma_delta/include -I/Users/icewalker/esp-idf/esp-idf/components/driver/spi/include -I/Users/icewalker/esp-idf/esp-idf/components/driver/temperature_sensor/include -I/Users/icewalker/esp-idf/esp-idf/components/driver/touch_sensor/include -I/Users/icewalker/esp-idf/esp-idf/components/driver/twai/include -I/Users/icewalker/esp-idf/esp-idf/components/driver/uart/include -I/Users/icewalker/esp-idf/esp-idf/components/driver/usb_serial_jtag/include -I/Users/icewalker/esp-idf/esp-idf/components/driver/touch_sensor/esp32/include -I/Users/icewalker/esp-idf/esp-idf/components/esp_pm/include -I/Users/icewalker/esp-idf/esp-idf/components/mbedtls/port/include -I/Users/icewalker/esp-idf/esp-idf/components/mbedtls/mbedtls/include -I/Users/icewalker/esp-idf/esp-idf/components/mbedtls/mbedtls/library -I/Users/icewalker/esp-idf/esp-idf/components/mbedtls/esp_crt_bundle/include -I/Users/icewalker/esp-idf/esp-idf/components/esp_app_format/include -I/Users/icewalker/esp-idf/esp-idf/components/bootloader_support/include -I/Users/icewalker/esp-idf/esp-idf/components/bootloader_support/bootloader_flash/include -I/Users/icewalker/esp-idf/esp-idf/components/esp_partition/include -I/Users/icewalker/esp-idf/esp-idf/components/app_update/include -I/Users/icewalker/esp-idf/esp-idf/components/esp_mm/include -I/Users/icewalker/esp-idf/esp-idf/components/spi_flash/include -I/Users/icewalker/esp-idf/esp-idf/components/pthread/include -I/Users/icewalker/esp-idf/esp-idf/components/esp_timer/include -I/Users/icewalker/esp-idf/esp-idf/components/app_trace/include -I/Users/icewalker/esp-idf/esp-idf/components/esp_event/include -I/Users/icewalker/esp-idf/esp-idf/components/nvs_flash/include -I/Users/icewalker/esp-idf/esp-idf/components/esp_phy/include -I/Users/icewalker/esp-idf/esp-idf/components/esp_phy/esp32/include -I/Users/icewalker/esp-idf/esp-idf/components/vfs/include -I/Users/icewalker/esp-idf/esp-idf/components/esp_netif/include -I/Users/icewalker/esp-idf/esp-idf/components/wpa_supplicant/include -I/Users/icewalker/esp-idf/esp-idf/components/wpa_supplicant/port/include -I/Users/icewalker/esp-idf/esp-idf/components/wpa_supplicant/esp_supplicant/include -I/Users/icewalker/esp-idf/esp-idf/components/esp_coex/include -I/Users/icewalker/esp-idf/esp-idf/components/esp_wifi/include -I/Users/icewalker/esp-idf/esp-idf/components/esp_wifi/wifi_apps/include -I/Users/icewalker/esp-idf/esp-idf/components/bt/include/esp32/include -I/Users/icewalker/esp-idf/esp-idf/components/bt/common/osi/include -I/Users/icewalker/es [... cut ...]
[build] [1125/1159  97% :: 149.362] Building CXX object esp-idf/esp_matter/CMakeFiles/__idf_esp_matter.dir/esp_matter_client.cpp.obj
[build] ninja: build stopped: subcommand failed.
[proc] The command: /Users/icewalker/.espressif/tools/cmake/3.24.0/CMake.app/Contents/bin/cmake --build /Users/icewalker/esp-idf/esp-matter/examples/zigbee_bridge/build --config Debug --target all -- exited with code: 1
```

Already set ``idf.py --preview set-target esp32c6`` .
SDK config ``Zigbee > Zigbee Coordinator or Router device checked`` 
and got this error.

shubhamdp, 2023-08-28:
Could be the problem with esp-idf commit. [Documentation](https://docs.espressif.com/projects/esp-matter/en/latest/esp32c6/developing.html#getting-the-repositories) suggest that you use `ea5e0ff298e6257b31d8e0c81435e6d3937f04c7`. Please try with that rather than using latest.

mrezzaaa, 2023-08-28:
Ok i'll try with commit hash that you give to me. I'll update later if it works or not

mrezzaaa, 2023-08-29:
@shubhamdp trying your suggest and got this error message when compiling 
```
/esp-idf/esp-idf/components/lwip/lwip/src/core/ipv4/dhcp.c:1144:11: error: 'DHCP_REQUEST_TIMEOUT_SEQUENCE' undeclared (first use in this function)
 1144 |   msecs = DHCP_REQUEST_TIMEOUT_SEQUENCE(dhcp->tries);
      |           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~

```

mrezzaaa, 2023-08-29:
When i moving to branch `v5.1.1` of esp-idf  and commit ``68e99de70266674ce172459136960bbb5c58dc0e`` of esp-matter and compiling it again for ``esp32c6`` target is working, but now i got this error. 

```
I (33855) chip[DIS]: mDNS service published: _matter._tcp
I (33855) chip[IM]: No subscriptions to resume
I (34305) chip[EM]: >>> [E:8594r S:0 M:185737137] (U) Msg RX from 0:01EB240DFC8722C7 [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
I (34315) chip[IN]: CASE Server received Sigma1 message . Starting handshake. EC 0x40821250
I (34325) chip[EM]: <<< [E:8594r S:0 M:107294229 (Ack:185737137)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (34335) chip[IN]: (U) Sending msg 107294229 to IP address 'UDP:[FE80::F4:8D89:1EC4:5901%st1]:56400'
I (34345) chip[SC]: Received Sigma1 msg
I (34355) chip[SC]: CASE matched destination ID: fabricIndex 1, NodeID 0x000000000000006E
I (34395) chip[EM]: <<< [E:8594r S:0 M:107294230 (Ack:185737137)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:31 (SecureChannel:CASE_Sigma2)
I (34415) chip[IN]: (U) Sending msg 107294230 to IP address 'UDP:[FE80::F4:8D89:1EC4:5901%st1]:56400'
I (34415) chip[SC]: Sent Sigma2 msg
I (34795) chip[EM]: >>> [E:8594r S:0 M:185737138 (Ack:107294230)] (U) Msg RX from 0:01EB240DFC8722C7 [0000] --- Type 0000:32 (SecureChannel:CASE_Sigma3)
I (34815) chip[EM]: <<< [E:8594r S:0 M:107294231 (Ack:185737138)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (34825) chip[IN]: (U) Sending msg 107294231 to IP address 'UDP:[FE80::F4:8D89:1EC4:5901%st1]:56400'
I (34835) chip[SC]: Received Sigma3 msg
E (34845) chip[SC]: The device does not support GetClock_RealTimeMS() API: 6c.  Falling back to Last Known Good UTC Time
I (34915) chip[EM]: <<< [E:8594r S:0 M:107294232 (Ack:185737138)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:40 (SecureChannel:StatusReport)
I (34925) chip[IN]: (U) Sending msg 107294232 to IP address 'UDP:[FE80::F4:8D89:1EC4:5901%st1]:56400'
I (34935) chip[SC]: SecureSession[0x4081f998, LSID:38013]: State change 'kEstablishing' --> 'kActive'
I (34945) chip[IN]: CASE Session established to peer: <2ED513DC3E2CC2BE, 1>
I (34915) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (35115) esp_netif_handlers: sta ip: 192.168.100.13, mask: 255.255.255.0, gw: 192.168.100.1

***ERROR*** A stack overflow in task sys_evt has been detected.
Core  0 register dump:
Stack dump detected
MEPC    : 0x40800606  RA      : 0x4080a8e8  SP      : 0x4081a7d0  GP      : 0x40816eb0  
0x40800606: panic_abort at /Users/icewalker/esp-idf/esp-idf/components/esp_system/panic.c:452

0x4080a8e8: __ubsan_include at /Users/icewalker/esp-idf/esp-idf/components/esp_system/ubsan.c:313

TP      : 0x407fa808  T0      : 0x40030dca  T1      : 0x0000000f  T2      : 0x00000001  
0x40030dca: memset in ROM

S0/FP   : 0x4081a7f4  S1      : 0x4081a7f4  A0      : 0x4081a7f4  A1      : 0x42173268  
A2      : 0x0000000f  A3      : 0x4081a821  A4      : 0x00000001  A5      : 0x4082f000  
A6      : 0x0000001c  A7      : 0x0164a8c0  S2      : 0x00001881  S3      : 0x00000001  
S4      : 0xffffffff  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x4083f590  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x00000000  T4      : 0x00000000  T5      : 0x00000001  T6      : 0x421a2a64  
MSTATUS : 0x00001801  MTVEC   : 0x40800001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
0x40800001: _vector_table at ??:?

MHARTID : 0x00000000  

Backtrace:

panic_abort (details=details@entry=0x4081a7f4 <xIsrStack+1412> "***ERROR*** A stack overflow in task sys_evt has been detected.") at /Users/icewalker/esp-idf/esp-idf/components/esp_system/panic.c:452
452	    *((volatile int *) 0) = 0; // NOLINT(clang-analyzer-core.NullDereference) should be an invalid operation on targets
#0  panic_abort (details=details@entry=0x4081a7f4 <xIsrStack+1412> "***ERROR*** A stack overflow in task sys_evt has been detected.") at /Users/icewalker/esp-idf/esp-idf/components/esp_system/panic.c:452
#1  0x4080a8e8 in esp_system_abort (details=details@entry=0x4081a7f4 <xIsrStack+1412> "***ERROR*** A stack overflow in task sys_evt has been detected.") at /Users/icewalker/esp-idf/esp-idf/components/esp_system/port/esp_system_chip.c:84
#2  0x4080d168 in vApplicationStackOverflowHook (xTask=<optimized out>, pcTaskName=<optimized out>) at /Users/icewalker/esp-idf/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:434
#3  0x4080c20e in vTaskSwitchContext () at /Users/icewalker/esp-idf/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:3729
#4  vTaskSwitchContext () at /Users/icewalker/esp-idf/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:3673
#5  0x4080d46c in isr_skip_decrement () at /Users/icewalker/esp-idf/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/portasm.S:87
Backtrace stopped: frame did not save the PC
ELF file SHA256: 0f65d4fbf96a72ac

Rebooting...
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0xc (SW_CPU),boot:0x1e (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001975a
0x4001975a: software_reset_cpu in ROM

```

shubhamdp, 2023-08-30:
> ***ERROR*** A stack overflow in task sys_evt has been detected.

This suggests that someone is using a lot more stack on default event loop. Are you using the default example without any change?

mrezzaaa, 2023-08-30:
I tried the default one, since error after connected to wifi and comissioning process. I tried to increasing stack, same error still same.

mrezzaaa, 2023-09-05:
yes i'd testing both bridge and rcp. 
I think C6 is built in for 802.15 hardware, so doesn't need any external like H2 hardware (just set to Native Zigbee on SDK config).
Based on log (not on log above), 802.15 network joined to Zigbee network successfully, but crashing after joining WiFi network. Some crashed before joined Zigbee network, and some crash after joined Zigbee network. Crashed happened after WiFi joining successfully.

mrezzaaa, 2023-09-22:
I am using IDF 5.1.1

baylf2000, 2023-10-02:
@mrezzaaa 
> I am using IDF 5.1.1

As I understand it, you cannot use both 802.15 and Wifi at the same time on H2 or C6 devices. You will need to use two devices together in that case.

mrezzaaa, 2023-10-02:
Any detail explanation why this couldn't work at same time?

baylf2000, 2023-10-03:
> Any detail explanation why this couldn't work at same time?

You'd have to ask Espressif for a definitive answer, however I believe it's because there is only a single radio in the device, which makes it more difficult to use both protocols at the same time.

mrezzaaa, 2023-10-03:
I think (correct me if i'm wrong) the release of C6 for increasing efficiency of it's predecessor ( H2 + Esp32 ) for Matter WiFi and Radio.🤔  [image]

baylf2000, 2023-10-03:
> I think (correct me if i'm wrong) the release of C6 for increasing efficiency of it's predecessor ( H2 + Esp32 ) for Matter WiFi and Radio.🤔 [image]

Unfortunately you are wrong. As it clearly states there... "ESP32-C6 *combined with* another Espressif SoC". At this stage I have not seen any examples of using WiFi and 802.15 together at the same time in a single device. You can use WiFi and Bluetooth together, and Bluetooth and 802.15 together, so I don't understand why it's different with WiFi and 802.15, but that's just the way it is right now.

mrezzaaa, 2023-10-03:
Ah i see. Thanks for the information. Closing this issue now

closed 2023-10-03 by mrezzaaa