espressif/esp-matter #596: Example controller: Fail to commission device

nguyenhuy5997, 2023-08-28:
**Describe the bug**
I have flashed example controller into ESP32S3. After that, i use command paring ble-wifi for a wifi socket, the log show i fail Attestation Information. Then i try paring onnetwork for Aquara Hub M2, the log show me the same result

**Environment**
- ESP-IDF Commit Id: v5.0.1
- SoC (eg: ESP32 or ESP32-C3): ESP32S3

**Here is the log pairing a wifi socket**
> matter esp controller pairing ble-wifi 0x000A ggwp 12345678 17670007 2570

I (2788516) chip[CTL]: Setting wifi credentials from parameters
I (2788516) chip[CTL]: Setting attestation nonce to random value
I (2788516) chip[CTL]: Setting CSR nonce to random value
I (2788526) chip[CTL]: Commission called for node ID 0x000000000000000A
Done
I (2788536) NimBLE: GAP procedure initiated: discovery;
I (2788546) NimBLE: own_addr_type=1 filter_policy=0 passive=1 limited=0 filter_duplicates=1
I (2788556) NimBLE: duration=60000ms
I (2788556) NimBLE:

                                                                                                                                                                                                                                            I (2788576) chip[BLE]: Device Discriminator match. Attempting to connect
I (2788586) chip[BLE]: Scan complete notification without an active scan
I (2788586) NimBLE: GAP procedure initiated: connect;
I (2788596) NimBLE: peer_addr_type=0 peer_addr=
I (2788606) NimBLE: 1c:90:ff:b8:69:b0
I (2788606) NimBLE:  scan_itvl=16 scan_window=16 itvl_min=24 itvl_max=40 latency=0 supervision_timeout=256 min_ce_len=0 m> ax_ce_len=0 own_addr_type=1
I (2788616) NimBLE:

I (2788686) NimBLE: GATT procedure initiated: exchange mtu

I (2788686) chip[DL]: BLE GAP connection established (con 1)
I (2788686) NimBLE: GATT procedure initiated: discover all services

I (2788786) NimBLE: GAP procedure initiated:
I (2788796) NimBLE: connection parameter update; conn_handle=1 itvl_min=50 itvl_max=80 latency=0 supervision_timeout=500 min_ce_len=0 max_ce_len=0
I (2788806) NimBLE:

I (2788836) NimBLE: GATT procedure initiated: discover all characteristics;
I (2788836) NimBLE: start_handle=1 end_handle=65535

I (2789036) NimBLE: GATT procedure initiated: discover all descriptors;
I (2789036) NimBLE: chr_val_handle=5 end_handle=6

I (2789136) NimBLE: GATT procedure initiated: discover all descriptors;
I (2789136) NimBLE: chr_val_handle=8 end_handle=65535

I (2789286) chip[BLE]: GATT discovery complete status:0 conn_handle:1
I (2789286) chip[DL]: HandlePlatformSpecificBLEEvent 16385
I (2789296) chip[DIS]: Closing all BLE connections
I (2789306) NimBLE: GATT procedure initiated: write;
I (2789306) NimBLE: att_handle=3 len=9

I (2789316) chip[EM]: <<< [E:6053i S:0 M:22373966] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:20 (SecureChannel:PBKDFParamRequest)
I (2789326) chip[IN]: (U) Sending msg 22373966 to IP address 'BLE'
I (2789386) chip[DL]: HandlePlatformSpecificBLEEvent 16387
I (2789386) NimBLE: GATT procedure initiated: write;
I (2789396) NimBLE: att_handle=6 len=2

I (2789486) chip[DL]: Subscribe complete: conn_handle=1, error=0, attr_handle=6
I (2789496) chip[DL]: HandlePlatformSpecificBLEEvent 16388
I (2789506) chip[BLE]: subscribe complete, ep = 0x3fcae4b0
I (2789506) chip[DL]: HandlePlatformSpecificBLEEvent 16389
I (2789506) chip[BLE]: peripheral chose BTP version 4; central expected between 4 and 4
I (2789516) chip[BLE]: using BTP fragment sizes rx 244 / tx 244.
I (2789526) chip[BLE]: local and remote recv window size = 5
I (2789526) NimBLE: GATT procedure initiated: write;
I (2789546) NimBLE: att_handle=3 len=73

I (2789686) chip[DL]: HandlePlatformSpecificBLEEvent 16387
I (2789696) chip[DL]: HandlePlatformSpecificBLEEvent 16389
[... 244 lines cut ...]
I (3123906) chip[CTL]: Received Attestation Information from the device
I (3123926) chip[CTL]: Successfully finished commissioning step 'SendAttestationRequest'
I (3123936) chip[CTL]: Commissioning stage next step: 'SendAttestationRequest' -> 'AttestationVerification'
I (3123936) chip[CTL]: Performing next commissioning step 'AttestationVerification'
I (3123946) chip[CTL]: Verifying attestation
E (3124196) chip[CTL]: Failed in verifying 'Attestation Information' command received from the device: err 101. Look at AttestationVerificationResult enum to understand the errors
I (3124206) chip[CTL]: Error on commissioning step 'AttestationVerification': 'Error CHIP:0x000000AC'
E (3124206) chip[CTL]: Failed to perform commissioning step 13
I (3124226) chip[CTL]: Going from commissioning step 'AttestationVerification' with lastErr = 'Error CHIP:0x000000AC' -> 'Cleanup'
I (3124236) chip[CTL]: Performing next commissioning step 'Cleanup' with completion status = 'Error CHIP:0x000000AC'
I (3124246) chip[CTL]: Expiring failsafe on proxy 0x3fca7930
I (3124246) chip[EM]: <<< [E:6068i S:3609 M:66046838] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0001:08 (IM:InvokeCommandRequest)
I (3124266) chip[IN]: (S) Sending msg 66046838 on secure session with LSID: 3609
I (3124276) chip[EM]: <<< [E:6067i S:3609 M:66046839 (Ack:230882929)] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (3124286) chip[IN]: (S) Sending msg 66046839 on secure session with LSID: 3609
I (3124326) chip[EM]: >>> [E:6067i S:3609 M:230882929 (Ack:66046836)] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:09 (IM:InvokeCommandResponse)
I (3124336) chip[EM]: <<< [E:6067i S:3609 M:66046840 (Ack:230882929)] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (3124356) chip[IN]: (S) Sending msg 66046840 on secure session with LSID: 3609
I (3124356) chip[EM]: >>> [E:6068i S:3609 M:230882930 (Ack:66046838)] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:09 (IM:InvokeCommandResponse)
I (3124366) chip[DMG]: Received Command Response Data, Endpoint=0 Cluster=0x0000_0030 Command=0x0000_0001
I (3124386) chip[CTL]: Failsafe disarmed
I (3124386) chip[CTL]: Successfully finished commissioning step 'Cleanup'
I (3124396) pairing_command: Device commissioning Failure: MatterError CHIP:0x000000AC
I (3124396) chip[SC]: SecureSession[0x3fcd1c4c, LSID:3609]: State change 'kActive' --> 'kPendingEviction'
E (3124406) chip[EM]: Failed to send Solitary ack for MessageCounter:230882930 on exchange 6068i:2

PSONALl, 2023-08-28:
@nguyenhuy5997 are you using test certificates in your application?
Can you check if the commissioning works with chip-tool?

nguyenhuy5997, 2023-08-28:
@PSONALl  no, my wifi socket and Aquara Hub M2 is verified certification, not test. 
But i also tried to test certificate with light examples and i got  the same result.

wqx6, 2023-08-29:
The `err 101` is `kPaaNotFound`, so the root certificate of your device's DAC is not the test certificate in the controller. The controller is using the hard-code [test PAA certificate](https://github.com/espressif/connectedhomeip/blob/5b4f8004662d00bdb111367fec7d3ea978c23372/credentials/test/attestation/Chip-Test-PAA-NoVID-Cert.pem) to do the device attestation verification. So please ensure that the DAC and PAI on the end-device is generated by that test PAA certificate and key.

dhrishi, 2023-09-04:
@nguyenhuy5997 Please share an update or close the issue if resolved

dhrishi, 2023-09-11:
Closing this now

closed 2023-09-11 by dhrishi