espressif/esp-matter #599: Question: zigbee-bridge on Zigbee Gateway v1.2

nevi-me, 2023-08-28:
**Describe the question/query that you have**

I have a esp-thread-border/zigbee-gateway board, and I have flashed the zigbee-gateway example (and the RCP on the h2).
I was able to commission it with Google Home instead of the chip-tool, however now I don't know how to actually discover and add Zigbee devices.

I have 2 questions:

1. The documentation is unclear (the README only mentions how to flash the RCP). If using this board, is there any additional configuration required? For example, on the thread-border-router example, I can run `idf.py menuconfig` and alter some settings beforehand.
2. How do I discover and add Zigbee devices to the bridge if I am not using the `chip-tool`?

**Additional context**

I used `esp-matter` at commit https://github.com/espressif/esp-matter/commit/39914437dfbe6f16c77f19df580b28ef90750c5c, `esp-idf` at the recommended `ea5e0ff298`.

dhrishi, 2023-08-28:
@wqx6 Please check

@nevi-me Meanwhile, can you confirm if you have flashed another ESP32-H2 with the Zigbee light_bulb example as mentioned [here](https://github.com/espressif/esp-matter/tree/main/examples/zigbee_bridge#22-setup-zigbee-bulb-on-esp32-h2) ? Once done, it should be added as a Matter accessory through the bridge.
The chip-tool commands provided just validate that the bridge is added. They shouldn't be required if you are using any ecosystem app.

nevi-me, 2023-08-28:
> can you confirm if you have flashed another ESP32-H2 with the Zigbee light_bulb example as mentioned [here](https://github.com/espressif/esp-matter/tree/main/examples/zigbee_bridge#22-setup-zigbee-bulb-on-esp32-h2) ?

I've just flashed the example (though it's called `HA_on_off_light`), and I'm getting the below error after the light bulb starts up:

```
I (3061) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: -1)
I (6731) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: -1)
I (10401) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: -1)
I (14071) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: -1)
```

wqx6, 2023-08-29:
> The documentation is unclear (the README only mentions how to flash the RCP). If using this board, is there any additional configuration required? For example, on the thread-border-router example, I can run idf.py menuconfig and alter some settings beforehand.

You don't need to do any additional configuration in menuconfig. If the RCP is connected well to the zigbee bridge, the bridge will form a ZigBee network and print log like this:
```
I (1280) esp_zboss: Start network formation
I (3060) esp_zboss: Formed network successfully (ieee extended address: f9:54:2d:01:a0:03:f7:84, PAN ID: 0x8651)
```
And if the HA_on_off_light is in the same channel of the bridge, it will join the network after it starts. Then the dynamic endpoint will be added to the bridge.

> How do I discover and add Zigbee devices to the bridge if I am not using the `chip-tool`?

Apple Home will detect the adding of dynamic endpoint and shows a new bridged device in their UI. But I'm uncertain about whether Google Home follows the same approach.

> and I'm getting the below error after the light bulb starts up

Could you please check whether the bridge has formed a Zigbee network? Look like the `HA_on_off_light` doesn't join the Zigbee network.

nevi-me, 2023-08-29:
> You don't need to do any additional configuration in menuconfig. If the RCP is connected well to the zigbee bridge, the bridge will form a ZigBee network and print log like this:

I've cleared the flash, rebuilt the application and recommissioned. Interestingly, there are no logs at all from `esp_zboss`, `zboss` or anything to indicate that zigbee is running or if the entrypoint `launch_app_zboss()` had an issue.

I'm not yet good at reading schematic diagrams, I'm trying to figure out if the below default UART settings map to what's expected on the border router.

```cpp
#define ESP_ZB_DEFAULT_RADIO_CONFIG()                      \
    {                                                      \
        .radio_mode = RADIO_MODE_UART_RCP,                 \
        .radio_uart_config = {                             \
            .port = 1,                                     \
            .uart_config =                                 \
                {                                          \
                    .baud_rate = 115200,                   \
                    .data_bits = UART_DATA_8_BITS,         \
                    .parity = UART_PARITY_DISABLE,         \
                    .stop_bits = UART_STOP_BITS_1,         \
                    .flow_ctrl = UART_HW_FLOWCTRL_DISABLE, \
                    .rx_flow_ctrl_thresh = 0,              \
                    .source_clk = UART_SCLK_APB,           \
                },                                         \
            .rx_pin = 4,                                   \
            .tx_pin = 5,                                   \
        },                                                 \
    }
```

What I think the ESP32-S3 is mapped with

```markdown
ESP32 pin           | ESP32-H2 pin
--------------------|-------------
  GND               |     G
  GPIO17 (UART RX)  |     TX
  GPIO18 (UART TX)  |     RX
  GPIO7             |     RST
  GPIO8  (SPI INTR) |     GPIO9 (BOOT)
  GPIO10 (SPI CS)   |     GPIO2
  GPIO11 (SPI MOSI) |     GPIO3
  GPIO12 (SPI CLK)  |     GPIO0
  GPIO13 (SPI MISO) |     GPIO1
```

wqx6, 2023-08-30:
The Zigbee bridge example is designed for two devkits connected to each other(not a single board), with UART pins in the radio configuration being 4 and 5. However, for the Zigbee Gateway v1.2, the configuration differs. In this case, the `rx_pin` in the radio configuration should be set to 17, while the `tx_pin` should be set to 18.

Since the USB/JTAG instead of the UART is used as the default console for Zigbee Gateway v1.2 board, I would suggest that you change the default console channel in menuconfig.

dhrishi, 2023-09-04:
@nevi-me Please close the issue if this helps

nevi-me, 2023-09-04:
Sorry, I forgot to get back here. I changed the pins to 17 and 18, and the logs indicated that the RCP connection was working, however the Zigbee network was still not being formed. I didn't investigate further after that, so I can close this issue.

Thanks for all your assistance :)

closed 2023-09-04 by nevi-me