espressif/esp-matter #600: Show OTA available in Apple Home?

law-ko, 2023-08-29:
Hello,

If we are using HTTPs OTA instead of Matter Requestor method, is there any method to display there's OTA available and trigger OTA updates directly in Apple Home app?

shubhamdp, 2023-08-30:
Communication between two Matter devices are strictly through clusters. OTA requestor cluster has the attributes which has has mechanism to notify the home app (Matter client). I don't think there is any provision right now. You can provision such things using manufacturer specific apps.

dhrishi, 2023-09-04:
@law-ko Please close your issue if the question is answered

law-ko, 2023-09-05:
@dhrishi can we use the requestor cluster to notify the Apple Home app and then trigger OTA update?

shubhamdp, 2023-09-05:
@law-ko I'd suggest you to please go through section `11.19. Over-the-Air (OTA) Software Update` of Matter Specification for better understanding of how Matter OTA works.

jonsmirl, 2023-09-18:
https://developers.home.google.com/matter/ota/test/vendors/espressif

dhrishi, 2023-09-21:
If you are using a different OTA mechanism than the Matter OTA then, the Apple Home cannot be used for performing the upgrade.

closed 2023-09-21 by dhrishi