espressif/esp-matter #601: add_definitions with string value casuses build error

damian-kurek-wizzdev, 2023-08-29:
When I add a definition with a string value in the project's main cmake of the project I get a build error. I tested it with multiple esp-idf versions and the esp-matter versions could reproduce the issue every time I tried.

I added this before including cmake for esp-idf and matter in light example and could see the issue as well

`add_definitions(-DTEST="TEST")
`
I can provide a full cmake file if required
**Environment**
- ESP-Matter Commit Id: v5.0 v5.1 
- ESP-IDF Commit Id: v5.1 v5.0 v.4.3.3 
- SoC (eg: ESP32 or ESP32-C3): ESP32
- Device Logs (Please attach the log file):
-- Build files have been written to: /home/damian/Projects/mill/build_mill_esp
[1/1] cd /home/damian/Projects/mill/build_mill_esp/bootlo...an/Projects/mill/build_mill_esp/bootloader/bootloader.binBootloader binary size 0x6800 bytes. 0x800 bytes (7%) free.
[6/428] Performing build step for 'chip_gn'FAILED: esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-build esp-idf/chip/lib/libCHIP.a 
cd /home/damian/Projects/mill/build_mill_esp/esp-idf/chip && ninja esp32
[0/1] Regenerating ninja filesERROR at build arg file (use "gn args <out_dir>" to edit):18:7728: Expected comma between items.
target_cflags_c = ["-DESP_PLATFORM", "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"", "-DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE", "-D_GNU_SOURCE", "-D_POSIX_READER_WRITER_LOCKS", "-isystem/home/damian/Projects/mill/esp-idf/components/freertos/include/freertos", "-isystem/home/damian/Projects/mill/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos", "-isystem/home/damian/Projects/mill/esp-idf/components/newlib/platform_include", "-isystem/home/damian/Projects/mill/esp-idf/components/freertos/FreeRTOS-Kernel/include", "-isystem/home/damian/Projects/mill/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include", "-isystem/home/damian/Projects/mill/esp-idf/components/freertos/esp_additions/include/freertos", "-isystem/home/damian/Projects/mill/esp-idf/components/freertos/esp_additions/include", "-isystem/home/damian/Projects/mill/esp-idf/components/freertos/esp_additions/arch/xtensa/include", "-isystem/home/damian/Projects/mill/esp-idf/components/esp_hw_support/include", "-isystem/home/damian/Projects/mill/esp-idf/components/esp_hw_support/include/soc", "-isystem/home/damian/Projects/mill/esp-idf/components/esp_hw_support/include/soc/esp32", "-isystem/home/damian/Projects/mill/esp-idf/components/esp_hw_support/port/esp32/.", "-isystem/home/damian/Projects/mill/esp-idf/components/esp_hw_support/port/esp32/private_include", "-isystem/home/damian/Projects/mill/esp-idf/components/heap/include", "-isystem/home/damian/Projects/mill/esp-idf/components/log/include", "-isystem/home/damian/Projects/mill/esp-idf/components/soc/include", "-isystem/home/damian/Projects/mill/esp-idf/components/soc/esp32", "-isystem/home/damian/Projects/mill/esp-idf/components/soc/esp32/include", "-isystem/home/damian/Projects/mill/esp-idf/components/hal/esp32/include", "-isystem/home/damian/Projects/mill/esp-idf/components/hal/include", "-isystem/home/damian/Projects/mill/esp-idf/components/hal/platform_port/include", "-isystem/home/damian/Projects/mill/esp-idf/components/esp_rom/include", "-isystem/home/damian/Projects/mill/esp-idf/components/esp_rom/include/esp32", "-isystem/home/damian/Projects/mill/esp-idf/components/esp_rom/esp32", "-isystem/home/damian/Projects/mill/esp-idf/components/esp_common/include", "-isystem/home/damian/Projects/mill/esp-idf/components/esp_system/include", "-isystem/home/damian/Projects/mill/esp-idf/components/esp_system/port/soc", "-isystem/home/damian/Projects/mill/esp-idf/components/esp_system/port/include/private", "-isystem/home/damian/Projects/mill/esp-idf/components/xtensa/include", "-isystem/home/damian/Projects/mill/esp-idf/components/xtensa/esp32/include", "-isystem/home/damian/Projects/mill/esp-idf/components/lwip/include", "-isystem/home/damian/Projects/mill/esp-idf/components/lwip/include/apps", "-isystem/home/damian/Projects/mill/esp-idf/components/lwip/include/apps/sntp", "-isystem/home/damian/Projects/mill/esp-idf/components/lwip/lwip/src/include", "-isystem/home/damian/Projects/mill/esp-idf/components/lwip/port/include", "-isystem/home/damian/Projects/mill/esp-idf/components/lwip/port/freertos/include", "-isystem/home/damian/Projects/mill/esp-idf/components/lwip/port/esp32xx/include", "-isystem/home/damian/Projects/mill/esp-idf/components/lwip/port/esp32xx/include/arch", "-isystem/home/damian/Projects/mill/esp-idf/components/esp_eth/include", "-isystem/home/damian/Projects/mill/esp-idf/components/esp_event/include", "-isystem/home/damian/Projects/mill/esp-idf/components/driver/include", "-isystem/home/damian/Projects/mill/esp-idf/components/driver/deprecated", "-isystem/home/damian/Projects/mill/esp-idf/components/driver/analog_comparator/include", "-isystem/home/damian/Projects/mill/esp-idf/components/driver/dac/include", "-isystem/home/damian/Projects/mill/esp-idf/components/driver/gpio/include", "-isystem/home/damian/Projects/mill/esp-idf/components/driver/gptimer/include", "-isystem/home/damian/Projects/mill/esp-idf/components/driver/i2c/include", "-isystem/home/damian/Projects/mill/esp-idf/components/driver/i2s/include", "-isystem/home/damian/Projects/mill/esp-idf/components/driver/ledc/include", "-isystem/home/damian/Projects/mill/esp-idf/components/driver/mcpwm/include", "-isystem/home/damian/Projects/mill/esp-idf/components/driver/parlio/include", "-isystem/home/damian/Projects/mill/esp-idf/components/driver/pcnt/include", "-isystem/home/damian/Projects/mill/esp-idf/components/driver/rmt/include", "-isystem/home/damian/Projects/mill/esp-idf/components/driver/sdio_slave/include", "-isystem/home/damian/Projects/mill/esp-idf/components/driver/sdmmc/include", "-isystem/home/damian/Projects/mill/esp-idf/components/driver/sigma_delta/include", "-isystem/home/damian/Projects/mill/esp-idf/components/driver/spi/include", "-isystem/home/damian/Projects/mill/esp-idf/components/driver/temperature_sensor/include", "-isystem/home/damian/Projects/mill/esp-idf/components/driver/touch_sensor/include", "-isystem/home/damian/Projects/mill/esp-idf/components/driver/twai/include", "-isystem/home/damian/Projects/mill/esp-idf/components/driver/uart/include", "-isystem/home/damian/Projects/mill/esp-idf/components/driver/usb_serial_jtag/include", "-isystem/home/damian/Projects/mill/esp-idf/components/driver/touch_sensor/esp32/include", "-isystem/home/damian/Projects/mill/esp-idf/components/esp_pm/include", "-isystem/home/damian/Projects/mill/esp-idf/components/esp_ringbuf/include", "-isystem/home/damian/Projects/mill/esp-idf/components/esp_timer/include", "-isystem/home/damian/Projects/mill/esp-idf/components/esp_wifi/include", "-isystem/home/damian/Projects/mill/esp-idf/components/esp_wifi/wifi_apps/include", "-isystem/home/damian/Projects/mill/esp-idf/components/esp_phy/include", "-isystem/home/damian/Projects/mill/esp-idf/components/esp_phy/esp32/include", "-isystem/home/damian/Projects/mill/esp-idf/components/esp_netif/include", "-isystem/home/damian/Projects/mill/esp-idf/components/mbedtls/port/include", "-isystem/home/damian/Projects/mill/esp-idf/components/mbedtls/mbedtls/include", "-isystem/home/damian/Projects/mill/esp-idf/components/mbedtls/mbedtls/library", "-isystem/home/damian/Projects/mill/esp-idf/components/mbedtls/esp_crt_bundle/include", "-isystem/home/damian/Projects/mill/esp-idf/components/fatfs/diskio", "-isystem/home/damian/Projects/mill/esp-idf/components/fatfs/vfs", "-isystem/home/damian/Projects/mill/esp-idf/components/fatfs/src", "-isystem/home/damian/Projects/mill/esp-idf/components/wear_levelling/include", "-isystem/home/damian/Projects/mill/esp-idf/components/esp_partition/include", "-isystem/home/damian/Projects/mill/esp-idf/components/sdmmc/include", "-isystem/home/damian/Projects/mill/esp-idf/components/app_update/include", "-isystem/home/damian/Projects/mill/esp-idf/components/bootloader_support/include", "-isystem/home/damian/Projects/mill/esp-idf/components/bootloader_support/bootloader_flash/include", "-isystem/home/damian/Projects/mill/esp-idf/components/esp_app_format/include", "-isystem/home/damian/Projects/mill/esp-idf/components/console", "-isystem/home/damian/Projects/mill/esp-idf/components/vfs/include", "-isystem/home/damian/Projects/mill/esp-idf/components/nvs_flash/include", "-isystem/home/damian/Projects/mill/esp-idf/components/spi_flash/include", "-I/home/damian/Projects/mill/mill-esp32/managed_components/espressif__md [... cut ...] "-fno-tree-switch-conversion", "-DconfigENABLE_FREERTOS_DEBUG_OCDAWARE=1", "-Wno-error=uninitialized", "-Wno-error=maybe-uninitialized", "-Wno-missing-field-initializers", "-Wno-error=array-bounds", "-Wno-format-nonliteral", "-Wno-format-security", "-std=gnu17", "-Wno-old-style-declaration", "-Os", "-I/home/damian/Projects/mill/build_mill_esp/config"]

- Host Machine OS: Ubuntu 22.04
- Commissioner app and versions if present: -
- Commissioner's logs if present: -

**Any additional details**
...

shubhamdp, 2023-08-30:
@damian-kurek-wizzdev If you have please share the patch. Also, can you attach the complete build logs in a file for debugging.

shubhamdp, 2023-08-30:
I tried the below patch and the build was successful. I had to rename the macro as somewhere in the code, `TEST` is being already in use.

```
diff --git a/examples/light/CMakeLists.txt b/examples/light/CMakeLists.txt
index f614a75..b2affab 100644
--- a/examples/light/CMakeLists.txt
+++ b/examples/light/CMakeLists.txt
@@ -30,6 +30,8 @@ set(PROJECT_VER_NUMBER 1)
 set(ESP_MATTER_PATH $ENV{ESP_MATTER_PATH})
 set(MATTER_SDK_PATH ${ESP_MATTER_PATH}/connectedhomeip/connectedhomeip)

+add_definitions(-DTEST_FOR_GH_ISSUE="TEST")
+
 # This should be done before using the IDF_TARGET variable.
 include($ENV{IDF_PATH}/tools/cmake/project.cmake)
 include($ENV{ESP_MATTER_DEVICE_PATH}/esp_matter_device.cmake)
```

damian-kurek-wizzdev, 2023-08-30:
I use a different PC today and it works fine as well. I will check the PC where it fails and provide you with more information.

damian-kurek-wizzdev, 2023-08-30:
I did not reproduce the previous issue but got a different one. When you mix definitions between "" and ''. I got a different error. Not sure if it is a bug since it looks handled a bit. 

add_definitions(-DTEST_FOR_GH_ISSUE="TEST")
add_definitions(-DTEST_FOR_GH_ISSUE_other='4')

FAILED: esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-configure 
cd /home/dkurek/Projects/mill/build_main_2/esp-idf/chip && /home/dkurek/Projects/mill/mill-esp32/main/src/matter_components/esp-matter/esp-matter/connectedhomeip/connectedhomeip/.environment/cipd/packages/pigweed/gn --root=/home/dkurek/Projects/mill/mill-esp32/main/src/matter_components/esp-matter/esp-matter/connectedhomeip/connectedhomeip/config/esp32 gen --check --fail-on-unused-args /home/dkurek/Projects/mill/build_main_2/esp-idf/chip && /usr/bin/cmake -E touch /home/dkurek/Projects/mill/build_main_2/esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-configure
ERROR at build arg file (use "gn args <out_dir>" to edit):15:1141: Invalid token.
target_cflags_c = ["-DAC__OLD_ACCOUNT_DEV", "-DAWS_IS_DEFAULT_CLOUD", "-DAWS__DEV", "-DBLE_DEVICE_NAME=\"Mill_Panel_G3\"", "-DCHIP_VERSION_ESP32", "-DCLOUD_RUNTIME_SWITCHING", "-DCOMPONENT_ABLE_CLOUD", "-DCOMPONENT_AWS_CLOUD", "-DCOMPONENT_BLE_CONFIGURATION", "-DCOMPONENT_CLOUD_CONTROL", "-DCOMPONENT_CLOUD_MISC_PARAMETERS", "-DCOMPONENT_DISPLAY_PANEL", "-DCOMPONENT_I2C", "-DCOMPONENT_LOCK_CHILD_OR_COMMERCIAL", "-DCOMPONENT_MATTER", "-DCOMPONENT_NVS_CONFIG_V2", "-DCOMPONENT_POWER_MONITOR", "-DCOMPONENT_SENSOR_THERMISTOR", "-DCOMPONENT_TEMPERATURE_MEASUREMENTS_MODULE", "-DCOMPONENT_TRIAC", "-DCOMPONENT_UI_GEN3", "-DCOMPONENT_UTC_TIME", "-DCOMPONENT_WEEKLY_TIMERS", "-DCOMPONENT_WIFI_LED", "-DENABLE_PRINTS_AND_LOGS", "-DFIRMWARE_VERSION_DEFINITION=\"\"0x230718\"\"", "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"", "-DMILL_PRODUCT_TYPE=\"H3P\"", "-DPRODUCT_CLASS_HEATER_PANEL", "-DPRODUCT_FAMILY_HEATER", "-DPRODUCT_GEN_3", "-DPRODUCT_HEATER", "-DPRODUCT_PANEL", "-DPRODUCT_TYPE_NAME=\"HeaterGen3Panel\"", "-DPRODUCT_VARIANT_HEATER_PANEL_BARE", "-DSUBPRODUCT_PANEL_BARE", "-DTEST_FOR_GH_ISSUE=\"TEST\"", "-DTEST_FOR_GH_ISSUE_other="'4'"", "-DUSE_REGULATOR_HYSTERESIS", "-DUSE_REGULATOR_PID", "-DUSE_SIMULATED_DISPLAY", "-DUSE_SIMULATED_SENSORS", "-DVERSION_IN_UI=\"4\"", "-isystem/home/dkurek/Projects/mill/mill-esp32/main/externals/esp-idf/components/freertos/include/freertos", "-isystem/home/dkurek/Projects/mill/mill-esp32/main/externals/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos", "-isystem/home/dkurek/Projects/mill/mill-esp32/main/externals/esp-idf/components/newlib/platform_include", "-isystem/home/dkurek/Projects/mill/mill-esp32/main/externals/esp-idf/components/freertos/include", "-isystem/home/dkurek/Projects/mill/mill-esp32/main/externals/esp-idf/components/freertos/include/esp_additions/freertos", "-isystem/home/dkurek/Projects/mill/mill-esp32/main/externals/esp-idf/components/freertos/port/xtensa/include", "-isystem/home/dkurek/Projects/mill/mill-esp32/main/externals/esp-idf/components/freertos/include/esp_additions", "-isystem/home/dkurek/Projects/mill/mill-esp32/main/externals/esp-idf/components/esp_hw_support/include", "-isystem/home/dkurek/Projects/mill/mill-esp32/main/externals/esp-idf/components/esp_hw_support/include/soc", "-isystem/home/dkurek/Projects/mill/mill-esp32/main/externals/esp-idf/components/esp_hw_support/include/soc/esp32", "-isystem/home/dkurek/Projects/mill/mill-esp32/main/externals/esp-idf/components/esp_hw_support/port/esp32/.", "-isystem/home/dkurek/Projects/mill/mill-esp32/main/externals/esp-idf/components/esp_hw_support/port/esp32/private_include", "-isystem/home/dkurek/Projects/mill/mill-esp32/main/externals/esp-idf/components/heap/include", "-isystem/home/dkurek/Projects/mill/mill-esp32/main/externals/esp-idf/components/log/include", "-isystem/home/dkurek/Projects/mill/mill-esp32/main/externals/esp-idf/components/lwip/include/apps", "-isystem/home/dkurek/Projects/mill/mill-esp32/main/externals/esp-idf/components/lwip/include/apps/sntp", "-isystem/home/dkurek/Projects/mill/mill-esp32/main/externals/esp-idf/components/lwip/lwip/src/include", "-isystem/home/dkurek/Projects/mill/mill-esp32/main/externals/esp-idf/components/lwip/port/esp32/include", "-isystem/home/dkurek/Projects/mill/mill-esp32/main/externals/esp-idf/components/lwip/port/esp32/include/arch", "-isystem/home/dkurek/Projects/mill/mill-esp32/main/externals/esp-idf/components/soc/include", "-isystem/home/dkurek/Projects/mill/mill-esp32/main/externals/esp-idf/components/soc/esp32/.", "-isystem/home/dkurek/Projects/mill/mill-esp32/main/externals/esp-idf/components/soc/esp32/include", "-isystem/home/dkurek/Projects/mill/mill-esp32/main/externals/esp-idf/components/hal/esp32/include", "-isystem/home/dkurek/Projects/mill/mill-esp32/main/externals/esp-idf/components/hal/include", "-isystem/home/dkurek/Projects/mill/mill-esp32/main/externals/esp-idf/components/hal/platform_port/include", "-isystem/home/dkurek/Projects/mill/mill-esp32/main/externals/esp-idf/components/esp_rom/include", "-isystem/home/dkurek/Projects/mill/mill-esp32/main/externals/esp-idf/components/esp_rom/include/esp32", "-isystem/home/dkurek/Projects/mill/mill-esp32/main/externals/esp-idf/components/esp_rom/esp32", "-isystem/home/dkurek/Projects/mill/mill-esp32/main/externals/esp-idf/components/esp_common/include", "-isystem/home/dkurek/Projects/mill/mill-esp32/main/externals/esp-idf/components/esp_system/include", "-isystem/home/dkurek/Projects/ [... cut ...]                                                                                                                                                                                                                                                                                                                                                                                                                                                            ^
Strings are delimited by " characters, not apostrophes.

damian-kurek-wizzdev, 2023-09-01:
Update, after cleaning everything up, downloading new repo and installing everything works fine

shubhamdp, 2023-09-01:
@damian-kurek-wizzdev Thanks for the update, can you please close the issue.

closed 2023-09-04 by dhrishi