espressif/esp-matter #602: examples->light_switch :  menuconfig <Enable Border Router>  building error!

MaplestoryAlen, 2023-08-30:
I want to enable the border router function. but there was an error in the compilation.

**Describe the bug**
home/alen/esp-idf/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Executing action: all (aliases: build)
Running ninja in directory /home/alen/esp-matter/examples/light_switch/build
Executing "ninja all"...
[0/1] Re-running CMake...-- Component directory /home/alen/esp-matter/examples/common/blemesh_platform does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/alen/esp-matter/examples/common/external_platform does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/alen/esp-matter/examples/common/relinker does not contain a CMakeLists.txt file. No component will be added
-- ccache will be used for faster recompilation
-- Building ESP-IDF components for target esp32h2
Skipping optional dependency: espressif/cmake_utilities
Skipping optional dependency: espressif/mdns
Processing 6 dependencies:
[6/6] idf (5.1.0)-- Found Python3: /usr/bin/python3.8 (found version "3.8.10") found components: Interpreter 
Skipping optional dependency: espressif/mdns
Skipping optional dependency: espressif/cmake_utilities
-- Project sdkconfig file /home/alen/esp-matter/examples/light_switch/sdkconfig
Loading defaults file /home/alen/esp-matter/examples/light_switch/sdkconfig.defaults...
Loading defaults file /home/alen/esp-matter/examples/light_switch/sdkconfig.defaults.esp32h2...
warning: default on the choice symbol SEC_CERT_DAC_PROVIDER (defined at /home/alen/esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip/Kconfig:786, /home/alen/esp-matter/components/esp_matter/Kconfig:34) will have no effect, as defaults do not affect choice symbols
warning: the choice symbol SEC_CERT_DAC_PROVIDER (defined at /home/alen/esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip/Kconfig:786, /home/alen/esp-matter/components/esp_matter/Kconfig:34) is defined with a prompt outside the choice
-- Found Python3: /home/alen/.espressif/python_env/idf5.1_py3.8_env/bin/python (found version "3.8.10") found components: Interpreter 
-- App "light_switch" version: 1.0
-- Adding linker script /home/alen/esp-matter/examples/light_switch/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /home/alen/esp-idf/components/esp_system/ld/esp32h2/sections.ld.in
-- Adding linker script /home/alen/esp-idf/components/esp_rom/esp32h2/ld/esp32h2.rom.ld
-- Adding linker script /home/alen/esp-idf/components/esp_rom/esp32h2/ld/esp32h2.rom.api.ld
-- Adding linker script /home/alen/esp-idf/components/esp_rom/esp32h2/ld/esp32h2.rom.libgcc.ld
-- Adding linker script /home/alen/esp-idf/components/esp_rom/esp32h2/ld/esp32h2.rom.newlib.ld
-- Adding linker script /home/alen/esp-idf/components/esp_rom/esp32h2/ld/esp32h2.rom.version.ld
-- Adding linker script /home/alen/esp-idf/components/esp_rom/esp32h2/ld/esp32h2.rom.wdt.ld
-- Adding linker script /home/alen/esp-idf/components/esp_rom/esp32h2/ld/esp32h2.rom.heap.ld
-- Adding linker script /home/alen/esp-idf/components/soc/esp32h2/ld/esp32h2.peripherals.ld
-- Found Python3: /home/alen/.espressif/python_env/idf5.1_py3.8_env/bin/python3 (found version "3.8.10") found components: Interpreter 
CMake Warning at /home/alen/esp-idf/tools/cmake/component.cmake:293 (message):
  No source files found for SRC_DIRS entry
  '/home/alen/esp-matter/connectedhomeip/connectedhomeip/src/app/clusters'.
Call Stack (most recent call first):
  /home/alen/esp-idf/tools/cmake/component.cmake:443 (__component_add_sources)
  /home/alen/esp-matter/components/esp_matter/CMakeLists.txt:26 (idf_component_register)

CMake Warning at /home/alen/esp-idf/tools/cmake/component.cmake:293 (message):
  No source files found for SRC_DIRS entry
  '/home/alen/esp-matter/connectedhomeip/connectedhomeip/src/app/clusters/ias-zone-client'.
Call Stack (most recent call first):
  /home/alen/esp-idf/tools/cmake/component.cmake:443 (__component_add_sources)
  /home/alen/esp-matter/components/esp_matter/CMakeLists.txt:26 (idf_component_register)

CMake Warning at /home/alen/esp-idf/tools/cmake/component.cmake:293 (message):
  No source files found for SRC_DIRS entry
  '/home/alen/esp-matter/connectedhomeip/connectedhomeip/src/app/clusters/ias-zone-server'.
Call Stack (most recent call first):
  /home/alen/esp-idf/tools/cmake/component.cmake:443 (__component_add_sources)
  /home/alen/esp-matter/components/esp_matter/CMakeLists.txt:26 (idf_component_register)

-- Components: app_bridge app_reset app_trace app_update bootloader bootloader_support bt chip cmock console cxx device driver efuse esp-tls esp_adc esp_app_format esp_coex esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_matter esp_matter_bridge esp_matter_console esp_matter_controller esp_matter_rainmaker esp_mm esp_netif esp_netif_stack esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump espressif__button espressif__esp_encrypted_img espressif__esp_secure_cert_mgr espressif__json_parser espressif__led_strip esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 iot_button json led_driver log lwip main mbedtls mqtt newlib nvs_flash openthread partition_table protobuf-c protocomm pthread riscv sdmmc soc spi_flash spiffs tcp_transport ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant
-- Component paths: /home/alen/esp-matter/examples/common/app_bridge /home/alen/esp-matter/examples/common/app_reset /home/alen/esp-idf/components/app_trace /home/alen/esp-idf/components/app_update /home/alen/esp-idf/components/bootloader /home/alen/esp-idf/components/bootloader_support /home/alen/esp-idf/components/bt /home/alen/esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip /home/alen/esp-idf/components/cmock /home/alen/esp-idf/components/console /home/alen/esp-idf/components/cxx /home/alen/esp-matter/device_hal/device /home/alen/esp-idf/components/driver /home/alen/esp-idf/components/efuse /home/alen/esp-idf/components/esp-tls /home/alen/esp-idf/components/esp_adc /home/alen/esp-idf/components/esp_app_format /home/alen/esp-idf/components/esp_coex /home/alen/esp-idf/components/esp_common /home/alen/esp-idf/components/esp_eth /home/alen/esp-idf/components/esp_event /home/alen/esp-idf/components/esp_gdbstub /home/alen/esp-idf/components/esp_hid /home/alen/esp-idf/components/esp_http_client /home/alen/esp-idf/components/esp_http_server /home/alen/esp-idf/components/esp_https_ota /home/alen/esp-idf/components/esp_https_server /home/alen/esp-idf/components/esp_hw_support /home/alen/esp-idf/components/esp_lcd /home/alen/esp-idf/components/esp_local_ctrl /home/alen/esp-matter/components/esp_matter /home/alen/esp-matter/components/esp_matter_bridge /home/alen/esp-matter/components/esp_matter_console /home/alen/esp-matter/components/esp_matter_controller /home/alen/esp-matter/components/esp_matter_rainmaker /home/alen/esp-idf/components/esp_mm /home/alen/esp-idf/components/esp_netif /home/alen/esp-idf/components/esp_netif_stack /home/alen/esp-idf/components/esp_partition /home/alen/esp-idf/components/esp_phy /home/alen/esp-idf/components/esp_pm /home/alen/esp-idf/components/esp_psram /home/alen/esp-idf/components/esp_ringbuf /home/alen/esp-idf/components/esp_rom /home/alen/esp-idf/components/esp_system /home/alen/esp-idf/components/esp_timer /home/alen/esp-idf/components/esp_wifi /home/alen/esp-idf/components/espcoredump /home/alen/esp-matter/examples/light_switch/managed_components/espressif__button /home/alen/esp-matter/examples/light_switch/managed_components/espressif__esp_encrypted_img /home/alen/esp-matter/examples/light_switch/managed_components/espressif__esp_secure_cert_mgr /home/alen/esp-matter/examples/light_switch/managed_components/espressif__json_parser /home/alen/esp-matter/examples/light_switch/managed_components/espressif__led_strip /home/alen/esp-idf/components/esptool_py /home/alen/esp-idf/components/fatfs /home/alen/esp-idf/components/freertos /home/alen/esp-idf/components/hal /home/alen/esp-idf/components/heap /home/alen/esp-idf/components/http_parser /home/alen/esp-idf/components/idf_test /home/alen/esp-idf/components/ieee802154 /home/alen/esp-matter/device_hal/button_driver/iot_button /home/ale [... cut ...]
-- Build files have been written to: /home/alen/esp-matter/examples/light_switch/build
ninja: error: '/home/alen/esp-idf/components/openthread/lib/esp32h2/libopenthread_br.a', needed by 'light_switch.elf', missing and no known rule to make it
ninja failed with exit code 1, output of the command is in the /home/alen/esp-matter/examples/light_switch/build/log/idf_py_stderr_output_970860 and /home/alen/esp-matter/examples/light_switch/build/log/idf_py_stdout_output_970860

 
**Environment**
- ESP-Matter  :latest version
- ESP-IDF Commit Id: ea5e0ff branch of  V5.1
- SoC : ESP32H2

shubhamdp, 2023-08-30:
This is what you should follow for building a border router: https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_br or https://github.com/espressif/esp-thread-br/tree/main/examples/basic_thread_border_router

MaplestoryAlen, 2023-08-30:
> This is what you should follow for building a border router: https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_br or https://github.com/espressif/esp-thread-br/tree/main/examples/basic_thread_border_router

  I have built the border router and it can be used normally, but I want to enable matter light_switch REED (Router Eligible End Device) for normal use when the leader is disconnected

wqx6, 2023-08-31:
Our examples are set as REED by default. The device will be promoted to router after it is attached to a Thread network.

wqx6, 2023-08-31:
One question: why do you want to enable Thread Border Router on ESP32-H2? The `openthread_br` lib for ESP32-H2 is not provided in ESP-IDF, and we are not planing to support BR on ESP32-H2 because of the limited RAM on it. We are planing to support BR feature with matter on other Wi-Fi chips like ESP32-S3. It can work as a Thread BR with a RCP and is able to be commissioned to a Matter fabric.

MaplestoryAlen, 2023-09-07:
> Our examples are set as REED by default. The device will be promoted to router after it is attached to a Thread network.

Okay, thank you

dhrishi, 2023-10-14:
Closing this now. Feel free to re-open in case of any problems

closed 2023-10-14 by dhrishi