espressif/esp-matter #603: Errors during initializing the matter node for Room Air Conditioner in the example code

aggaddam, 2023-08-30:
Describe the bug
Observed few warnings/errors while running the matter room air conditioner example under esp-matter sdk

Environment :

    ESP-Matter Commit Id: https://github.com/espressif/esp-matter/commit/68e99de70266674ce172459136960bbb5c58dc0e
    ESP-IDF Commit Id: e088c3766ba440e72268b458a68f27b6e7d63986
    SoC (eg: ESP32 or ESP32-C3): ESP32-C3
    Device Logs (Please attach the log file): 
[image]

    Host Machine OS: Linux
    Commissioner app and versions if present: 
    Commissioner's logs if present:

Any additional details
...

PSONALl, 2023-08-30:
@aggaddam This problem has been fixed in internal repo, the fix will be available soon

dhrishi, 2023-09-14:
@aggaddam This if fixed and pushed to Github

closed 2023-09-14 by dhrishi

aggaddam, 2023-09-19:
@dhrishi Which commit of ESP-matter has this fix?