espressif/esp-matter #604: App component outside main folder casuses error during build

damian-kurek-wizzdev, 2023-08-30:
**Describe the bug**
If I put my esp component that builds executable for mcu in a folder different than "main" I get cmake error while somewhere in matter. If I rename the folder to main everything works fine. 
I think that renaming the folder in any example and adding it to EXTRA_COMPONENT_DIRS should trigger the issue.
Esp-idf projects worked fine when an executable component was in a different folder than "main".

CMake Error at main/externals/esp-idf/tools/cmake/component.cmake:368 (__component_get_property):
  __component_get_property Function invoked with incorrect arguments for
  function named: __component_get_property
Call Stack (most recent call first):
  main/src/matter_components/esp-matter/esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip/CMakeLists.txt:322 (idf_component_get_property)

cmake extra component: 
set(EXTRA_COMPONENT_DIRS
    "main2"
    "${ESP_MATTER_PATH}/examples/common"
    "${MATTER_SDK_PATH}/config/esp32/components"
    "${ESP_MATTER_PATH}/components"
    "${ESP_MATTER_PATH}/device_hal/device"
    ${extra_components_dirs_append})

**Environment**
- ESP-Matter Commit Id: release 1.0
- ESP-IDF Commit Id: 4.4.3
- SoC (eg: ESP32 or ESP32-C3): ESP32
- Device Logs (Please attach the log file):
- Host Machine OS: Ubuntu 22.04 
- Commissioner app and versions if present:
- Commissioner's logs if present:

**Any additional details**
...

shubhamdp, 2023-10-16:
@damian-kurek-wizzdev I think that is expected, please check [here](https://github.com/espressif/connectedhomeip/blob/4088a77f557e8571a39338fad51a1d8eb0131d79/config/esp32/components/chip/CMakeLists.txt#L322C1-L323C1). This file explicitly tries to link the `main` component.

If you want this to build, you may have to change here as well. Since the śp-idf projects uses only CMake build system that may have worked there. But, Matter uses CMake along with GN so its a bit complicated setup.

damian-kurek-wizzdev, 2023-10-16:
@shubhamdp If it is expected behaviour some meaningful error should be thrown to a user. For example:
"Your executable component should be located in a directory called 'main'." Or something like that. For now, it is black magic that took me 1 week to figure out.

shubhamdp, 2024-07-25:
@damian-kurek-wizzdev @ShahilDholariya I have added the changes in this PR to support renaming main... https://github.com/project-chip/connectedhomeip/pull/34497

Will update the submodule once it gets merged.

dhrishi, 2024-09-02:
Fixed with the connectedhomeip submodule update.

closed 2024-09-02 by dhrishi