espressif/esp-matter #605: Menuconfig configuration for only on network pairing

damian-kurek-wizzdev, 2023-08-30:
**Describe the question/query that you have**
I want to configure my existing project with matter to use only On-network pairing. I looked into documentation ([https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#]) and examples. I could not find any info on how to do it.
All my tries with disabling stuff in "CHIP device Layer" ended up in compilation errors

Could someone provide me with some links to examples or documentation that can help me with menuconfig configuration?

shubhamdp, 2023-08-30:
For on network pairing your device needs to be on the wifi network. For that you can either use the cli on the device with command `matter esp wifi connect ssid passphrase`. Or use the `CONFIG_DEFAULT_WIFI_SSID` and `CONFIG_DEFAULT_WIFI_PASSWORD` options to configure wifi network. 

Once your device is connected to wifi you can use the on network pairing.

damian-kurek-wizzdev, 2023-08-30:
A bit of explanation. I have an existing code base. The firmware already connects to wifi and I want to let a user connect my device using matter to home automation devices. I don't want to edit the current functionality since there are thousands of devices in the market and the code for wifi handling has been working fine for a few years. 
Can I configure esp-matter to start on network pairing without reconfiguration of wifi or BLE? (I will provide a Wi-Fi connection myself)
Could you point me to some examples or documentation?
For now, I need to use esp-matter 1.0 since I don't want to update esp-idf to 5.x for now

shubhamdp, 2023-09-01:
Thanks for the more explanation. Well in that case, here's the similar issue: https://github.com/espressif/esp-matter/issues/551.

If you do not want to use the BLE for commissioning then you can disable `CONFIG_ENABLE_CHIPOBLE` options. 

For, wifi, call to [`InitWiFiStack`](https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_core.cpp#L1010), if this asserts/fails, try commenting it and see if that works.

damian-kurek-wizzdev, 2023-09-01:
I had to change some things in CHIP API since I work with esp-matter 1.0 but I managed to run something.  Now I'm note sure if it works. I found somewhere that the logs can be ignored. Is this expected behaviour?

I (9204) chip[DMG]: AccessControlCluster: initializing
I (9214) chip[ZCL]: 0x3f4349edep 0 clus 0x0000_0030 attr 0x0000_0000 not supported
I (9224) chip[ZCL]: Initiating Admin Commissioning cluster.
E (9224) chip[ZCL]: Duplicate attribute override registration failed
I (9234) chip[DIS]: Updating services using commissioning mode 0
I (9244) chip[DIS]: CHIP minimal mDNS started advertising.
E (9254) chip[SVR]: ERROR setting up transport: 6
I (9254) esp_matter_core: Dynamic endpoint 0 added
I (9264) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0003's Attribute 0x0001 is 0 **********
I (9274) MatterThermostatDriver: matterAttributeUpdate: PRE_UPDATE
I (9274) esp_matter_core: Dynamic endpoint 1 added
E (9284) chip[DL]: Long dispatch time: 149 ms, for event type 2
I (9294) chip[DL]: Updating advertising data

damian-kurek-wizzdev, 2023-09-01:
One more thing. Can I use this QR code (link below) for device pairing with Google Home app?

https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#:~:text=Alternatively%2C%20you%20can%20scan%20the%20below%20QR%20code%20image%20using%20Matter%20commissioners.

damian-kurek-wizzdev, 2023-09-01:
I managed to commission a new device, but I got an error from google home that the device was not certified. I guess it is expected and I need to use a chip tool to communicate with it.

damian-kurek-wizzdev, 2023-09-04:
@shubhamdp I did something similar and managed to make it work (I guess). I will post a patch for connecthomeip when I'm sure that it works (patch for release 1.0). I think your changes will allow the same functionality. My only concern is that if a user configured matter to use WiFi AP via menuconfig and then created WiFi AP with a different ssid it won't work for him. I guess some additional checks would be required to check if the WiFi AP name on esp32 is the same as the name and password as one configured in menuconfig. Not sure if it explains my concerns.

damian-kurek-wizzdev, 2023-09-04:
@dhrishi Thanks that helped a lot. 
In case someone will look here for a solution for the "device was not certified":

1. Go to https://console.home.google.com/projects and add your test device
2. Enable developer mode on your mobile phone 
3. Remember to use the same email when adding the test device and in Google Home app

damian-kurek-wizzdev, 2023-09-05:
I can commission the device and everything works fine, but after I reboot, the device goes offline at Google Home and never returns. I can see that the matter server starts on esp32. Any idea @shubhamdp?

shubhamdp, 2023-09-05:
Can you share the logs?

vaemc, 2023-09-05:
> @damian-kurek-wizzdev [project-chip/connectedhomeip#29017](https://github.com/project-chip/connectedhomeip/pull/29017) might help in your case.

@shubhamdp  Hi. when will the connectedhomeip of esp-matter sdk be updated to project-chip/connectedhomeip#29017?

damian-kurek-wizzdev, 2023-09-05:
[logs.txt](https://github.com/espressif/esp-matter/files/12520731/logs.txt)

Ignore error messages in comms and MatterThermostatDriver I added them for debuging

damian-kurek-wizzdev, 2023-09-05:
Release 1.0 patch I used to make it work

diff --git a/src/app/server/CommissioningWindowManager.cpp b/src/app/server/CommissioningWindowManager.cpp
index 6ce32d01bb..805ac8ac05 100644
--- a/src/app/server/CommissioningWindowManager.cpp
+++ b/src/app/server/CommissioningWindowManager.cpp
@@ -24,6 +24,7 @@
 #include <platform/CHIPDeviceLayer.h>
 #include <platform/CommissionableDataProvider.h>
 #include <platform/DeviceControlServer.h>
+#include "esp_log.h"
 
 using namespace chip::app::Clusters;
 using namespace chip::System::Clock;
@@ -280,13 +281,14 @@ CHIP_ERROR CommissioningWindowManager::OpenBasicCommissioningWindow(Seconds16 co
                                                                     CommissioningWindowAdvertisement advertisementMode)
 {
     RestoreDiscriminator();
+    SetBLE(false);
 
 #if CONFIG_NETWORK_LAYER_BLE
     // Enable BLE advertisements if commissioning window is to be opened on all supported
     // transports, and BLE is supported on the current device.
-    SetBLE(advertisementMode == chip::CommissioningWindowAdvertisement::kAllSupported);
+//SetBLE(advertisementMode == chip::CommissioningWindowAdvertisement::kAllSupported);
 #else
-    SetBLE(false);
+   // SetBLE(false);
 #endif // CONFIG_NETWORK_LAYER_BLE
 
     mFailedCommissioningAttempts = 0;
diff --git a/src/include/platform/internal/GenericPlatformManagerImpl.ipp b/src/include/platform/internal/GenericPlatformManagerImpl.ipp
index 4cee5dab0b..beefce15fb 100644
--- a/src/include/platform/internal/GenericPlatformManagerImpl.ipp
+++ b/src/include/platform/internal/GenericPlatformManagerImpl.ipp
@@ -93,12 +93,12 @@ CHIP_ERROR GenericPlatformManagerImpl<ImplClass>::_InitChipStack()
 
     // Initialize the CHIP BLE manager.
 #if CHIP_DEVICE_CONFIG_ENABLE_CHIPOBLE
-    err = BLEMgr().Init();
-    if (err != CHIP_NO_ERROR)
-    {
-        ChipLogError(DeviceLayer, "BLEManager initialization failed: %" CHIP_ERROR_FORMAT, err.Format());
-    }
-    SuccessOrExit(err);
+//    err = BLEMgr().Init();
+//    if (err != CHIP_NO_ERROR)
+//    {
+//        ChipLogError(DeviceLayer, "BLEManager initialization failed: %" CHIP_ERROR_FORMAT, err.Format());
+//    }
+//    SuccessOrExit(err);
 #endif
 
     // Initialize the Connectivity Manager object.
diff --git a/src/platform/ESP32/PlatformManagerImpl.cpp b/src/platform/ESP32/PlatformManagerImpl.cpp
index 074750654c..f7af125ba1 100644
--- a/src/platform/ESP32/PlatformManagerImpl.cpp
+++ b/src/platform/ESP32/PlatformManagerImpl.cpp
@@ -44,6 +44,7 @@
 #endif
 #include "esp_system.h"
 #include "esp_wifi.h"
+#include "esp_log.h"
 
 namespace chip {
 namespace DeviceLayer {
@@ -70,56 +71,55 @@ CHIP_ERROR PlatformManagerImpl::_InitChipStack(void)
     // Make sure the LwIP core lock has been initialized
     ReturnErrorOnFailure(Internal::InitLwIPCoreLock());
 
[... 77 lines cut ...]
+//        // Initialize the ESP WiFi layer.
+//        cfg = WIFI_INIT_CONFIG_DEFAULT();
+//        err = esp_wifi_init(&cfg);
+//        if (err != ESP_OK)
+//        {
+//            goto exit;
+//        }
+
+//        esp_wifi_get_mode(&mode);
+//        if ((mode == WIFI_MODE_AP) || (mode == WIFI_MODE_APSTA))
+//        {
+//            esp_fill_random(ap_mac, sizeof(ap_mac));
+//            /* Bit 0 of the first octet of MAC Address should always be 0 */
+//            ap_mac[0] &= (uint8_t) ~0x01;
+//            err = esp_wifi_set_mac(WIFI_IF_AP, ap_mac);
+//            if (err != ESP_OK)
+//            {
+//                goto exit;
+//            }
+//        }
+//    }
+//#endif // CHIP_DEVICE_CONFIG_ENABLE_WIFI
     ReturnErrorOnFailure(chip::Crypto::add_entropy_source(app_entropy_source, NULL, 16));
 
     // Call _InitChipStack() on the generic implementation base class

shubhamdp, 2023-09-05:
> @shubhamdp Hi. when will the connectedhomeip of esp-matter sdk be updated to [project-chip/connectedhomeip#29017](https://github.com/project-chip/connectedhomeip/pull/29017)?

We are in the process of updating the submodule, will be out soon.

damian-kurek-wizzdev, 2023-09-05:
You are right uncommenting the IP/wifi handlers seems to fix it. I will fix the patch and post it again.

sxsx2yzyz, 2023-09-06:
Can this plan pass CSA testing and can this product be launched? (Users must use my app+blufi for network configuration before they can use Google home, Alexa, or Apple home for network pairing)

damian-kurek-wizzdev, 2023-09-11:
Everything works fine and I can control my device using matter. Thanks for help @shubhamdp  @dhrishi

closed 2023-09-11 by damian-kurek-wizzdev