espressif/esp-matter #606: Level Control Cluster with Bridge-app

HARRYhelenH, 2023-08-31:
In the bridge-app It is given for the on /off light where I need to integrate the Dimmer instead on on/off light.
Does bridge-app example support Dimmers.
The changes made for Dimmer:
[dimmer_changes.txt](https://github.com/espressif/esp-matter/files/12483778/dimmer_changes.txt)
with these changes I observed that:

1. the attribute is mismatched(current level)
2. Once it read the callback doesn't go to write callback.

Kindly share the resources if available.

wqx6, 2023-08-31:
Looks like you are using the bridge-app from upstream `connectedhomeip` repo. For level control cluster, we must call MatterLevelControlPluginServerInitCallback before enabling the dynamic endpoint to make the cluster work.

If you use APIs from esp-matter-bridge this PluginServerInit Callback will be call in [plugin_init_callback_endpoin](https://github.com/espressif/esp-matter/blob/main/components/esp_matter_bridge/esp_matter_bridge.cpp#L159) and the dimmable-light can be create with [create_device](https://github.com/espressif/esp-matter/blob/main/components/esp_matter_bridge/esp_matter_bridge.cpp#L246).

HARRYhelenH, 2023-08-31:
@wqx6  I'm using bridge-app examples from [here](https://github.com/espressif/connectedhomeip/tree/4088a77f557e8571a39338fad51a1d8eb0131d79/examples/bridge-app/esp32), where we don't have any matter related file includes in it. When I try to add the header files it throws the error as
 No such file or directory
 #include <esp_matter.h>
any reference docs for the matter bridge?

wqx6, 2023-08-31:
> I'm using bridge-app examples from [here](https://github.com/espressif/connectedhomeip/tree/4088a77f557e8571a39338fad51a1d8eb0131d79/examples/bridge-app/esp32).

In that example, you should call `MatterLevelControlPluginServerInitCallback()` after you call `Esp32AppServer::Init()` to use the LevelControl Cluster in Dynamic endpoint.

> any reference docs for the matter bridge?

**9.12. Bridge for non-Matter devices** in SPEC might help you

HARRYhelenH, 2023-08-31:
Could you share the location of the file where the **MatterLevelControlPluginServerInitCallback(**) function exist.

wqx6, 2023-08-31:
Sorry for the wrong reply. I have changed the function in the last comment. emberAfLevelControlClusterServerInitCallback() is in `connectedhomeip/src/app/clusters/level-control.cpp`

wqx6, 2023-08-31:
The `DECLARE_DYNAMIC_CLUSTER` doesn't include the init function so the cluster initialization function is not called for the dynamic endpoint, that's why the level control cluster doesn't work on dynamic endpoint.

wqx6, 2023-09-04:
The input of `emberAfLevelControlClusterServerInitCallback` is endpoint ID, not endpoint index.

HARRYhelenH, 2023-09-04:
named the variable as endpoint_index, but passing the endpoint only.

HARRYhelenH, 2023-09-08:
The logs are as this when the level functions are not being called
[level_bridge.txt](https://github.com/espressif/esp-matter/files/12557892/level_bridge.txt)

HARRYhelenH, 2023-10-03:
I'm using 
**esp-idf: v4.4.3
Matter: release/v1.0**
 When ever I change the level in the dimmer through alexa the level cluster is not being called. Tried the above configurations as suggested but still getting the same issue.

wqx6, 2024-04-25:
[bridge-level-control.txt](https://github.com/espressif/esp-matter/files/15105625/bridge-level-control.txt)
Could you apply this patch to your example? I tested and the external write callback was called when I sent the movetolevel command to the endpoint.

dhrishi, 2024-12-23:
Closing this now

closed 2024-12-23 by dhrishi