espressif/esp-matter #608: BLE Memory is not released when commissionMgr.CloseCommissioningWindow() is called and closed succesfully

aggaddam, 2023-08-31:
**Describe the bug**
I am doing custom onboarding of my device and then stopping ble commissioning window and then connecting to wifi network received over our custom onboarding process. I was able to make the device connect to wifi network. But seeing available memory is less.

It appears that BLE memory is not released though the Ble commissioning window got closed successfully by calling commissionMgr.CloseCommissioningWindow() forcibly from the application and if there are no fabrics in the device.

**Environment**
- ESP-Matter Commit Id: https://github.com/espressif/esp-matter/commit/68e99de70266674ce172459136960bbb5c58dc0e
- ESP-IDF Commit Id: v5.1.1 (Stable)
- SoC (eg: ESP32 or ESP32-C3): ESP32C3
- Device Logs (Please attach the log file):
- Host Machine OS:
- Commissioner app and versions if present:
- Commissioner's logs if present:

**Any additional details**
...

shubhamdp, 2023-09-01:
@aggaddam Right now, BLE memory is released when one receives an event [`kCHIPoBLEConnectionClosed`](https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_core.cpp#L916). (Assuming the relevant config options are enabled). This gets called during very first standard Matter commissioning flow.

For manufacturer specific commissioning flows, you can free the ble memory yourself after calling `commissionMgr.CloseCommissioningWindow()` or in event `kCommissioningWindowClosed`.

aggaddam, 2023-09-01:
@shubhamdp Is there any API to free up memory or otherwise I will do this in application like it does in esp_matter_core.cpp  ?

```

#if CONFIG_BT_NIMBLE_ENABLED
            if (!ble_hs_is_enabled()) {
                ESP_LOGI(TAG, "BLE already deinited");
                return;
            }

            if (nimble_port_stop() != 0) {
                ESP_LOGE(TAG, "nimble_port_stop() failed");
                return;
            }
            vTaskDelay(100); // TODO: BT-3248
            nimble_port_deinit();
#if ESP_IDF_VERSION < ESP_IDF_VERSION_VAL(5, 0, 0)
            err = esp_nimble_hci_and_controller_deinit();
#endif
#endif /* CONFIG_BT_NIMBLE_ENABLED */
#if CONFIG_IDF_TARGET_ESP32
            err |= esp_bt_mem_release(ESP_BT_MODE_BTDM);
#elif CONFIG_IDF_TARGET_ESP32C2 || CONFIG_IDF_TARGET_ESP32C3 || CONFIG_IDF_TARGET_ESP32S3 || CONFIG_IDF_TARGET_ESP32H2
            err |= esp_bt_mem_release(ESP_BT_MODE_BLE);
#endif
            if (err != ESP_OK) {
                ESP_LOGE(TAG, "BLE deinit failed");
                return;
            }
            ESP_LOGI(TAG, "BLE deinit successful and memory reclaimed");
```

shubhamdp, 2023-09-01:
There isn't any abstract API right now, you will have to do it in application.

aggaddam, 2023-09-01:
@shubhamdp  The graceful approach for freeing the chip ble memory is to reboot the device after chip wifi station gets provisioned with custom commission flow. This way I don't need manage code to release the ble free memory instead it handled during chip stack intialization

shubhamdp, 2023-09-05:
@aggaddam If things are looking good, can you please close the issue.

closed 2023-09-06 by aggaddam