espressif/esp-matter #609: Unable to commission using new generated manufacturing binary

anuarrozman, 2023-09-01:
**Unable to commission using new generated manufacturing binary**
I currently have multiple ESP32 for testing. I want to have different setup passcode. Im following [Manufacturing Partition Generator Utility](https://github.com/espressif/esp-matter/blob/fd842d0/tools/mfg_tool/README.md#manufacturing-partition-generator-utility). I can't manage to successfully commission the device. How can I change it?

**Command used:**
`./mfg_tool.py -v 0xFFF1 -p 0x8000 --passcode 20202023 --discriminator 3000`
`esptool.py -p /dev/tty.usbserial-0001 erase_flash`
`idf.py -p /dev/tty.usbserial-0001 build erase-flash flash`
`esptool.py -p /dev/tty.usbserial-0001 write_flash 0x10000 ${HOME}/esp/esp-matter/tools/mfg_tool/out/fff1_8000/44293a61-1948-4a9e-a7ca-d9f3de2e8510/44293a61-1948-4a9e-a7ca-d9f3de2e8510-partition.bin`

shubhamdp, 2023-09-01:
@anuarrozman What would help is more details about the problem. I'ts hard to just figure our what could be the problem by above details. Please fill in the below details.

**Environment**
- ESP-Matter Commit Id:
- ESP-IDF Commit Id:
- SoC (eg: ESP32 or ESP32-C3):
- Device Logs (Please attach the log file):
- Host Machine OS:
- Commissioner app and versions if present:
- Commissioner's logs if present:

anuarrozman, 2023-09-04:
Hi @shubhamdp , thanks for your reply. Here are the details that you requested:

**Environment**

- ESP-Matter Commit Id: 08b277eb92f7231aa0c511d79b76acd68fb8e571
- ESP-IDF Commit Id: ea5e0ff298e6257b31d8e0c81435e6d3937f04c7
- SoC (eg: ESP32 or ESP32-C3): ESP32 
- Device Logs : [Device_Log_ESP32.docx](https://github.com/espressif/esp-matter/files/12508762/Device_Log_ESP32.docx)
- Host Machine OS: MacOS
- Commissioner app and versions if present: AppleHomeKit App

shripad621git, 2023-09-04:
@anuarrozman you need to use the qrcode generated in the output folder of mfg_tool while generating the factory partition. In your case it is at ```${HOME}/esp/esp-matter/tools/mfg_tool/out/fff1_8000/44293a61-1948-4a9e-a7ca-d9f3de2e8510/44293a61-1948-4a9e-a7ca-d9f3de2e8510-qrcode.png.```
When you flash a factory_partition binary using mfg_tool you can check the associated onboarding codes in out/{vid_pid}/{uuid}/uuid-onb_codes.csv.
**for ex:**: ```out/fff1_8000/44293a61-1948-4a9e-a7ca-d9f3de2e8510/44293a61-1948-4a9e-a7ca-d9f3de2e8510-onb_codes.csv```.

anuarrozman, 2023-09-04:
Thank you for you response. Im using the QR code generated and also getting the pairing key and QR code from `matter onboardingcodes none` command. Is it necessary to change the device configuration in as per this guide [ESP-Matter mfg_tool.py](https://github.com/espressif/esp-matter/tree/main/tools/mfg_tool) ?

shripad621git, 2023-09-06:
@anuarrozman  Yes, you need to change the device configuration. You can refer to [this](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/esp32/factory_data.md) link for having a better understanding of what menuconfig options to choose for the data flashed in factory partition.

anuarrozman, 2023-09-06:
Thank you for your reply. For the last part in the ConfigurationOption, should I change to fctry or nfs ? 

`ESP32 implementation reads factory data from nvs partition, chip-factory data must be flashed into the configured nvs partition. Factory partition can be configured using CONFIG_CHIP_FACTORY_NAMESPACE_PARTITION_LABEL option, default is "nvs".

[Component config -> CHIP Device Layer -> Matter Manufacturing Options -> chip-factory namespace partition label]`

Because, Ive followed the configuration guide before but when using `matter config` it shows error 45 and does not display the correct value of the new passcode I generated.

shripad621git, 2023-09-08:
@anuarrozman, if you are flashing the generated partition at the address which corresponds to the ```nvs``` label, then leave the option to be nvs. If you set ```CONFIG_CHIP_FACTORY_NAMESPACE_PARTITION_LABEL``` option to ```fctry``` then you need to flash the partition at the address corresponding to the ```fctry``` label which you can get in the ```partitions.csv``` file of the application.

dhrishi, 2023-09-15:
With https://github.com/espressif/esp-matter/issues/612 , I assume this can be closed too

closed 2023-09-15 by dhrishi

anuarrozman, 2023-09-15:
Yeap, Thank you so much for your help.