espressif/esp-matter #610: Temporarily unmount and restore endpoint capabilities

maenkai, 2023-09-01:
**Is your feature request related to a problem? Please describe.**
The current sdk erases all of its endpoint information once it destory the endpoint

**Describe the solution you'd like**
For example, I can temporarily unbridge endpoint 2, modify endpoint 2's properties (such as name) and cluster, and then resuspend

**Describe alternatives you've considered**
null

**Additional context**
null

jadhavrohit924, 2023-09-15:
I would suggest don't destroy the endpoint. Instead add the properties that you want on the endpoint and re-enable the endpoint as discussed here https://github.com/espressif/esp-matter/issues/576#issuecomment-1687997467

jadhavrohit924, 2023-09-25:
@maenkai Does the above comment help? Please close the issue if your issue is resolved.