espressif/esp-matter #611: Difference between client and controller of esp-matter

maenkai, 2023-09-01:
**Describe the question/query that you have**
1. Creating an endpoint in client mode can control standard matter devices, while controller mode can also control standard matter devices. I can't tell the difference between the two. It seems that the controller mode can directly replace the client mode without creating an endpoint
2. I want to implement the same role as homepod on esp's platform, should I choose to use the first (client) or the second (controller)
3. If the second method is used, does it need to be certified by matter? The second does not require an endpoint to be built and cannot be tested by the TH tool

wqx6, 2023-09-04:
> Creating an endpoint in client mode can control standard matter devices, while controller mode can also control standard matter devices. I can't tell the difference between the two. It seems that the controller mode can directly replace the client mode without creating an endpoint

The client cluster only implies that the device might send commands related to that particular cluster to other devices, but it is not mandatory for the device to have the ability to send those commands. In fact, every Matter device can send commands to other devices if it is able to form the Matter packets.

> I want to implement the same role as homepod on esp's platform, should I choose to use the first (client) or the second (controller)

HomePod is a controller in Matter fabric, so the second (controller) is recommended.

> If the second method is used, does it need to be certified by matter? The second does not require an endpoint to be built and cannot be tested by the TH tool

It needs to be certified as a Software Component.

maenkai, 2023-09-04:
Ok, thank you very much. I would like to know if esp32 supports csa Alliance software component certification? I would also like to know the documentation related to software component certification

wqx6, 2023-09-11:
> I would like to know if esp32 supports csa Alliance software component certification?

We don't have the Software Component certification currently

> I would also like to know the documentation related to software component certification

You can fetch the documents from the CSA working group.

maenkai, 2023-09-11:
> > I would like to know if esp32 supports csa Alliance software component certification?
> 
> We don't have the Software Component certification currently
> 
> > I would also like to know the documentation related to software component certification
> 
> You can fetch the documents from the CSA working group.

Can the controller I developed with esp32 be certified as a software component?  I need to develop it into a formal matter controller product

closed 2023-12-04 by maenkai