espressif/esp-matter #612: Generate Unique Commissioning QR Code for Multiple ESP Matter Device

anuarrozman, 2023-09-04:
In the current setup, we employ identical QR codes for all matters or items under consideration. Is there a way to generate unique QR codes for each individual device or item within our system? Additionally, we are looking for guidance on the process of modifying the product and vendor IDs associated with these devices. Any insights or recommendations to resolve these matters would be greatly appreciated.

sayondeep, 2023-09-04:
@anuarrozman You can create unique QR code for each device by generating and flashing factory partition to your devices. You can refer [this] (https://github.com/espressif/esp-matter/tree/main/tools/mfg_tool) for how to generate and flash the factory partition.

anuarrozman, 2023-09-04:
I follow that guide, but simply got error when trying to display Matter device config with this command `matter config`. Is it necessary to change the device configuration in as per this guide [ESP-Matter mfg_tool.py](https://github.com/espressif/esp-matter/tree/main/tools/mfg_tool) ?

anuarrozman, 2023-09-05:
Hi, I would like to share that I managed to commission using chip tool but chip tool command that use <discriminator> and <passcode>. But when I try to use chip tool command that use <qrcode_payload-or-manual_code> it show this error:

`[1693874122859] [26252:168038] [SC] Failed to verify peer's MAC. This can happen when setup code is incorrect.`
`[1693874122859] [26252:168038] [SC] Failed during PASE session setup: src/crypto/CHIPCryptoPALOpenSSL.cpp:1437: CHIP Error 0x000000AC: Internal error
`

This is how I generate the new pairing key:

`./mfg_tool.py --discriminator 3131 --passcode 20201111 --vendor-id 65521 --product-id 32768 `

And here are what I get from running "matter config" on my matter device log:

_**> matter config

VendorId:        65521 (0xFFF1)
ProductId:       32768 (0x8000)
HardwareVersion: 0 (0x0)
PinCode:         20202021
Discriminator:   c3b
Done
> matter onboardingcodes none

QRCode:            MT:Y.K90MPP03780648G00
QRCodeUrl:         https://project-chip.github.io/connectedhomeip/qrcode.html?data=MT%3AY.K90MPP03780648G00
ManualPairingCode: 30054912331
Done**_

The PinCode didn't change to the new one which is 20201111.

dhrishi, 2023-09-05:
@anuarrozman The CLIs `matterconfig` and `matter onboardingcodes none` do not fetch the data from the factory partition. And hence, they should not be used in this case. Instead you should use the manual code that is generated in the CSV file:

`tools/mfg_tool/out/<vid_pid>/<abc-xyz-lmn>$ vi abc-xyz-lmn-onb_codes.csv`

This path will be printed when you use the mfg tool to generate the factory partition. With this, you should be able to commission the device.

anuarrozman, 2023-09-05:
Thanks @dhrishi . Im having problem when using chip-tool commission option 
`./chip-tool pairing code <node_id> <qrcode_payload-or-manual_code>`
Here is my command:
`./chip-tool pairing code 1 30054912331`

Commissioning unsuccessful.

dhrishi, 2023-09-05:
@anuarrozman How did you get the manual pairing code? What are the values of the discriminator and passcode that have been generated through the mfg tool?

anuarrozman, 2023-09-05:
I get the manual pairing code from the generated .csv file. These are the discriminator and passcode:

- Discriminator: 3131
- Passcode: 20201111
- Vendor ID: 0xFFF1
- Prod ID: 0x8000

dhrishi, 2023-09-06:
@anuarrozman The manual pairing code `30054912331` does not seem to match the above discriminator and passcode values. Please check if you have used the correct file (like abc-lmn-pqr-xyz-onb_codes.csv) that is generated by the mfg_tool in tools/mfg_too/out/<vid>_<pid>/<abc-lmn-pqr-xyz>` folder.

You can use chip-tool command to parse the manual code
```
chip-tool payload parse-setup-payload <setup-payload>
```

Yes, you will have to change the menuconfig options so that the appropriate values are picked up from the factory partition.

anuarrozman, 2023-09-06:
Thank you for your reply. Im sorry, I share the wrong pairing code. For the menuconfig options , i need to use nfs partition or change to factory partition. If change to factory, I need to use factory partition during flashing the partition bin right ? instead of using default nvs partition

anuarrozman, 2023-09-12:
Hi, sorry for the late reply. I was not available these past few days. 

I already managed to create unique QR Code and Pairing Key, but when trying to read the `matter onboardingcodes none`, it printed the wrong one. So the correct one is in the csv file generated during creating the unique QR Code and Pairing Key. 

I've followed all the menuconfig steps, when using `matter config` it shows error and `matter onboardingcodes none` still shows the wrong pairing key and QR Code.

anuarrozman, 2023-09-14:
Oh, thanks. My last question about the setup payload is about the serial number. Using `mfg_tool.py`, there's an option to add serial number, but how can i read it ? Is there a way to retrieve the serial number ?

shubhamdp, 2023-09-15:
Setup payload do not contain the serial number. Please check Matter Specification section 5.1.1 for  more info on Onboarding Payload Contents.
Serial number is an attribute in basic information cluster. You can read it as any other attribute.

anuarrozman, 2023-09-15:
Thanks @shubhamdp . Thats really helpful. Ill close this issue. Thank you @dhrishi for helping me.

closed 2023-09-15 by anuarrozman