espressif/esp-matter #613: Can I enable both Blufi and Matter to configure the network at the same time?

sxsx2yzyz, 2023-09-04:
In my project, using blufi+my app to configure the network has become very stable. Now I want to integrate matter into my project. When users do not want to use my own app but use app home or Alexa, they can also configure the network through matter. Is this possible?
And when users use Blufi+our app to configure the network and connect their devices to WiFi, they can also use app home, Alexa+"ip commissioning" to add devices and control them.

dhrishi, 2023-09-04:
>When users do not want to use my own app but use apple home or Alexa, they can also configure the network through matter. Is this possible?

Yes, exactly. That is possible. Matter setup, called as commissioning, involves setting up the Wi-Fi network on the device too.

>And when users use Blufi+our app to configure the network and connect their devices to WiFi, they can also use apple home, Alexa+" network pairing" to add devices and control them.

Yes, this should be possible too.

sxsx2yzyz, 2023-09-05:
This is my initialization process：
```
void app_main()
{
    some_task_init();
    ble_init();
    wifi_init();
    httpServer_init();
    ssdp_init();
    matter_init();
}
```
```
void matter_init()
{
    esp_err_t err = ESP_OK;
    /* Create a Matter node and add the mandatory Root Node device type on endpoint 0 */
    node::config_t node_config;
    node_t *node = node::create(&node_config, app_attribute_update_cb, app_identification_cb);

    extended_color_light::config_t light_config;
    light_config.on_off.on_off = false;
    light_config.on_off.lighting.start_up_on_off = nullptr;
    light_config.level_control.current_level = 64;
    light_config.level_control.lighting.start_up_current_level = 64;
    light_config.color_control.color_mode = EMBER_ZCL_COLOR_MODE_COLOR_TEMPERATURE;
    light_config.color_control.enhanced_color_mode = EMBER_ZCL_ENHANCED_COLOR_MODE_COLOR_TEMPERATURE;
    light_config.color_control.color_temperature.startup_color_temperature_mireds = nullptr;
    endpoint_t *endpoint = extended_color_light::create(node, &light_config, ENDPOINT_FLAG_NONE, NULL);

    /* These node and endpoint handles can be used to create/add other endpoints and clusters. */
    if (!node || !endpoint) {
        ESP_LOGE(TAG, "Matter node creation failed");
    }

    light_endpoint_id = endpoint::get_id(endpoint);

    /* Matter start */
    err = esp_matter::start(app_event_cb);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "Matter start failed: %d", err);
    }
}
```
In menuconfig,
```
CHIP Device Layer ->WiFi Station Options disabled
CHIP Device Layer ->WiFi AP Options disabled
CHIP Device Layer ->BLE Options ->CHIP over BLE Support enabled
CHIP Device Layer ->BLE Options ->Use Ble only for commissioning enabled
Set the Bluetooth Host to Bluedroid - Dual mode in menuconfig's Bluetooth
```
I found that after setting this up, I cannot use my own application to connect to the device through Bluetooth. I can use the ESP Matter application of iOS to connect, but it returns an error of 80

[image]

This is the log output:

```
W (00:00:08.633) esp_matter_core: Command 0x00000047 on cluster 0x00000300 already exists. Not creating again.
I (1930) chip[DL]: NVS set: chip-counters/reboot-count = 6 (0x6)
I (1937) chip[DL]: Real time clock set to 946684800 (0100/00/01 00:00:00 UTC)
I (1939) chip[SVR]: Initializing subscription resumption storage...
I (1942) chip[BLE]: GATT Event: 0 (if 4)
I (1945) chip[SVR]: Server initializing...
I (1953) chip[TS]: Last Known Good Time: 2023-09-05T10:02:41
I (1958) chip[DMG]: AccessControl: initializing
I (1963) chip[DMG]: Examples::AccessControlDelegate::Init
I (1970) chip[DMG]: AccessControl: setting
I (1974) chip[DMG]: DefaultAclStorage: initializing
[... 202 lines cut ...]
E (22757) chip[SVR]: Commissioning failed (attempt 1): 32
I (22764) chip[BLE]: Releasing end point's BLE connection back to application.
I (22772) chip[BLE]: Got notification regarding chip connection closure
I (22780) chip[DIS]: Updating services using commissioning mode 1
I (22786) chip[DIS]: CHIP minimal mDNS started advertising.
E (22793) chip[DIS]: Failed to get primary mac address of device. Generating a random one.
I (22801) chip[DIS]: Advertise commission parameter vendorID=65521 productID=32768 discriminator=3840/15 cm=1
I (22812) chip[DIS]: CHIP minimal mDNS configured as 'Commissionable node device'; instance name: 8FAE969F9372FA5D.
I (22826) chip[DIS]: mDNS service published: _matterc._udp
E (22829) chip[ZCL]: OpCreds: Got FailSafeTimerExpired
E (22834) chip[ZCL]: OpCreds: Proceeding to FailSafeCleanup on fail-safe expiry!
E (22843) chip[FP]: Reverting pending fabric data for fabric 0x1
I (22849) chip[TS]: Pending Last Known Good Time: 2023-09-05T10:02:41
I (22858) chip[TS]: Previous Last Known Good Time: 2023-09-05T10:02:41
I (22863) chip[TS]: Reverted Last Known Good Time to previous value
I (00:00:20.933) esp_matter_attribute: ********** W : Endpoint 0x0000's Cluster 0x00000030's Attribute 0x00000000 is 0 **********
I (00:00:20.945) matter: Commissioning failed, fail safe timer expired
I (22890) chip[FS]: Fail-safe cleanly disarmed
I (00:00:20.957) matter: Commissioning window opened
I (22907) chip[BLE]: GATT Event: 5 (if 4)
E (22907) chip[BLE]: no endpoint for BLE sent data ack
I (22986) chip[BLE]: GATT Event: 15 (if 3)
I (22986) chip[BLE]: GATT Event: 15 (if 4)
I (22987) chip[DL]: BLE GATT connection closed (con 0, reason 19)
```

sxsx2yzyz, 2023-09-06:
The problem may be that MATTER's ble initialization conflicts with my blufi initialization
If MATTER's BLE is not enabled and only my app+blufi is used to configure the network, everything is normal, my app is working properly, and Google Home can also complete network pairing.
What I want to ask is whether this solution can pass CSA testing and whether this product can be launched? (Users must first use my app for network configuration before they can use Google home, Alexa, or Apple home for network pairing)

dhrishi, 2023-09-06:
@sxsx2yzyz Currently, for such cases, you need to use the [external platform option](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#creating-the-external-platform-directory) and make your changes as required.

The BLE Mesh to Matter bridge example uses it. For reference, please check https://github.com/espressif/esp-matter/tree/main/examples/common/blemesh_platform/platform/ESP32_custom. It can then be included as https://github.com/espressif/esp-matter/blob/main/examples/blemesh_bridge/main/CMakeLists.txt

You can create your own platform and make changes as required.

sxsx2yzyz, 2023-09-06:
@dhrishi thank u 
But I still want to know if this solution can pass CSA testing and bring the product to market? (Users must first use my app for network configuration before they can use Google home, Alexa, or Apple home for network pairing)

dhrishi, 2023-09-06:
@sxsx2yzyz I think, it should be possible. You will anyway have to execute and pass all the (applicable) test cases for certification.

sxsx2yzyz, 2023-09-06:
thank u

sxsx2yzyz, 2023-09-07:
> @sxsx2yzyz Currently, for such cases, you need to use the [external platform [...]

Do you have more detailed documentation

dhrishi, 2023-09-07:
I've already provided the reference to the documentation above: https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#creating-the-external-platform-directory (sections 2.4.4.1 to 2.4.4.4)

sxsx2yzyz, 2023-09-07:
> I've already provided the reference to the documentation above: https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#creating-the-external-platform-directory (sections 2.4.4.1 to 2.4.4.4)

Thank you, I have already completed these. I want to know what these files are for and how to modify them to achieve the desired functionality.
[image]

sxsx2yzyz, 2023-09-07:
For example, I want to write the external platform folder myself from scratch, without using the files shown in the above picture, where should I start

dhrishi, 2023-09-07:
> I want to write the external platform folder myself from scratch, 

The external platform is a copy of platform support for ESP platform in the upstream connectedhomeip repository: https://github.com/project-chip/connectedhomeip/tree/master/src/platform/ESP32. It is not advisable to change everything there or write it from scratch. Instead you should only modify the parts relevant to you. If you want to make change in BLE, please check `nimble/BLEManagerImpl.cpp` and `ConnectivityManagerImpl_WiFi.cpp`.

sxsx2yzyz, 2023-09-07:
If I don't modify this section, how can I make MATTER initialize without initializing ble? And the subsequent process does not conflict with my blufi function
```
void app_main()
{
    blufi_init();
    wifi_init();
    matter_init();
}
```

sxsx2yzyz, 2023-09-07:
I have created a custom external platform directory:`myproject/platform/ESP32_ Custom/ `and successfully compiled in my project.
What you mean is that I can modify` myproject/platform/ESP32_ custom/bluedroid/BLEManagerImpl.cpp` or `myproject/platform/ESP32_custom` Are there any other files in the custom directory?

dhrishi, 2023-09-22:
@sxsx2yzyz The custom folder has all the files that are present in `connectedhomeip/src/platform/ESP32` folder. These files are symbolic links for all files in the ESP32 platform folder. You can modify the files you want, depending on the functionality you want to change.

closed 2023-09-22 by sxsx2yzyz