espressif/esp-matter #614: Can I disable cooling mode for theromostate device?

damian-kurek-wizzdev, 2023-09-04:
**Describe the question/query that you have**
Can  I disable the cooling mode for my thermostat device when I configure the endpoint? or do I need to return an error when someone wants to set it?

I found how to add this feature esp_err_t add(cluster_t *cluster, config_t *config);
But it seems to be added by default.

Where can I find documentation that describes what is expected effect when changing parameters via matter?

**Additional context**
...

PSONALl, 2023-09-05:
@damian-kurek-wizzdev Since the cooling feature is optional it is not added by default in the thermostat

damian-kurek-wizzdev, 2023-09-05:
Oh, it is the Google Home app. It displays cooling/cooling heating if you don't block it. Handling system settings like this:
`    ESP_LOGI(TAG, "New mode acquired %d", mode);
    if (mode == 4)
    {
        ESP_LOGI(TAG, "Acquired mode heating", mode);
    }
    else if (mode == 0)
    {
        ESP_LOGI(TAG, "Acquired mode off", mode);
    }
    else
    {
        ESP_LOGI(TAG, "Acquired mode unsupported %d", mode);
        return ESP_FAIL;
    }`

Disables cooling settings there

closed 2023-09-05 by damian-kurek-wizzdev