espressif/esp-matter #615: Node Id Question

ddd-Rex, 2023-09-05:
Hello, I am trying to use the CHIP-Tool undled with the esp-matter SDK to comission an ESP32-C3 controller running the light example also bundled with the SDK, I am able to succesfully monitor the cotroller as well as initiate the chip-tool, but the documentation mentions a "node_id" wich is seemingly randomly generated, however, I do not know how or where to find this randomly generated ID, so far I have tried looking at the output of my controller when monitoring it, but I can't find anything that matches what is shown in the documentation. 

Below is the aforementioned output with personal details redacted:

```
Executing action: monitor
Serial port /dev/ttyACM0
Connecting...
Detecting chip type... ESP32-C3
Running idf_monitor in directory /path/to/esp/esp-matter/examples/light
Executing "/path/to/.espressif/python_env/idf5.1_py3.10_env/bin/python /path/to/esp/esp-idf/tools/idf_monitor.py -p /dev/ttyACM0 -b 115200 --toolchain-prefix riscv32-esp-elf- --target esp32c3 --revision 3 --decode-panic backtrace /path/to/esp/esp-matter/examples/light/build/light.elf -m '/path/to/.espressif/python_env/idf5.1_py3.10_env/bin/python' '/path/to/esp/esp-idf/tools/idf.py'"...
--- idf_monitor on /dev/ttyACM0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4004c512
0x4004c512: uart_rx_one_char in ROM

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x1704
load:0x403cc710,len:0x968
load:0x403ce710,len:0x2f68
entry 0x403cc710
I (24) boot: ESP-IDF v5.1.1 2nd stage bootloader
I (24) boot: compile time Sep  5 2023 09:15:03
I (25) boot: chip revision: v0.3
I (27) boot.esp32c3: SPI Speed      : 80MHz
I (32) boot.esp32c3: SPI Mode       : DIO
I (37) boot.esp32c3: SPI Flash Size : 4MB
I (42) boot: Enabling RNG early entropy source...
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 esp_secure_cert  unknown          3f 06 0000d000 00002000
I (65) boot:  1 nvs              WiFi data        01 02 00010000 0000c000
I (73) boot:  2 nvs_keys         NVS keys         01 04 0001c000 00001000
I (80) boot:  3 otadata          OTA data         01 00 0001d000 00002000
I (88) boot:  4 phy_init         RF data          01 01 0001f000 00001000
I (95) boot:  5 ota_0            OTA app          00 10 00020000 001e0000
I (103) boot:  6 ota_1            OTA app          00 11 00200000 001e0000
I (110) boot:  7 fctry            WiFi data        01 02 003e0000 00006000
I (118) boot: End of partition table
I (122) esp_image: segment 0: paddr=00020020 vaddr=3c110020 size=395a0h (234912) map
I (168) esp_image: segment 1: paddr=000595c8 vaddr=3fc94c00 size=03464h ( 13412) load
I (171) esp_image: segment 2: paddr=0005ca34 vaddr=40380000 size=035e4h ( 13796) load
I (176) esp_image: segment 3: paddr=00060020 vaddr=42000020 size=10bbf4h (1096692) map
I (357) esp_image: segment 4: paddr=0016bc1c vaddr=403835e4 size=11444h ( 70724) load
I (371) esp_image: segment 5: paddr=0017d068 vaddr=50000010 size=00004h (     4) load
I (378) boot: Loaded app from partition at offset 0x20000
I (378) boot: Disabling RNG early entropy source...
I (391) cpu_start: Unicore app
I (392) cpu_start: Pro cpu up.
I (400) cpu_start: Pro cpu start user code
I (400) cpu_start: cpu freq: 160000000 Hz
I (400) cpu_start: Application information:
I (403) cpu_start: Project name:     light
I (408) cpu_start: App version:      1.0
I (412) cpu_start: Compile time:     Sep  5 2023 09:14:53
I (419) cpu_start: ELF file SHA256:  5b4d3b76d5a5e921...
I (425) cpu_start: ESP-IDF:          v5.1.1
I (429) cpu_start: Min chip rev:     v0.3
I (434) cpu_start: Max chip rev:     v0.99 
I (439) cpu_start: Chip rev:         v0.3
I (444) heap_init: Initializing. RAM available for dynamic allocation:
I (451) heap_init: At 3FCA96F0 len 00033020 (204 KiB): DRAM
I (457) heap_init: At 3FCDC710 len 00002950 (10 KiB): STACK/DRAM
I (464) heap_init: At 50000014 len 00001FD4 (7 KiB): RTCRAM
I (471) spi_flash: detected chip: generic
I (475) spi_flash: flash io: dio
[... 122 lines cut ...]
I (1758) led_driver_ws2812: led set r:0, g:0, b:0
I (1768) led_driver_ws2812: led set r:0, g:0, b:0
I (1768) led_driver_ws2812: led set r:0, g:0, b:0
> I (2288) main_task: Returned from app_main()
Writing to serial is timing out. Please make sure that your application supports an interactive console and that you have picked the correct console for serial communication.
I (31458) chip[DL]: Configuring CHIPoBLE advertising (interval 500 ms, connectable)
I (31458) chip[DL]: Device already advertising, stop active advertisement and restart
I (31468) NimBLE: GAP procedure initiated: stop advertising.

I (31468) NimBLE: GAP procedure initiated: advertise; 
I (31478) NimBLE: disc_mode=2
I (31478) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=800 adv_itvl_max=800
I (31488) NimBLE: 

I (901448) chip[SVR]: Closing pairing window
I (901448) chip[DIS]: Updating services using commissioning mode 0
I (901448) chip[DIS]: CHIP minimal mDNS started advertising.
I (901458) NimBLE: GAP procedure initiated: stop advertising.

I (901468) chip[DL]: CHIPoBLE advertising stopped
I (901478) app_main: Commissioning window closed
Writing to serial is timing out. Please make sure that your application supports an interactive console and that you have picked the correct console for serial communication.
```

Any help is appreciated,thanks

shubhamdp, 2023-09-05:
Pasting an example command to commission the device, `pairing ble-wifi 3 <ssid> <passphrase> 20202021 3840`.

You can find more details here: https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#commissioning

ddd-Rex, 2023-09-05:
Thanks for the reply, I have read the documentation, that's where my confusion stemmed from in the first place since I wasn't sure if "chosen at random" meant the node ID was being generated and displayed somewhere by my controller every time I restarted it. Going by your response I take it it can be pretty much anything I want as long as it adheres to the format 0xXXXX?

shubhamdp, 2023-09-05:
@ddd-Rex its just the hex representation, eg: 20 decimal can be 0x14.

Also, for more details about the node-id range, i'd suggest you to look at section `2.5.5. Node Identifier` from Matter specification.

shubhamdp, 2023-09-05:
@ddd-Rex Please close the issue if your query is resolved.

closed 2023-09-05 by ddd-Rex