espressif/esp-matter #617: chipDie chipDie chipDie

jonsmirl, 2023-09-05:
Using 5.1 IDF
commit f0437b945fe016bd9aa8281bca1095294785a5fe (HEAD, origin/release/v5.1)
Merge: 865b48039d bd21c215c7

~~~
I (30745) chip[DL]: BLE GAP connection terminated (con 1 reason 0x213)
I (41275) app_main: Free internal 123431 biggest block 31732 total 123431
I (61275) app_main: Free internal 123431 biggest block 31732 total 123431
I (73355) chip[FS]: Fail-safe timer expired
E (73355) chip[SVR]: Failsafe timer expired
I (73355) chip[SC]: SecureSession[0x3fca5690, LSID:19989]: State change 'kActive' --> 'kPendingEviction'
E (73355) chip[SVR]: Commissioning failed (attempt 1): 32
I (73365) chip[DIS]: Updating services using commissioning mode 1
I (73365) chip[DIS]: Advertise commission parameter vendorID=65522 productID=32769 discriminator=3840/15 cm=1
E (73365) chip[ZCL]: OpCreds: Got FailSafeTimerExpired
E (73365) chip[ZCL]: OpCreds: Proceeding to FailSafeCleanup on fail-safe expiry!
I (73365) chip[TS]: Pending Last Known Good Time: 2023-09-05T16:22:20
I (73375) chip[TS]: Previous Last Known Good Time: 2023-09-05T16:22:20
I (73375) chip[TS]: Reverted Last Known Good Time to previous value
I (73375) esp_matter_attribute: ********** W : Endpoint 0x0000's Cluster 0x00000030's Attribute 0x00000000 is 0 **********
I (73375) ep: Commissioning failed, fail safe timer expired
I (73375) chip[FS]: Fail-safe cleanly disarmed
I (73375) chip[DL]: Configuring CHIPoBLE advertising (interval 25 ms, connectable)
I (73375) NimBLE: GAP procedure initiated: advertise; 
I (73375) NimBLE: disc_mode=2
I (73375) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40
I (73375) NimBLE: 

I (73375) chip[DL]: CHIPoBLE advertising started
I (73385) ep: Commissioning window openI (73495) chip[DL]: BLE GAP connection established (con 1)
I (73495) chip[DL]: CHIPoBLE advertising stopped
I (74045) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (74045) chip[BLE]: local and remote recv window sizes = 5
I (74045) chip[BLE]: selected BTP version 4
I (74045) chip[BLE]: using BTP fragment sizes rx 244 / tx 244.
I (75745) chip[DL]: BLE GAP connection terminated (con 1 reason 0x213)
E (75745) chip[DL]: Chip stack locking error at 'src/system/SystemLayerImplFreeRTOS.cpp:101'. Code is unsafe/racy
E (75745) chip[-]: chipDie chipDie chipDie

abort() was called at PC 0x420754df on core 0
0x420754df: chip::Platform::Internal::AssertChipStackLockedByCurrentThread(char const*, int) at /home/jonsmirl/lowpan/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/support/CodeUtils.h:508
 (inlined by) chipDie at /home/jonsmirl/lowpan/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/support/CodeUtils.h:518
 (inlined by) chip::Platform::Internal::AssertChipStackLockedByCurrentThread(char const*, int) at /home/jonsmirl/lowpan/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/LockTracker.cpp:36

Backtrace: 0x40375c5a:0x3fcdf3a0 0x403816c9:0x3fcdf3c0 0x40387486:0x3fcdf3e0 0x420754df:0x3fcdf450 0x420760cd:0x3fcdf470 0x42093ad9:0x3fcdf490 0x42093d90:0x3fcdf4b0 0x4206f895:0x3fcdf4d0 0x4208302b:0x3fcdf4f0 0x420831af:0x3fcdf550 0x42056f99:0x3fcdf570 0x42057cb1:0x3fcdf590 0x42057cd8:0x3fcdf5f0 0x420546ee:0x3fcdf650 0x420548bf:0x3fcdf670 0x42053f39:0x3fcdf690 0x4037efe7:0x3fcdf6b0 0x40377de1:0x3fcdf6d0 0x420828fe:0x3fcdf6f0 0x40383c29:0x3fcdf710
0x40375c5a: panic_abort at /home/jonsmirl/esp/esp-idf/components/esp_system/panic.c:452

0x403816c9: esp_system_abort at /home/jonsmirl/esp/esp-idf/components/esp_system/port/esp_system_chip.c:84

0x40387486: abort at /home/jonsmirl/esp/esp-idf/components/newlib/abort.c:38

0x420754df: chip::Platform::Internal::AssertChipStackLockedByCurrentThread(char const*, int) at /home/jonsmirl/lowpan/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/support/CodeUtils.h:508
 (inlined by) chipDie at /home/jonsmirl/lowpan/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/support/CodeUtils.h:518
 (inlined by) chip::Platform::Internal::AssertChipStackLockedByCurrentThread(char const*, int) at /home/jonsmirl/lowpan/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/LockTracker.cpp:36

0x420760cd: chip::System::LayerImplFreeRTOS::CancelTimer(void (*)(chip::System::Layer*, void*), void*) at /home/jonsmirl/lowpan/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/system/SystemLayerImplFreeRTOS.cpp:101

0x42093ad9: chip::Ble::BLEEndPoint::StopReceiveConnectionTimer() at /home/jonsmirl/lowpan/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/ble/BLEEndPoint.cpp:1487

0x42093d90: chip::Ble::BLEEndPoint::DoClose(unsigned char, chip::ChipError) at /home/jonsmirl/lowpan/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/ble/BLEEndPoint.cpp:331

0x4206f895: chip::Ble::BleLayer::HandleConnectionError(unsigned short, chip::ChipError) at /home/jonsmirl/lowpan/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/ble/BleLayer.cpp:738

0x4208302b: chip::DeviceLayer::Internal::BLEManagerImpl::HandleGAPDisconnect(ble_gap_event*) at /home/jonsmirl/lowpan/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/nimble/BLEManagerImpl.cpp:1287

0x420831af: chip::DeviceLayer::Internal::BLEManagerImpl::ble_svr_gap_event(ble_gap_event*, void*) at /home/jonsmirl/lowpan/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/nimble/BLEManagerImpl.cpp:1370

0x42056f99: ble_gap_call_event_cb at /home/jonsmirl/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:773

0x42057cb1: ble_gap_conn_broken at /home/jonsmirl/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:1331

0x42057cd8: ble_gap_rx_disconn_complete at /home/jonsmirl/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:1359 (discriminator 4)

0x420546ee: ble_hs_hci_evt_disconn_complete at /home/jonsmirl/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:281
 (inlined by) ble_hs_hci_evt_disconn_complete at /home/jonsmirl/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:204

0x420548bf: ble_hs_hci_evt_process at /home/jonsmirl/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:1023

0x42053f39: ble_hs_event_rx_hci_ev at /home/jonsmirl/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:526

0x4037efe7: npl_freertos_event_run at /home/jonsmirl/esp/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/src/npl_os_freertos.c:451

0x40377de1: ble_npl_event_run at /home/jonsmirl/esp/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:185
 (inlined by) nimble_port_run at /home/jonsmirl/esp/esp-idf/components/bt/host/nimble/nimble/porting/nimble/src/nimble_port.c:256

0x420828fe: chip::DeviceLayer::Internal::BLEManagerImpl::bleprph_host_task(void*) at /home/jonsmirl/lowpan/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/nimble/BLEManagerImpl.cpp:878

0x40383c29: vPortTaskWrapper at /home/jonsmirl/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

~~~

aggaddam, 2023-09-06:
I too observed this issue in esp-idf v5.1.1 and esp-matter commit "68e99de70266674ce172459136960bbb5c58dc0e". But this issue does not occur very frequently.

shubhamdp, 2023-09-06:
Looking into it, it looks like nimble is trying to execute the CHIP code. Will raise a fix.

shubhamdp, 2023-09-12:
I have raised a PR, https://github.com/project-chip/connectedhomeip/pull/29180. PTAL @jonsmirl @aggaddam

aggaddam, 2023-10-13:
@shubhamdp Is this issue fixed in the latest esp-matter SDK main branch ?

jonsmirl, 2023-11-06:
I am up to date and just hit this again in a different location
~~~
E (11905) chip[DL]: Chip stack locking error at 'src/system/SystemLayerImplFreeRTOS.cpp:101'. Code is unsafe/racy
E (11915) chip[-]: chipDie chipDie chipDie

abort() was called at PC 0x42046aa3 on core 0
0x42046aa3: chip::Platform::Internal::AssertChipStackLockedByCurrentThread(char const*, int) at /home/jonsmirl/lowpan/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/support/CodeUtils.h:508
 (inlined by) chipDie at /home/jonsmirl/lowpan/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/support/CodeUtils.h:518
 (inlined by) chip::Platform::Internal::AssertChipStackLockedByCurrentThread(char const*, int) at /home/jonsmirl/lowpan/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/LockTracker.cpp:36

Backtrace: 0x40375966:0x3fcdd1b0 0x40380d69:0x3fcdd1d0 0x40387252:0x3fcdd1f0 0x42046aa3:0x3fcdd260 0x42047679:0x3fcdd280 0x42063719:0x3fcdd2a0 0x420639c0:0x3fcdd2c0 0x4203d58d:0x3fcdd2e0 0x420523bf:0x3fcdd300 0x42052543:0x3fcdd360 0x4206be35:0x3fcdd380 0x4206d391:0x3fcdd3a0 0x4206d3be:0x3fcdd400 0x42068e32:0x3fcdd460 0x4206919d:0x3fcdd480 0x42068621:0x3fcdd4a0 0x4037e463:0x3fcdd4c0 0x40377876:0x3fcdd4e0 0x42051cc6:0x3fcdd500 0x403834ca:0x3fcdd520
0x40375966: panic_abort at /home/jonsmirl/esp/esp-idf/components/esp_system/panic.c:452

0x40380d69: esp_system_abort at /home/jonsmirl/esp/esp-idf/components/esp_system/port/esp_system_chip.c:84

0x40387252: abort at /home/jonsmirl/esp/esp-idf/components/newlib/abort.c:38

0x42046aa3: chip::Platform::Internal::AssertChipStackLockedByCurrentThread(char const*, int) at /home/jonsmirl/lowpan/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/support/CodeUtils.h:508
 (inlined by) chipDie at /home/jonsmirl/lowpan/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/support/CodeUtils.h:518
 (inlined by) chip::Platform::Internal::AssertChipStackLockedByCurrentThread(char const*, int) at /home/jonsmirl/lowpan/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/LockTracker.cpp:36

0x42047679: chip::System::LayerImplFreeRTOS::CancelTimer(void (*)(chip::System::Layer*, void*), void*) at /home/jonsmirl/lowpan/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/system/SystemLayerImplFreeRTOS.cpp:101

0x42063719: chip::Ble::BLEEndPoint::StopReceiveConnectionTimer() at /home/jonsmirl/lowpan/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/ble/BLEEndPoint.cpp:1487

0x420639c0: chip::Ble::BLEEndPoint::DoClose(unsigned char, chip::ChipError) at /home/jonsmirl/lowpan/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/ble/BLEEndPoint.cpp:331

0x4203d58d: chip::Ble::BleLayer::HandleConnectionError(unsigned short, chip::ChipError) at /home/jonsmirl/lowpan/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/ble/BleLayer.cpp:738

0x420523bf: chip::DeviceLayer::Internal::BLEManagerImpl::HandleGAPDisconnect(ble_gap_event*) at /home/jonsmirl/aosp/esp-matter/examples/common/blemesh_platform/platform/ESP32_custom/nimble/BLEManagerImpl.cpp:997

0x42052543: chip::DeviceLayer::Internal::BLEManagerImpl::ble_svr_gap_event(ble_gap_event*, void*) at /home/jonsmirl/aosp/esp-matter/examples/common/blemesh_platform/platform/ESP32_custom/nimble/BLEManagerImpl.cpp:1080

0x4206be35: ble_gap_call_event_cb at /home/jonsmirl/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:773

0x4206d391: ble_gap_conn_broken at /home/jonsmirl/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:1344

0x4206d3be: ble_gap_rx_disconn_complete at /home/jonsmirl/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:1372 (discriminator 4)

0x42068e32: ble_hs_hci_evt_disconn_complete at /home/jonsmirl/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:294

0x4206919d: ble_hs_hci_evt_process at /home/jonsmirl/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:1051

0x42068621: ble_hs_event_rx_hci_ev at /home/jonsmirl/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:526

0x4037e463: npl_freertos_event_run at /home/jonsmirl/esp/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/src/npl_os_freertos.c:451

0x40377876: ble_npl_event_run at /home/jonsmirl/esp/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:185
 (inlined by) nimble_port_run at /home/jonsmirl/esp/esp-idf/components/bt/host/nimble/nimble/porting/nimble/src/nimble_port.c:248

0x42051cc6: chip::DeviceLayer::Internal::BLEManagerImpl::bleprph_host_task(void*) at /home/jonsmirl/aosp/esp-matter/examples/common/blemesh_platform/platform/ESP32_custom/nimble/BLEManagerImpl.cpp:666

0x403834ca: vPortTaskWrapper at /home/jonsmirl/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

~~~

shubhamdp, 2023-11-08:
@jonsmirl https://github.com/project-chip/connectedhomeip/pull/29180 just got merged and submodule is not yet updated, so please wait till the submodule is updated.

dhrishi, 2024-01-31:
Closing this as the connectedhomeip submodule in esp-matter is now updated

closed 2024-01-31 by dhrishi