espressif/esp-matter #619: Usage for chip lock/unlock in application task

aggaddam, 2023-09-06:
**I wanted to understand the correct method of using locking and unlocking while accessing matter resources from the application thread.*

In FAQ, it was suggested to follow the below method 

```
lock::chip_stack_lock(portMAX_DELAY);
... // eg: access Matter attribute, open/close commissioning window.
lock::chip_stack_unlock();
```

But in esp-matter SDK, it was implemented like below

```
    /* Take lock if not already taken */
    lock::status_t lock_status = lock::chip_stack_lock(portMAX_DELAY);
    if (lock_status == lock::FAILED) {
        ESP_LOGE(TAG, "Could not get task context");
        return ESP_FAIL;
    }
    ... // eg: access Matter attribute, open/close commissioning window.

    if (lock_status == lock::SUCCESS) {
        lock::chip_stack_unlock();
    }
```
    
Also, I would like to know, do i need to use the locking  and unlocking while accessing matter resources in the matter app callback i.e. app_event_cb 

**Additional context**
...

jonsmirl, 2023-09-14:
Those two are the same thing. The second one is checking for errors, the first one isn't.

All of the main CHIP APIs have debug code so that if you call one which needs locking without holding the lock it will print on the console. I am not certain if all of the minor APIs have this code, if you find one missing the debug locking check, raise an issue. I only have the main APIs locked in my code and have not encountered any problems using the minor ones without locking.

aggaddam, 2023-09-15:
@jonsmirl I am using approach 2 right now. Sometimes I do see random crashes in the chip SDK.

jonsmirl, 2023-09-15:
If you are using any of the emberAF...callback() functions, those need to be locked.

shubhamdp, 2023-09-15:
If you are using CHIP SDK API and you are calling them from thread other than chip thread then they needs to be either called by acquiring/releasing locks or dispatching them to chip queue (using StartTimer, ScheduleWork).

I can't say which one needs locking specifically, because, assert are not placed in every part of code. They are placed where it is really required the synchronized access to chip sdk resources.

On the other hand if lock tracking is in place because we want thread safe access to chip sdk resources. If you want to get rid of lock tracking disable `CHIP_STACK_LOCK_TRACKING_ERROR_FATAL`. This option is not exposed using menuconfig. But I'd suggest strongly against it during development.

dhrishi, 2023-09-30:
Closing this as the question is answered

closed 2023-09-30 by dhrishi