espressif/esp-matter #620: Failed to use chip-tool Bluetooth network distribution with error CHIP Error 0x000000AC

MaplestoryAlen, 2023-09-07:
The (./chip tool pairing onnetwork) command can be used for network configuration, but Bluetooth cannot be used for network configuration

**Describe the bug**
 chip-tool log: Please refer to the following document
 [log.txt](https://github.com/espressif/esp-matter/files/12544358/log.txt)

**Environment**
- ESP-Matter Commit Id: 5b4f800
- ESP-IDF Commit Id:ESP-IDF v5.1.1
- SoC :ESP32-S3
- Device Logs (Please attach the log file):
- Host Machine OS: Ubuntu20.04
- Commissioner app and versions if present:
- Commissioner's logs if present:

**Any additional details**
...

MaplestoryAlen, 2023-09-08:
ESP32-H2
 chip-tool log
 CHIP:TOO: Run command failure: src/controller/AutoCommissioner.cpp:218: CHIP Error 0x0000002F: Invalid argument

 ESP32-H2 monitor log:

I (1775) app_main: Commissioning window opened
I (1775) led_driver_ws2812: led set r:63, g:51, b:40
I (1855) main_task: Returned from app_main()
> I (10935) chip[DL]: BLE GAP connection established (con 1)
I (10945) chip[DL]: CHIPoBLE advertising stopped
I (12265) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (12265) chip[BLE]: local and remote recv window sizes = 5
I (12275) chip[BLE]: selected BTP version 4
I (12285) chip[BLE]: using BTP fragment sizes rx 244 / tx 244.
I (12365) chip[DL]: Write request/command received for CHIPoBLE TX CCCD characteristic (con 1 ) indicate = 1
I (12365) chip[DL]: CHIPoBLE subscribe received
I (12365) NimBLE: GATT procedure initiated: indicate; 
I (12385) NimBLE: att_handle=14

I (12465) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
I (12465) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (12485) chip[EM]: >>> [E:9410r S:0 M:98748697] (U) Msg RX from 0:549AB147F7B35FA9 [0000] --- Type 0000:20 (SecureChannel:PBKDFParamRequest)
I (12495) chip[EM]: <<< [E:9410r S:0 M:107189204] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:21 (SecureChannel:PBKDFParamResponse)
I (12505) chip[IN]: (U) Sending msg 107189204 to IP address 'BLE'
I (12515) NimBLE: GATT procedure initiated: indicate; 
I (12515) NimBLE: att_handle=14

I (12535) chip[SVR]: Commissioning session establishment step started
I (12615) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
I (12615) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (12635) chip[EM]: >>> [E:9410r S:0 M:98748698] (U) Msg RX from 0:549AB147F7B35FA9 [0000] --- Type 0000:22 (SecureChannel:PASE_Pake1)
I (12745) chip[EM]: <<< [E:9410r S:0 M:107189205] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:23 (SecureChannel:PASE_Pake2)
I (12745) chip[IN]: (U) Sending msg 107189205 to IP address 'BLE'
I (12755) NimBLE: GATT procedure initiated: indicate; 
I (12755) NimBLE: att_handle=14

I (12865) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
I (12865) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (12885) chip[EM]: >>> [E:9410r S:0 M:98748699] (U) Msg RX from 0:549AB147F7B35FA9 [0000] --- Type 0000:24 (SecureChannel:PASE_Pake3)
I (12895) chip[EM]: <<< [E:9410r S:0 M:107189206] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:40 (SecureChannel:StatusReport)
I (12905) chip[IN]: (U) Sending msg 107189206 to IP address 'BLE'
I (12915) NimBLE: GATT procedure initiated: indicate; 
I (12915) NimBLE: att_handle=14

I (12925) chip[SC]: SecureSession[0x40815278, LSID:63917]: State change 'kEstablishing' --> 'kActive'
I (12935) chip[SVR]: Commissioning completed session establishment step
I (12935) chip[DIS]: Updating services using commissioning mode 0
E (12945) chip[DIS]: Failed to remove advertised services: 3
I (12945) chip[DIS]: Advertise commission parameter vendorID=65521 productID=32768 discriminator=3840/15 cm=0
E (12965) chip[DIS]: Failed to advertise extended commissionable node: 3
E (12975) chip[DIS]: Failed to finalize service update: 3
I (12975) chip[SVR]: Device completed Rendezvous process
I (12975) app_main: Commissioning session started
I (12995) app_main: Commissioning window closed
I (13015) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
I (13025) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (13035) chip[EM]: >>> [E:9411r S:63917 M:31816379] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:02 (IM:ReadRequest)
I (13045) chip[DMG]: Registered a ReadHandler that will schedule a report between system Timestamp: 13046 and system Timestamp 13046.
I (13055) esp_matter_attribute: ********** R : Endpoint 0x0000's Cluster 0x00000030's Attribute 0x00000000 is 0 **********
I (13075) chip[EM]: <<< [E:9411r S:63917 M:214414024] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0001:05 (IM:ReportData)
I (13075) chip[IN]: (S) Sending msg 214414024 on secure session with LSID: 63917
I (13095) NimBLE: GATT procedure initiated: indicate; 
I (13095) NimBLE: att_handle=14

I (13165) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
I (13165) NimBLE: GATT procedure initiated: indicate; 
I (13185) NimBLE: att_handle=14

I (13265) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
[... 169 lines cut ...]
I (16105) esp_matter_attribute: ********** W : Endpoint 0x0000's Cluster 0x00000030's Attribute 0x00000000 is 0 **********
I (16115) app_main: Commissioning failed, fail safe timer expired
I (16125) chip[FS]: Fail-safe cleanly disarmed
I (16135) chip[DL]: Configuring CHIPoBLE advertising (interval 25 ms, connectable)
I (16145) NimBLE: GAP procedure initiated: advertise; 
I (16145) NimBLE: disc_mode=2
I (16145) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40
I (16155) NimBLE: 

I (16165) chip[DL]: CHIPoBLE advertising started
I (16165) app_main: Commissioning window opened
I (16175) app_main: Fabric will be removed
E (16175) chip[BLE]: no endpoint for BLE sent data ack
E (16185) chip[BLE]: no endpoint for unsub recvd
[... cut ...]
I (18665) chip[DL]: BLE GAP connection terminated (con 1 reason 0x213)
I (18665) chip[DL]: Configuring CHIPoBLE advertising (interval 25 ms, connectable)
I (18675) chip[DL]: Device already advertising, stop active advertisement and restart
I (18685) NimBLE: GAP procedure initiated: stop advertising.

I (18695) NimBLE: GAP procedure initiated: advertise; 
I (18695) NimBLE: disc_mode=2
I (18695) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40
I (18715) NimBLE:

VaishaliAvhale, 2023-09-13:
Please provide the complete device and chip-tool logs in a file format.

MaplestoryAlen, 2023-09-14:
> Please provide the complete device and chip-tool logs in a file format.
Please review
[log.txt](https://github.com/espressif/esp-matter/files/12602913/log.txt)

shubhamdp, 2023-09-15:
@MaplestoryAlen `log.txt` shows that device failed to connect to wifi network.

In first comment you said `SoC :ESP32-S3` and in subsequent comment you said you are commissioning H2. Can you confirm which SoC you are using.

H2 is a 802.15.4 device and `ble-wifi` won't work on that. You would need to use `ble-thread` command and you must need a border router for commissioning the h2.

MaplestoryAlen, 2023-09-18:
> @MaplestoryAlen `log.txt` shows that device failed to connect to wifi network.
>
> In first comment you said `SoC :ESP32-S3` and in subsequent comment you said you are commissioning H2. Can you confirm [...]

I have also set up the border router and it can work normally. However, H2 and S3 cannot access the network using Bluetooth, and the hub uses Apple audio

S-A-M-J, 2024-03-03:
I have the same issue on an esp32-c3. 
ESP-IDF: 5.2
When using an amazon alexa device to discover the device, the device is found but Alexa tells me to enter wifi creds in a loop while never fully connecting. This is despite the chip being provisioned already. The code i'm using is the light example from esp matter. Logs below
If i disable chip over ble the device is found by alexa but alexa is unable to connect.
[log1.txt](https://github.com/espressif/esp-matter/files/14474880/log1.txt)

closed 2024-08-12 by MaplestoryAlen