espressif/esp-matter #621: How to determine whether an interface is a complex interface or a simple interface

maenkai, 2023-09-08:
**Describe the question/query that you have**
1. I found that the contoller source code only supports a few simple instructions, and no other instructions are related to complex interfaces here?
2. I want to add a new command interface by myself, but I can't determine whether the interface belongs to a simple interface or a complex interface

maenkai, 2023-09-08:
I need to determine whether the interface is uncomplicated and thus does not affect software component authentication

wqx6, 2023-09-11:
> no other instructions are related to complex interfaces here?

Could you clarify what you mean by a 'complex interface'? Perhaps you could provide an example to illustrate the concept?

> I want to add a new command interface by myself, but I can't determine whether the interface belongs to a simple interface or a complex interface.

We will provide interfaces that you can used to set the handlers for other commands that are not supported now later.

maenkai, 2023-09-11:
> Could you clarify what you mean by a 'complex interface'? Perhaps you could provide an example to illustrate the concept?
>
>
> We will provide interfaces that you can used to set the handlers for other [...]

1. In the software component certification specification, in the Type 2 UIC certification, if a complex interface needs to pass the ATL certification, the csa's official software component certification process also needs to inform whether it is a complex interface
2. Whether the controller can provide a general method, rather than adding new source code every time you add support for one instruction.

closed 2023-10-11 by maenkai