espressif/esp-matter #622: Upon matter factory reset, fails to clear wifi credentials from nvs

aggaddam, 2023-09-08:
**Describe the bug**
Upon issuing a factory reset after matter commissioning, the device fails to clear wifi credentials from nvs. There is an error occurred while restoring Wi-Fi credentials.  Check out the logs below for reference

**Environment**
- ESP-Matter Commit Id:  68e99de70266674ce172459136960bbb5c58dc0e
- ESP-IDF Commit Id: v5.1.1 Stable Release
- SoC (eg: ESP32 or ESP32-C3): ESP32-C3
- Device Logs (Please attach the log file):
```
I (51111) chip[DL]: Performing factory reset
D (51122) nvs: nvs_open_from_partition chip-config 1
D (51122) nvs: nvs_erase_all

D (51112) wifi:connect status 5 -> 0
D (51134) wifi:Start wifi disconnect
I (51134) wifi:state: run -> init (0)
D (51137) wifi:connect status 0 -> 6
D (51137) wifi:stop beacon/connect timer, send diassoc(8)
D (51148) wifi:sta leave
D (51141) nvs: nvs_close 181
D (51149) nvs: nvs_open_from_partition chip-counters 1
D (51160) nvs: nvs_erase_all

I (51149) wifi:pm stop, total sleep time: 20613992 us / 25972561 us

D (51166) nvs: nvs_close 182
E (51172) chip[DL]: esp_wifi_restore() failed: UNKNOWN ERROR
D (51173) nvs: nvs_open_from_partition CHIP_KVS 1
D (51184) nvs: nvs_erase_all

D (51173) wifi:stop CSA timer
I (51185) wifi:<ba-del>idx:0, tid:0
D (51185) wifi:remove b4:75:0e:bb:00:6e from rc list
I (51197) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
D (51198) wifi:filter: set rx policy=8
D (51196) nvs: nvs_close 183
I (51209) chip[DL]: System restarting
D (51210) wifi:Send disconnect event, reasonI (302) cpu_start: Unicore app
```

- Host Machine OS: Ubuntu
- Commissioner app and versions if present: chip-tool
- Commissioner's logs if present: NA

**Any additional details**
...

shripad621git, 2023-09-11:
@aggaddam I have tried calling a ```factory_reset``` from the code on the ```commissioning complete``` event with your specified esp-matter sdk commit id and idf version.It works fine on my side.Can you please give it a try once again? It works fine on the latest main as well.

aggaddam, 2023-09-14:
@shripad621git  I took the latest changes from main (Yesterday's commit). I am still seeing the issue when I perform a factory reset after adding the device to the fabric

```
D (147921) nvs: nvs_open_from_partition node 1
D (147921) nvs: nvs_erase_all

D (147927) nvs: nvs_open_from_partition endpoint_0 1
D (147928) nvs: nvs_erase_all

D (147933) nvs: nvs_open_from_partition endpoint_1 1
D (147934) nvs: nvs_erase_all

I (147950) esp_matter_core: Erasing attribute data completed
I (147951) chip[DL]: Performing factory reset
D (147952) nvs: nvs_open_from_partition chip-config 1
D (147962) nvs: nvs_erase_all

D (147952) wifi:connect status 5 -> 0
D (147964) wifi:Start wifi disconnect
I (147975) wifi:state: run -> init (0)
D (147975) wifi:connect status 0 -> 6
D (147976) wifi:stop beacon/connect timer, send diassoc(8)
D (147970) nvs: nvs_close 168
D (147987) nvs: nvs_open_from_partition chip-counters 1
D (147988) wifi:sta leave
I (147998) wifi:pm stop, total sleep time: 27210069 us / 36830234 us

D (147999) wifi:stop CSA timer
I (147999) wifi:<ba-del>idx:0, tid:0
D (148010) wifi:remove a8:88:1f:1c:2f:ab from rc list
I (148011) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
D (148022) wifi:filter: set rx policy=8
D (148022) wifi:Send disconnect event, reason=8, AP number=0
D (148023) wifi:connect status 6 -> 0
D (148033) wifi:filter: set rx policy=8D (147988) nvs: nvs_erase_all

D (148035) wifi:filter: set rx policy=2
D (148035) wifi:clear scan ap list
D (148046) wifi:clear blacklist
D (148046) wifi:clear rc list
D (148040) nvs: nvs_close 169
E (148048) chip[DL]: esp_wifi_restore() failed: UNKNOWN ERROR
D (148059) nvs: nvs_open_from_partition CHIP_KVS 1
D (148060) nvs: nvs_erase_all

D (148023) event: running post WIFI_EVENT:5 with handler 0x42099e0e and context 0x3fcb5658 on loop 0x3fcb2988
D (148087) esp_netif_handlers: esp_netif action disconnected with netif0x3fcb5364 from event_id=5
D (148099) esp_netif_lwip: check: remote, if=0x3fcb5364 fn=0x42096f44

D (148110) esp_netif_lwip: esp_netif_down_api esp_netif:0x3fcb5364
I (148098) wifi:flush txq
I (148111) wifi:stop sw txq
D (148111) wifi:filter: set rx policy=0
I (148122) wifi:lmac stop hw txq
D (148087) nvs: nvs_close 170
```

aggaddam, 2023-09-14:
Here is the exact error code while restoring wifi setting. I'm calling the factory reset function from the application thread not from matte app callback

[image]

shripad621git, 2023-09-14:
@aggaddam, can you please share your patch of changes?

shripad621git, 2023-09-15:
@[aggaddam](https://github.com/aggaddam) can you try the two patches provided patch and check if it solves your issue.Basically there are two issues.The nvs_commit and nvs_close calls were missing in factory_reset and erase_persistent_data api calls for which a fix has been raised.

The second issue might be with  esp_wifi_restore() api for which we added the esp_wifi_stop() api with a delay to ensure that wifi operation is stopped and it then calls the restore api.

[connectedhomeip.txt](https://github.com/espressif/esp-matter/files/12615461/connectedhomeip.txt)
[esp-matter.txt](https://github.com/espressif/esp-matter/files/12615458/esp-matter.txt)

Apply the first patch in connectedhomeip submodule of esp-matter and second one in esp-matter repository.
Please let me know if this helps you resolve the issue or not and your observations if any.

shripad621git, 2023-09-25:
@aggaddam, any update on this?

aggaddam, 2023-10-13:
@shripad621git  I applied patch still it is failing. Here is the log 

```
D (48998) nvs: nvs_open_from_partition node 1
D (48998) nvs: nvs_erase_all

D (49003) nvs: nvs_close 122
D (49004) nvs: nvs_open_from_partition endpoint_0 1
D (49015) nvs: nvs_erase_all

D (49020) nvs: nvs_close 123
D (49020) nvs: nvs_open_from_partition endpoint_1 1
D (49021) nvs: nvs_erase_all

D (49028) nvs: nvs_close 124
I (49029) esp_matter_core: Erasing attribute data completed
I (49030) chip[DL]: Performing factory reset
D (49040) nvs: nvs_open_from_partition chip-config 1
D (49041) nvs: nvs_erase_all

D (49030) wifi:connect status 5 -> 0
D (49053) wifi:Start wifi disconnect
I (49053) wifi:state: run -> init (0)
D (49055) wifi:connect status 0 -> 6
D (49065) wifi:stop beacon/connect timer, send diassoc(8)
D (49066) wifi:sta leave
I (49067) wifi:pm stop, total sleep time: 12648873 us / 16577663 us

D (49078) wifi:stop CSA timer
I (49078) wifi:<ba-del>idx:0, tid:0
D (49079) wifi:remove b4:75:0e:bb:00:6e from rc list
D (49057) nvs: nvs_close 125
D (49090) nvs: nvs_open_from_partition chip-counters 1
D (49090) nvs: nvs_erase_all

I (49091) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
D (49102) wifi:filter: set rx policy=8
D (49104) wifi:Send disconnect event, reason=8, AP number=0
D (49114) wifi:connect status 6 -> 0
D (49115) wifi:filter: set rx policy=8
D (49115) wifi:filter: set rx policy=2
D (49126) wifi:clear scan ap list
D (49126) wifi:clear blacklist
D (49127) wifi:clear rc list
D (49107) nvs: nvs_close 126
D (49116) event: running post WIFI_EVENT:5 with handler 0x4209a380 and context 0x3fcb519c on loop 0x3fcb24cc
D (49139) esp_netif_handlers: esp_netif action disconnected with netif0x3fcb4ea8 from event_id=5
D (49150) esp_netif_lwip: check: remote, if=0x3fcb4ea8 fn=0x420974b6

D (49161) esp_netif_lwip: esp_netif_down_api esp_netif:0x3fcb4ea8
D (49162) esp_netif_lwip: esp_netif_internal_dhcpc_cb lwip-netif:0x3fcb4f0c
D (49173) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3fcb4ea8
D (49174) esp_netif_lwip: if0x3fcb4ea8 start ip lost tmr: interval=120
D (49185) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3fcb4ea8
D (49196) esp_netif_lwip: if0x3fcb4ea8 start ip lost tmr: already started
D (49196) esp_netif_lwip: check: local, if=0x3fcb4ea8 fn=0x4209814e

D (49207) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3fcb4ea8
D (49208) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (49219) event: running post WIFI_EVENT:5 with handler 0x42117316 and context 0x3fcafd48 on loop 0x3fcb24cc
D (49230) event: running post WIFI_EVENT:3 with handler 0x4209a1fe and context 0x3fcb513c on loop 0x3fcb24cc
D (49241) esp_netif_handlers: esp_netif action stopped with netif0x3fcb4ea8 from event_id=3
D (49242) esp_netif_lwip: check: remote, if=0x3fcb4ea8 fn=0x420977a8

D (49253) esp_netif_lwip: call api in lwip: ret=0x5002, give sem
D (49264) event: running post WIFI_EVENT:3 with handler 0x42117316 and context 0x3fcafd48 on loop 0x3fcb24cc
E (50128) chip[DL]: esp_wifi_restore() failed: ESP_ERR_WIFI_STOP_STATE
D (50129) nvs: nvs_open_from_partition CHIP_KVS 1
D (50129) nvs: nvs_erase_all

D (50152) nvs: nvs_close 127
I (50152) chip[DL]: System restarting
I (311) cpu_start: Unicore app
I (311) cpu_start: Pro cpu up.
D (319) clk: RTC_SLOW_CLK calibration value: 3573350
I (325) cpu_start: Pro cpu start user code
I (325) cpu_start: cpu freq: 160000000 Hz
I (325) cpu_start: Application information:
I (328) cpu_start: Project name:     ac
I (332) cpu_start: App version:      1cb20eb-dirty
I (338) cpu_start: Compile time:     Oct  9 2023 19:09:24
I (344) cpu_start: ELF file SHA256:  eb39e1c520f40dde...
I (350) cpu_start: ESP-IDF:          4bfe99a4d-dirty
I (356) cpu_start: Min chip rev:     v0.3
I (360) cpu_start: Max chip rev:     v0.99 
I (365) cpu_start: Chip rev:         v0.3
```

Here are the changes I applied 

[image]

[image]

[image]

aggaddam, 2023-10-18:
@shripad621git 

Here are the scenarios i observed 

1) Factory reset works fine when it is initiated upon a fabric being removed and if 0 fabrics in the device from the callback. Here is the snapshot of logic

```
static void app_event_cb(const ChipDeviceEvent *event, intptr_t arg)
{
    switch (event->Type) {
    case chip::DeviceLayer::DeviceEventType::kInterfaceIpAddressChanged:
        ESP_LOGI(TAG, "Interface IP Address changed");
        break;

    case chip::DeviceLayer::DeviceEventType::kCommissioningComplete:
        ESP_LOGI(TAG, "Commissioning complete");
        break;

    case chip::DeviceLayer::DeviceEventType::kFailSafeTimerExpired:
        ESP_LOGI(TAG, "Commissioning failed, fail safe timer expired");
        break;

    case chip::DeviceLayer::DeviceEventType::kCommissioningSessionStarted:
        ESP_LOGI(TAG, "Commissioning session started");
        break;

    case chip::DeviceLayer::DeviceEventType::kCommissioningSessionStopped:
        ESP_LOGI(TAG, "Commissioning session stopped");
        break;

    case chip::DeviceLayer::DeviceEventType::kCommissioningWindowOpened:
        ESP_LOGI(TAG, "Commissioning window opened");
        break;

    case chip::DeviceLayer::DeviceEventType::kCommissioningWindowClosed:
        ESP_LOGI(TAG, "Commissioning window closed");
        break;

    case chip::DeviceLayer::DeviceEventType::kFabricRemoved:
        {
            ESP_LOGI(TAG, "Fabric removed successfully");
            if (chip::Server::GetInstance().GetFabricTable().FabricCount() == 0)
            {
                PlatformMgr().ScheduleWork(matter_helperfcn_factory_reset,
                                           reinterpret_cast<intptr_t>(nullptr));
            }
            break;
        }

    case chip::DeviceLayer::DeviceEventType::kFabricWillBeRemoved:
        ESP_LOGI(TAG, "Fabric will be removed");
        break;

    case chip::DeviceLayer::DeviceEventType::kFabricUpdated:
        ESP_LOGI(TAG, "Fabric is updated");
        break;

    case chip::DeviceLayer::DeviceEventType::kFabricCommitted:
        ESP_LOGI(TAG, "Fabric is committed");
        break;
    default:
        break;
    }
}
    
void matter_helperfcn_factory_reset(intptr_t context)
{
    esp_matter::factory_reset();
}    

```

2) Factory reset fails if initiated from any application task after successful matter commissioning. We want factory reset operation gracefully irrespective of chip stack state.

```  
void matter_helperfcn_factory_reset(intptr_t context)
{
    esp_matter::factory_reset();
}    

void app1_task(void)
{    
    // Calling after matter commission 
    PlatformMgr().ScheduleWork(matter_helperfcn_factory_reset,
                           reinterpret_cast<intptr_t>(nullptr));
}

```

shreyassheth1, 2023-10-19:
@aggaddam 
Hey I guess the factory reset fails because it is called twice, here are the wifi debug binaries which will help us if that's the real cause of problem. Also if possible please attach complete debug console logs, thanks. 
[matter_libs.zip](https://github.com/espressif/esp-matter/files/13038954/matter_libs.zip)
Please copy these libraries in esp-idf/components/esp_wifi/lib/ folder and try to reproduce the error again.

aggaddam, 2023-10-25:
@shreyassheth1 

Here are the logs i collected for both success and failure cases

1) Success log Scenario: Factory reset works fine when it is initiated upon a fabric being removed and if 0 fabrics in the device from the callback. Here is the snapshot of logic
[Factory_reset_success_when_called_if_fabric_is_removed.txt](https://github.com/espressif/esp-matter/files/13162594/Factory_reset_success_when_called_if_fabric_is_removed.txt)

2) Failure log Scenario: Factory reset fails if initiated from any application task after successful matter commissioning. 
[Factory_reset_failed_when_called_async_after_commissioning.txt](https://github.com/espressif/esp-matter/files/13162597/Factory_reset_failed_when_called_async_after_commissioning.txt)

shreyassheth1, 2023-10-31:
Hey @aggaddam
This is not wifi specific issue. It seems that either esp_wifi_restore is getting called twice, or esp_wifi_stop then esp_wifi_restore is getting called. Can you please help us understand what does this print mean?
> I (88752) PAL: RESETTING

aggaddam, 2023-10-31:
@shreyassheth1 It is just a log print before calling matter_helperfcn_factory_reset()

aggaddam, 2023-11-07:
Closing this issue as we figured out that our application is causing the issue for the factory reset to fail.

closed 2023-11-07 by aggaddam