espressif/esp-matter #624: VSCode Linux VM USB Connection to esp32-c3

tpitman, 2023-09-08:
**Describe the question/query that you have**
I am running a virtual machine on an M1 Mac in Parallels. I am doing this because I can't get the esp32 matter stuff to work native on the Mac yet. Still pursuing that, but wanted to try linux in case it was better supported.

I am trying to get VSCode set up to develop matter devices on the esp32.

I installed VSCode and was able to build the blink example.

When I plug my esp32-c3 dev board into my computer it allows me to attach it to my VM.

lsusb shows the following device that shows up:

`Espressif USB JTAG/serial debug unit`

ls /dev/tty* shows this device that goes away if I unplug it:

`/dev/ttyACM0`

There are no devices with ttyUSB in the name as other examples seem to indicate.

When I run the flash command and say yes to starting OpenOCD I see the following:

`/home/tpitman/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/bin/openocd`
` `
`Open On-Chip Debugger v0.12.0-esp32-20230419 (2023-04-18-22:02)`
`Licensed under GNU GPL v2`
`For bug reports, read`
`	http://openocd.org/doc/doxygen/bugs.html`
` `
`/home/tpitman/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/bin/openocd`
` `
`[OpenOCD]`
`Open On-Chip Debugger v0.12.0-esp32-20230419 (2023-04-18-22:02)`
`Licensed under GNU GPL v2`
`For bug reports, read`
`	http://openocd.org/doc/doxygen/bugs.html`
`debug_level: 2`
` `
`[OpenOCD]`
`adapter speed: 20000 kHz`
` `
`[OpenOCD]`
`WARNING: boards/esp32-wrover.cfg is deprecated, and may be removed in a future release.`
`         If your board is ESP32-WROVER-KIT, use board/esp32-wrover-kit-1.8v.cfg instead.`
` `
`[OpenOCD]`
`Info : Listening on port 6666 for tcl connections`
`Info : Listening on port 4444 for telnet connections`
` `
`[OpenOCD]`
`❌ Error: unable to open ftdi device with description '*', serial '*' at bus location '*'`
`/home/tpitman/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts/target/esp_common.cfg:4: Error: `
`at file "/home/tpitman/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts/target/esp_common.cfg", line 4`
` `
`[OpenOCD]`
`[Stopped] : OpenOCD Server`
`[Flash]`
`Can't perform JTag flash, because OpenOCD server is not running!`
` `
`Any ideas to get it so I can flash my esp32 board in this situation? I will also want to debug, but hope that if I can flash I will be able to debug.`

sayondeep, 2023-09-11:
@tpitman 
1)Have you followed the instructions given [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/jtag-debugging/index.html#jtag-debugging-run-openocd) ?

2)Are you using ESP32-C3's built-in JTAG interface or USB Serial/JTAG Controller?

tpitman, 2023-09-11:
Thank you for the link. I will try it out.

I am using the built in usb serial / jtag.

dhrishi, 2023-09-21:
@tpitman Please close this issue if it works

tpitman, 2023-09-21:
Through some help of others I was able to get this working on my Mac instead, so I do not know if this will fix the issue. Since I will not be pursuing this, I will close the issue...

closed 2023-09-21 by tpitman