espressif/esp-matter #626: matter light example build error

tpitman, 2023-09-10:
**Describe the bug**
I just did a clean install of the VSCode plugin for esp-idf and esp-matter.

I used the command menu to make a copy of the matter light example.

When I build the example I get the following build error(s):

`[2/543] Performing build step for 'bootloader'`
`[1/1] cd /Users/tpitman/Projects/Mindfire/LimeLight/esp32/light/build/bootloader/esp-idf/esptool_py && /Users/tpitman/.espressif/python_env/idf5.1_py3.11_env/bin/python /Users/tpitman/esp/esp-idf/components/partition_table/check_sizes.py --offset 0xc000 bootloader 0x0 /Users/tpitman/Projects/Mindfire/LimeLight/esp32/light/build/bootloader/bootloader.bin
Bootloader binary size 0x5300 bytes. 0x6d00 bytes (57%) free.`
`[5/543] Performing configure step for 'chip_gn'`
`FAILED: esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-configure /Users/tpitman/Projects/Mindfire/LimeLight/esp32/light/build/esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-configure 
cd /Users/tpitman/Projects/Mindfire/LimeLight/esp32/light/build/esp-idf/chip && /Users/tpitman/.espressif/esp-matter/connectedhomeip/connectedhomeip/.environment/cipd/packages/pigweed/gn --root=/Users/tpitman/.espressif/esp-matter/connectedhomeip/connectedhomeip/config/esp32 gen --check --fail-on-unused-args /Users/tpitman/Projects/Mindfire/LimeLight/esp32/light/build/esp-idf/chip && /opt/homebrew/Cellar/cmake/3.27.1/bin/cmake -E touch /Users/tpitman/Projects/Mindfire/LimeLight/esp32/light/build/esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-configure`
`ERROR at //third_party/connectedhomeip/src/app/chip_data_model.gni:192:26: Script returned non-zero exit code.
      _cluster_sources = exec_script("${_app_root}/zap_cluster_list.py",`
`                         ^----------`
`Current dir: /Users/tpitman/Projects/Mindfire/LimeLight/esp32/light/build/esp-idf/chip/
Command: python /Users/tpitman/.espressif/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/app/zap_cluster_list.py --zap_file ../../../../../../../../.espressif/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/controller/data_model/controller-clusters.zap`
`Returned 1.`
`stderr:`
` `
`  File "/Users/tpitman/.espressif/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/app/zap_cluster_list.py", line 12
    def get_cluster_sources(clusters: typing.Set[str],`
`                                    ^`
`SyntaxError: invalid syntax`
` `
`See //third_party/connectedhomeip/src/controller/data_model/BUILD.gn:43:1: whence it was called.`
`chip_data_model("data_model") {`
`^------------------------------`
`See //third_party/connectedhomeip/src/lib/shell/commands/BUILD.gn:66:11: which caused the file to be included.
        [ "${chip_root}/src/controller/data_model:data_model_zapgen_config" ]`
`          ^----------------------------------------------------------------`
`[7/543] Linking C static library esp-idf/esp_adc/libesp_adc.a`
`ninja: build stopped: subcommand failed.`

**Environment**
- ESP-Matter Commit Id: 41b143e0bc600d7e72a522ec7df0051951d07e98
- ESP-IDF Commit Id: not sure. latest installed for VSCode
- SoC (eg: ESP32 or ESP32-C3): esp32-c6
- Device Logs (Please attach the log file): NA
- Host Machine OS: MacOS Apple Silicon
- Commissioner app and versions if present: NA
- Commissioner's logs if present: NA

sayondeep, 2023-09-11:
@tpitman can you please try to install esp-matter in different directory and not inside esp-idf tools directory [do not select (use current esp-idf tools directory) while installing from the extension]. You can do this instead,
choose existing repository --> github --> select a directory to install esp-matter.

tpitman, 2023-09-11:
I followed your instructions and chose to install the esp-matter in a different folder.

I then ran the built process and it didn't give those same errors, BUT

I then went to flash the project to my esp32-c6 dev board and realized I had not set those yet.

So I then set the device configuration by choosing the esp32-c6 and then the serial port of the usb-jtag.

I then went to build and flash and got the same build errors as mentioned above.

Hopefully this gives a little more information so you can help me resolve it.

sayondeep, 2023-09-12:
@tpitman can you please share the log file. (do not paste the logs)

tpitman, 2023-09-12:
> @tpitman can you please share the log file. (do not paste the logs)

Which log file exactly do you want me to share?

tpitman, 2023-09-12:
Sorry to keep asking questions, but I want to make sure I send the right thing.

There is output in the terminal when I build and that is where the error is shown. It is really just what I pasted above in the original issue report. I assume that is not what you want because I have already pasted it.

I have found a few logs inside the project in different places, but none of them contain the error shown above or during the build.

Please tell me what log in what folder you are asking for so I can provide the right thing.

sayondeep, 2023-09-12:
@tpitman it cannot be same because...the location of esp-matter has changed...if it is exactly same that would mean ESP_MATTER_PATH is not set to the new correct path..

sayondeep, 2023-09-12:
@tpitman can you please go in the directory of esp-matter and run `./install.sh` . Does it builds successfully?

tpitman, 2023-09-12:
When I run source /path/to/esp-idf/export.sh it hangs on "Checking other ESP-IDF version."

When I did the install a while back the first time it didn't do that. It ran right through it.

When I discovered I can install it all via the VSCode plugin, that worked without hanging as well. In fact, if I do the VSCode esprissif plugin install it works fine.

Only when I tried to run export at the command line does it hang up like this.

I let it run for 15 minutes before I finally hit ctrl-c to stop it.

shubhamdp, 2023-09-13:
export.sh has to be sourced and not executed. `source esp-idf/export.sh`.

tpitman, 2023-09-13:
Yes it was. I did that just before doing install.sh. I have tried it many times. It hangs every time.