espressif/esp-matter #629: Numbering of bridged endpoints

elektro-NIK, 2023-09-12:
I am working on developing a bridge between Matter and non-Matter devices.

My bridge must be able to dynamically create and delete bridged endpoints. I do this using the `esp_matter_bridge::create_device` and `esp_matter::endpoint::destroy` APIs and I noticed that endpoint IDs for new devices are issued one after the other, even if there are endpoints released (using `destroy`) before that. This leads to the fact that when I create and delete device IDs, they reach the maximum number for the bridge and no more are created.

I've researched this behavior a bit, and it's because, in `esp_matter_core.cpp`, in the `endpoint::create` function, the `min_unused_endpoint_id` field is always incremented and stored in NVS.

Is this the correct behavior and is there any way around it?

wqx6, 2023-09-12:
Yes, the SPEC says we should select endpoint address for dynamic endpoint by`incrementing from the highest previously (ever) used endpoint address`. And the SPEC also says that when the endpoint address reaches 65535(UINT16_MAX), `it MAY wrap around to the lowest unused endpoint address`. We are planning to add this logic later.

elektro-NIK, 2023-09-12:
Ok, thank you.

I continued experimenting and ran into the following problem. My devices stop being created after 125 creation-deletions. Because 2 namespaces are used for each endpoint, and with 'esp_matter_bridge::erase_bridged_device_info', the 'nvs_erase_all' API is called, which erases all key-value pairs, but the namespace itself remains, and NVS can only have 255 namespaces.

Is there a way around this?

wqx6, 2023-09-13:
@elektro-NIK Thanks for report. Will fix it ASAP.

dhrishi, 2024-02-27:
This has been fixed with https://github.com/espressif/esp-matter/commit/98c6e8070b582c9b18058256f490e6b02a4264e1

closed 2024-02-27 by dhrishi