espressif/esp-matter #630: Build error with changing the vendor and product ID

SYESHASWINI, 2023-09-14:
I have changed the vendor and product ID's for our product and tried building the project, which throws error. 
Attaching the error log
[deviceID_errors.txt](https://github.com/espressif/esp-matter/files/12608299/deviceID_errors.txt)

I have generated the Certs don't where to place exactly to solve this errors.

dhrishi, 2023-09-14:
@SYESHASWINI Please share the details about the mfg_tool commands used and upload your sdkconfig file.

SYESHASWINI, 2023-09-15:
@dhrishi I have used chip cert commands for generating the certificates. 
This is the sdk file:
[sdk_cert.txt](https://github.com/espressif/esp-matter/files/12615527/sdk_cert.txt)

shubhamdp, 2023-09-15:
@SYESHASWINI Please take a look at [mfg_tool](https://github.com/espressif/esp-matter/tree/main/tools/mfg_tool).

You have configured the VID/PID to non-test and still using the Test DAC provider and it is causing the problem. Test DAC provider only have test certs from VID:0xFFF1-0xFFF4 and PID:0x8000:0x8063.

And hence it is failing to compile. You will have to generate the manufacturing binary and flash it, then set the correct config options, build/flash firmware and that would work. All the info you need is in mfg_tool's readme.

dhrishi, 2023-09-15:
I am closing this issue. Please refer the README and and it should help. Feel free to re-open in case of anything.

closed 2023-09-15 by dhrishi

SYESHASWINI, 2023-09-15:
@shubhamdp I tried generating partition using the command

./mfg_tool.py -cn "My bulb" -v 0xFFF2 -p 0x8001 --pai \
   -k path/to/esp-matter/connectedhomeip/connectedhomeip/credentials/test/attestation/Chip-Test-PAI-FFF2-8001-Key.pem \
   -c path/to/esp-matter/connectedhomeip/connectedhomeip/credentials/test/attestation/Chip-Test-PAI-FFF2-8001-Cert.pem \
   -cd path/to/esp-matter/connectedhomeip/connectedhomeip/credentials/test/certification-declaration/Chip-Test-CD-FFF2-8001.der

where it throws the following error:

Traceback (most recent call last):
  File "/home/ubuntu/esp-matter/esp-matter/tools/mfg_tool/./mfg_tool.py", line 630, in <module>
    main()
  File "/home/ubuntu/esp-matter/esp-matter/tools/mfg_tool/./mfg_tool.py", line 623, in main
    write_per_device_unique_data(args)
  File "/home/ubuntu/esp-matter/esp-matter/tools/mfg_tool/./mfg_tool.py", line 360, in write_per_device_unique_data
    generate_onboarding_data(args, int(row['Index']), int(chip_factory_get_val('discriminator')), int(row['PIN Code']))
  File "/home/ubuntu/esp-matter/esp-matter/tools/mfg_tool/./mfg_tool.py", line 438, in generate_onboarding_data
    if args.commissioning_flow == CommissioningFlow.Standard:
NameError: name 'CommissioningFlow' is not defined