espressif/esp-matter #631: How do I return TLV from a custom command?

jonsmirl, 2023-09-14:
All of the examples only show sending TLV. How do I return it?

PSONALl, 2023-09-15:
@jonsmirl The TLV types for commands are defined for different clusters and their commands in connectedhomeip, e.g The Response type for ViewScene command has kGroupID and kSceneID [reference](https://github.com/project-chip/connectedhomeip/blob/master/zzz_generated/app-common/app-common/zap-generated/cluster-objects.cpp#L1065)

So, The TLV type for a custom command needs to be defined in similar way

jonsmirl, 2023-09-15:
How do I do AddResponse in the esp-matter model?

PSONALl, 2023-09-18:
The ESP-Matter command callback API is designed to provide an abstraction to users, The TLV Writer's handling is expected to be done internally from connectedhomeip, Which is why it's not implemented currently in esp-matter

jonsmirl, 2023-10-31:
This needs to be done using Matter APIs. I have implemented now using those APIs.

closed 2023-10-31 by jonsmirl