espressif/esp-matter #634: Read write of custom complex attributes - lists, arrays, structs

jonsmirl, 2023-09-14:
I gave up trying to figure how to implement this. I think all of the necessary APIs are there, but I am not sure how to combine them all. I would be uses to have an a working example.

I switched over to using custom commands instead. But now I can't figure out how to return a structure from the command.
https://github.com/espressif/esp-matter/issues/631

This would be a good demo to add to the bridge examples. I need a custom cluster for bridge control so I am building one.  Attributes with things like how BLE address is mapped to EP.  Event saying a BLE device needs to be commissioned, way to pass in OOB data, etc.

For a first demo. Custom cluster with an array attribute showing BLEaddr -> EP mapping.  Then if you delete a row and write the attribute back, that device gets removed from the bridge. Next add another column for on/off-line status.

wqx6, 2023-09-15:
Complex attributes always have `AttributeAccessInterface` implementation for its cluster. so for the clusters defined in SPEC, when we create the complex attributes in those cluster, we will use a null pointer for it value as this attribute is managed internally.   If you are trying to create a custom cluster with complex attribute. we will recommend to use an `AttributeAccessInterface` implementation for it and call `registerAttributeAccessOverride` in the cluster's `PluginServerInitCallback`.

jonsmirl, 2023-09-20:
I've coded this up and it does work. However, it is complex to implement this on both the front and back sides. I'm exploring other ways to achieve what I need so that I can avoid this complexity.

jonsmirl, 2024-05-28:
I am able to do this, but it requires a lot of CHIP internal knowledge to do it.

closed 2024-05-28 by jonsmirl