espressif/esp-matter #635: Looking for ways to read DAC and confirm its OK

eme-bennytang, 2023-09-15:
I am currently using the esp_secure_cert partition as my DAC provider
I have configured the relevant settings and proved its working.

I am wondering is there an API I can call such that if I accidentally erased / left that partition un-programmed, I can know the DAC is not read properly and prompt error in my manufacturing review.

I know there are ESP32SecureCertDACProvider, but I just can't make it work.
Is there an example code on how to read the DAC and confirm its ok?

Thanks

YOGYUI, 2023-09-16:
`ESP32SecureCertDACProvider` works with **CONFIG_SEC_CERT_DAC_PROVIDER=y** option in sdkconfig.
(you can check this in `setup_providers()` function in `components/esp_matter/esp_matter_providers.cpp` code)

If you implemented esp_secure_cert partition and flashed proper secure cert binary file into the partition, then you can read and confirm DAC like below psuedo code.

``` c
#include <credentials/CHIPCert.h>
#include <credentials/DeviceAttestationCredsProvider.h>
#include <platform/CommissionableDataProvider.h>

using namespace chip;
using namespace chip::DeviceLayer;
using namespace chip::Credentials;
uint16_t discriminator;

CHIP_ERROR chip_err = GetCommissionableDataProvider()->GetSetupDiscriminator(discriminator);
if (chip_err == CHIP_NO_ERROR) {
    ESP_LOGI(TAG, "fctry successfully read");
} else {
    ESP_LOGI(TAG, "fctry read failed");
}

uint8_t der_cert_buf[kMaxDERCertLength];
MutableByteSpan der_cert_span(der_cert_buf);
chip_err = GetDeviceAttestationCredentialsProvider()->GetDeviceAttestationCert(der_cert_span);
if (chip_err == CHIP_NO_ERROR) {
    ESP_LOGI(TAG, "esp_secure_cert successfully read");
} else {
    ESP_LOGI(TAG, "esp_secure_cert read failed");
}

AttestationCertVidPid dacVidPid;
chip_err = ExtractVIDPIDFromX509Cert(der_cert_span, dacVidPid);
if (chip_err == CHIP_NO_ERROR) {
    ESP_LOGI(TAG, "VID in DAC: %d, PID in DAC: %d", dacVidPid.mVendorId.Value(), dacVidPid.mProductId.Value());
} else {
    ESP_LOGI(TAG, "DAC Format Invalid");
}
```

shubhamdp, 2023-09-18:
@eme-bennytang, I think [mfg_test_app](https://github.com/espressif/esp-matter/blob/main/examples/mfg_test_app) would answer your doubts.

eme-bennytang, 2023-09-18:
Hi @shubhamdp, 
This looks like exactly what I am looking for!
My IDF release/v5.1 haven't got this, so I assume there is no such thing....

Thanks anyway

eme-bennytang, 2023-09-19:
Hi @shubhamdp,

I have re-opened this issue
I am able to verify the DAC according to the examples.
However I want to verify the CD's VID/PID (or origin VID/PID) against the DAC VID/PID. (as specified in Matter SPEC)

I am able to retrieve the DAC VID/PID, and I can get CD from Factory Data, but is there any API I can call to retrieve VID/PID from CD?

Thanks again for the great help

eme-bennytang, 2023-09-20:
[image]
I have tried the above ways, but the CD decode seems fail, no idea why

shubhamdp, 2023-10-04:
@eme-bennytang, sorry for the late response.

I think you are looking for [`DecodeCertificationElements`](https://github.com/project-chip/connectedhomeip/blob/master/src/credentials/CertificationDeclaration.h#L137-L145).

Or [`DeviceAttestationVerifier`](https://github.com/project-chip/connectedhomeip/blob/master/src/credentials/attestation_verifier/DefaultDeviceAttestationVerifier.h#L57) could be of your interest.

eme-bennytang, 2023-10-06:
@shubhamdp  Thanks for the response!
However I have used this DecodeCertificationElements as the above reply.

And it will always return decode error, not sure where the problem is

eme-bennytang, 2023-10-06:
My CD is generated by below command
out/host/chip-cert gen-cd \
--format-version 1 \
--device-type-id 0016 \
--vendor-id 1463 --product-id 0001 \
--dac-origin-vendor-id 131B --dac-origin-product-id 8013 \
--certificate-id CSA00000SWC00000-01 -l 0 -i 0 -n 1 -t 0 \
--key credentials/test/certification-declaration/Chip-Test-CD-Signing-Key.pem \
--cert credentials/test/certification-declaration/Chip-Test-CD-Signing-Cert.pem \
--out ../../EME_Test_Certs/CD/EME-Test-CD-1463-0001-131B-8013.der

And I have confirmed the read out CD byte span is exactly the one I have programmed.

shubhamdp, 2023-10-06:
Ohh, I overlooked the screenshot, sorry for that. Well, in that case what is the error returned by the API? We can check the error returned and see whats wrong in the code.

shubhamdp, 2023-10-06:
I looked through the source again and the problem is we are providing CMS encoded data to the API and it expects the TLV encoded data.

Try this, extract the TLV out of CMS encoded CD. I've tested it with test cert and works!
```
ByteSpan tlv_cd_content;
CMS_ExtractCDContent(cd_span, tlv_cd_content);
DecodeCertificationElements(tlv_cd_content, certElementsOut);
```

eme-bennytang, 2023-10-06:
Oh great! thanks for the help!
I will try that ASAP and let you know if I see any other issue on this matter

dhrishi, 2023-10-13:
@eme-bennytang Please close this if your issue is resolved

eme-bennytang, 2023-10-16:
Sorry for not closing, solution from [shubhamdp](https://github.com/shubhamdp) works perfectly
Thanks everyone

closed 2023-10-16 by eme-bennytang