espressif/esp-matter #636: Error while generating partition

SYESHASWINI, 2023-09-15:
In reference to the issue #630 
@shubhamdp I tried generating partition using the command

./mfg_tool.py -cn "My bulb" -v 0xFFF2 -p 0x8001 --pai
-k path/to/esp-matter/connectedhomeip/connectedhomeip/credentials/test/attestation/Chip-Test-PAI-FFF2-8001-Key.pem
-c path/to/esp-matter/connectedhomeip/connectedhomeip/credentials/test/attestation/Chip-Test-PAI-FFF2-8001-Cert.pem
-cd path/to/esp-matter/connectedhomeip/connectedhomeip/credentials/test/certification-declaration/Chip-Test-CD-FFF2-8001.der

where it throws the following error:

Traceback (most recent call last):
File "/home/ubuntu/esp-matter/esp-matter/tools/mfg_tool/./mfg_tool.py", line 630, in
main()
File "/home/ubuntu/esp-matter/esp-matter/tools/mfg_tool/./mfg_tool.py", line 623, in main
write_per_device_unique_data(args)
File "/home/ubuntu/esp-matter/esp-matter/tools/mfg_tool/./mfg_tool.py", line 360, in write_per_device_unique_data
generate_onboarding_data(args, int(row['Index']), int(chip_factory_get_val('discriminator')), int(row['PIN Code']))
File "/home/ubuntu/esp-matter/esp-matter/tools/mfg_tool/./mfg_tool.py", line 438, in generate_onboarding_data
if args.commissioning_flow == CommissioningFlow.Standard:
NameError: name 'CommissioningFlow' is not defined

jonsmirl, 2023-09-15:
Did you install the requirements.txt file in the mfg_tool directory?
pip install -r requirements.txt

shripad621git, 2023-09-18:
@SYESHASWINI please ensure the following things before you execute the command to create a partition:
1. Please ensure that you have exported the esp-idf and esp-matter path.
2. Please install the requirements as mentioned [here](https://github.com/espressif/esp-matter/tree/main/tools/mfg_tool#install-python-dependencies).
3. Please check that you provide the appropriate path of the files mentioned in the command.
   The command should execute fine if you folllow all the above steps well.

[image]

SYESHASWINI, 2023-09-20:
I have installed the requirements, still getting the same error.

error:
[2023-09-20 11:38:45,700] [   INFO] - Input Root certificate type PAI
[2023-09-20 11:38:45,700] [   INFO] - Number of manufacturing NVS images to generate: 1
[2023-09-20 11:38:45,849] [   INFO] - Generating Config CSV...
[2023-09-20 11:38:45,850] [   INFO] - Writing chip manifest CSV header...
[2023-09-20 11:38:45,858] [   INFO] - Generated PAI certificate in DER format: out/105e_49de/staging/pai_cert.der
[2023-09-20 11:38:45,868] [   INFO] - Generated DAC certificate: out/105e_49de/bc59141d-5fbe-4e33-94f7-83bd0120d072/internal/DAC_cert.pem
[2023-09-20 11:38:45,868] [   INFO] - Generated DAC private key: out/105e_49de/bc59141d-5fbe-4e33-94f7-83bd0120d072/internal/DAC_key.pem
[2023-09-20 11:38:45,868] [   INFO] - Generated DAC certificate in DER format: out/105e_49de/bc59141d-5fbe-4e33-94f7-83bd0120d072/internal/DAC_cert.der
[2023-09-20 11:38:45,868] [   INFO] - Generated DAC private key in binary format: out/105e_49de/bc59141d-5fbe-4e33-94f7-83bd0120d072/internal/DAC_private_key.bin
[2023-09-20 11:38:45,868] [   INFO] - Generated DAC public key in binary format: out/105e_49de/bc59141d-5fbe-4e33-94f7-83bd0120d072/internal/DAC_public_key.bin
[2023-09-20 11:38:45,869] [   INFO] - Appending chip master CSV row...
Traceback (most recent call last):
  File "/home/ubuntu/esp-matter/esp-matter/tools/mfg_tool/./mfg_tool.py", line 630, in <module>
    main()
  File "/home/ubuntu/esp-matter/esp-matter/tools/mfg_tool/./mfg_tool.py", line 623, in main
    write_per_device_unique_data(args)
  File "/home/ubuntu/esp-matter/esp-matter/tools/mfg_tool/./mfg_tool.py", line 360, in write_per_device_unique_data
    generate_onboarding_data(args, int(row['Index']), int(chip_factory_get_val('discriminator')), int(row['PIN Code']))
  File "/home/ubuntu/esp-matter/esp-matter/tools/mfg_tool/./mfg_tool.py", line 438, in generate_onboarding_data
    if args.commissioning_flow == CommissioningFlow.Standard:
NameError: name 'CommissioningFlow' is not defined

shripad621git, 2023-09-20:
The error you are getting is due to the CommissioningFlow is not getting imported correctly which is getting imported from` esp-matter/connectedhomeip/connectedhomeip/src/setup_payload/python/generate_setup_payload.py.`
Please perform the steps mentioned below:
1.Please use the latest main branch.
2.Go to  esp-matter/connectedhomeip/connectedhomeip and do a` rm -rf .environment/`.
3.Now traverse back to esp-matter and do a` install.sh` and `export.sh` before running the mfg_tool command.
Please feel free to reach out in case the issue is not addressed.

SYESHASWINI, 2023-09-20:
Tried the following steps. The commissioning flow is not getting imported.

1.Please use the latest main branch.
2.Go to esp-matter/connectedhomeip/connectedhomeip and do a rm -rf .environment/.
3.Now traverse back to esp-matter and do a install.sh and export.sh before running the mfg_tool command.

shripad621git, 2023-09-21:
@SYESHASWINI, I have tried the command on the release/v1.0 commit and I was able to reproduce your issue.
On release/v1.0 of esp-matter can you please follow the steps below?
1. git reset --hard 8cf029456afcfb71748d77649851e6df6fcf4bfc.
2. Please try the mfg_tool command on this commit.

SYESHASWINI, 2023-09-21:
@shripad621git Now I'm able to generate the bin files, Thank You.

Could you help me with few more questions:

- I have observed that the partition table chooses **"chip-factory"** partition by default, where the sub-type is not mentioned. how can I change it to other one.
- In the project we have esp_secure_cert partition. Is Matter attestation info is by default stored in this partition.
- By executing this command let's the certification info into the program or do we need to perform some more steps to include the cert info.
 `esptool.py -p <serial_port> write_flash <address> path/to/<uuid>-partition.bin`

Could you share docs if any for the reference.

shripad621git, 2023-09-22:
> @shripad621git Now I'm able to generate the bin files, Thank You.
>
> Could you help me with few more questions:
>
> * I have observed that the partition table chooses **"chip-factory"** partition by [...]

@SYESHASWINI you can refer [this](https://github.com/espressif/esp-matter/tree/release/v1.0/tools/mfg_tool#configure-your-app) for configuring your app.
You can refer [this](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/esp32/secure_cert_partition.md) to generate a esp_secure_cert partition on release/v1.0. 
Please enabe the option `CONFIG_SEC_CERT_DAC_PROVIDER` after flashing the esp_secure_cert_partition at the appropriate address.
Executing the esptool.py will flash the factory partition at the specified address.You do not need to do anthing else to include cert information.

SYESHASWINI, 2023-09-22:
@shripad621git 
I have followed [this](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/esp32/secure_cert_partition.md) file, generated esp_secure_cert.bin file by the command

> configure_esp_secure_cert.py --private-key Esp-Development-DAC-Key-01.der \
>     --device-cert Esp-Development-DAC-01.der \
>     --ca-cert Esp-Development-PAI-Cert.der \
>     --target_chip esp32c3 \
>     --keep_ds_data_on_host \
>     --port /dev/ttyUSB0

and flashed at esp_secure_cert partition(0xd000)

Generated factory partition using:

> ./generate_esp32_chip_factory_bin.py -d 3434 -p 99663300 \
>     --product-name ESP-lighting-app --product-id 0x8000 \
>     --vendor-name Test-vendor --vendor-id 0xFFF1 \
>     --hw-ver 1 --hw-ver-str DevKit \
>     --cd esp_dac_fff1_8000.der
and flashed at(0x3E000)

when I enabled CONFIG_SEC_CERT_DAC_PROVIDER as the esp_secure_cert is flased at sec_cert partiton , during commissioning it shows **FailedCertificateChainREquest.**

when changed it to Factory partition the device is commissioned successfully.

It has to work with esp_secure_cert partition when DAC certification information is flashed at that partition instead works with factory partition.

shripad621git, 2023-09-25:
@SYESHASWINI Please follow the below steps to successfully commission the device by flashing esp_secure_cert partition.
1.Generate the `secure_cert_partition` with the same command as mentioned above.
2.In release/v1.0, you need to provide dac-cert,dac-key and pai-cert as arguments to the factory_partition script.So,generate the factory_partition using the above command and the three additional arguments.The value of these arguments will be same as those used in generating esp_secure_cert_partition.
3.Please replace the partitons.csv file of generic_switch example with light example partitons.csv file.A internal fix will be raised for change in the partitions.csv file of the generic_switch example.
4.Flash the partitions at appropriate addresses.
5.Please ensure that you enable the config options mentioned in the esp_secure_cert document.Also please **do not disable** `CONFIG_ENABLE_ESP32_DEVICE_INFO_PROVIDER.`
6.During commissioning please provide the argument --paa-trusted-store-path.
```
For ex:./chip-tool pairing ble-wifi [node-id] [ssid [password] 99663300 3434 --paa-trust-store-path /path/to/esp-matter/connectedhomeip/connectedhomeip/credentials/development/attestation/
```
Use the above command to commission the device.I have given it a try on my side.The commissioning is successful with these steps.

SYESHASWINI, 2023-09-27:
@shripad621git As we are flashing factory-partition.bin Is there any use of esp_secure_cert partiton. I flashed only factory-partition bin at fctry partition address with dac keys and all, the commission is succesfull. So is there any purpose for esp_secure_cert_partition to be flashed.

shripad621git, 2023-09-27:
@SYESHASWINI ,there are two ways for flashing the partitons.If you enable the `CONFIC_SEC_CERT_DAC_PROVIDER` and flash the esp_secure_cert_partition along with factory_partiton then, then the attestation information like `DAC, DAC private key, and PAI certificate` is read from the `esp_secure_cert_partiton` and `certificate-declaration(cd)` is read from the` factory_partiton `along with other data.
It is fine if you flash the `factory_partiton `with `dac keys `and the commissioning is successful.In that case ,the `FACTORY_PARTITION_DAC_PROVIDER `reads the attestation information from factory_partition.The commissioning is successful because this option is automatically set based on `CONFIG_ENABLE_ESP32_FACTORY_DATA_PROVIDER.`
For more info about esp_secure_cert partiton ,you can refer [this](https://github.com/espressif/esp_secure_cert_mgr#esp-secure-cert-partition) link.

SYESHASWINI, 2023-09-27:
Yes @shripad621git but, when I tried enabling the esp_secure_cert partition in the menuconfig the commissioning window is getting closed.

The changes I made to enable esp-secure_cert partiton are:

**Entered the partiton name as esp_secure_Cert**

> Set chip-factory namespace partition label [Component config → CHIP Device Layer → Matter Manufacturing Options → chip-factory namespace partition label]
> Set config option [CHIP_FACTORY_NAMESPACE_PARTITION_LABEL](https://github.com/project-chip/connectedhomeip/blob/master/config/esp32/components/chip/Kconfig#L856) to choose the label of the partition to store key-values in the "chip-factory" namespace. The dafault chosen partition label is nvs.

**Choosed attestation to secure_cert.**

> Enable Attestation -secure-cert [ Component config → ESP Matter → DAC Provider options → Attestation - secure-cert]
> Enable config option CONFIG_FACTORY_PARTITION_DAC_PROVIDER to use DAC certificates from the factory partition during Attestation.

shripad621git, 2023-09-27:
> @SYESHASWINI Please follow the below steps to successfully commission the device by flashing esp_secure_cert partition. 1.Generate the `secure_cert_partition` with the same command as mentioned [...]

Make sure you follow these steps thoroughly. Please share your sdkconfig file and device logs.

SYESHASWINI, 2023-09-28:
I'm trying to commission through GH and Alexa following the above steps.
Attaching the log and sdk file.
[cmmsn_closed.txt](https://github.com/espressif/esp-matter/files/12745503/cmmsn_closed.txt)
[sdk_certt.txt](https://github.com/espressif/esp-matter/files/12745504/sdk_certt.txt)

shripad621git, 2023-10-03:
@SYESHASWINI ,please set the `CONFIG_CHIP_FACTORY_NAMESPACE_PARTITION_LABEL="fctry"` and give a try. If you flash a secure_cert_partiton  along with factory_partiton you need to enable the configuration options mentioned [here](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/esp32/secure_cert_partition.md#15-build-the-firmware-with-below-configuration-options).

SYESHASWINI, 2023-10-03:
@shripad621git I have set the partition name as fctry and attestation as esp_secure_cert which shows that the validation of chain is getting failed.
[certificatechainfail.txt](https://github.com/espressif/esp-matter/files/12791037/certificatechainfail.txt)

shripad621git, 2023-10-04:
@SYESHASWINI please send the chip-tool logs.

closed 2023-10-13 by dhrishi