espressif/esp-matter #637: BLE:Error writing Characteristics in Chip service on the device when commissioning

richardklingler, 2023-09-15:
Hello

Installed latest ESP-IDF and matter framework as of today on macOS 13.5.2.

Compiling and flashing the light example works fine..and I can connect via serial port to the ESP32_Devkitc_v4 board.

Either through the iOS Espressif app or chiptool I can't commission the device at all, resulting in the same error outputs:

```
[1694778139060] [22643:294475] [TOO] Command: pairing ble-wifi 0x7283 mySSID myPassword 20202021 3840 
[1694778139062] [22643:294477] [CTL] Setting attestation nonce to random value
[1694778139062] [22643:294477] [CTL] Setting CSR nonce to random value
[1694778139065] [22643:294477] [CTL] Generating NOC
[1694778139068] [22643:294477] [FP] Validating NOC chain
[1694778139076] [22643:294477] [FP] NOC chain validation successful
[1694778139076] [22643:294477] [FP] Updated fabric at index: 0x1, Node ID: 0x000000000001B669
[1694778139076] [22643:294477] [TS] Last Known Good Time: 2023-09-15T10:49:58
[1694778139076] [22643:294477] [TS] New proposed Last Known Good Time: 2021-01-01T00:00:00
[1694778139076] [22643:294477] [TS] Retaining current Last Known Good Time
[1694778139078] [22643:294477] [FP] Metadata for Fabric 0x1 persisted to storage.
[1694778139079] [22643:294477] [TS] Committing Last Known Good Time to storage: 2023-09-15T10:49:58
[1694778139080] [22643:294477] [CTL] Joined the fabric at index 1. Fabric ID is 0x0000000000000001 (Compressed Fabric ID: E702B60C0D26E86E)
[1694778139080] [22643:294477] [DIS] Updating services using commissioning mode 0
[1694778139087] [22643:294477] [BLE] NewConnection
[1694778139087] [22643:294477] [CTL] Setting wifi credentials from parameters
[1694778139087] [22643:294477] [CTL] Setting attestation nonce to random value
[1694778139087] [22643:294477] [CTL] Setting CSR nonce to random value
[1694778139087] [22643:294477] [CTL] Commission called for node ID 0x0000000000007283
[1694778139119] [22643:294477] [BLE] CBManagerState: ON
[1694778139154] [22643:294477] [BLE] Connecting to device 0x600003bf81a0 with discriminator: 3840
[1694778140020] [22643:294478] [DIS] Closing all BLE connections
[1694778140020] [22643:294478] [IN] BleConnectionComplete: endPoint 0x102bedf58
[1694778140020] [22643:294478] [BLE] ATT MTU = 256
[1694778140021] [22643:294477] [BLE] BLE:Error writing Characteristics in Chip service on the device: [The specified UUID is not allowed for this operation.]
[1694778140022] [22643:294478] [IN] SecureSession[0x600003cf4200]: Allocated Type:1 LSID:42770
[1694778140022] [22643:294478] [SC] Assigned local session key ID 42770
[1694778140022] [22643:294478] [EM] <<< [E:469i S:0 M:26728876] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:20 (SecureChannel:PBKDFParamRequest)
[1694778140022] [22643:294478] [IN] (U) Sending msg 26728876 to IP address 'BLE'
[1694778140022] [22643:294478] [IN] Message appended to BLE send queue
[1694778140022] [22643:294478] [SC] Sent PBKDF param request
[1694778140022] [22643:294478] [IN] Failed to establish BLE connection: src/platform/Darwin/BleConnectionDelegateImpl.mm:478: Ble Error 0x00000407: GATT write characteristic operation failed
[1694778140022] [22643:294478] [IN] Clearing BLE pending packets.
[1694778140022] [22643:294478] [BLE] Auto-closing end point's BLE connection.
[1694778157024] [22643:294480] [SC] PASESession timed out while waiting for a response from the peer. Expected message type was 33
[1694778157024] [22643:294480] [IN] SecureSession[0x600003cf4200]: Released - Type:1 LSID:42770
[1694778157024] [22643:294480] [TOO] Secure Pairing Failed
[1694778157024] [22643:294480] [TOO] Run command failure: examples/chip-tool/commands/pairing/PairingCommand.cpp:315: CHIP Error 0x00000003: Incorrect state
[1694778157024] [22643:294480] [DIS] Closing all BLE connections
[1694778157024] [22643:294480] [TOO] Pairing Failure: src/protocols/secure_channel/PASESession.cpp:255: CHIP Error 0x00000032: Timeout
[1694778157024] [22643:294480] [TOO] Run command failure: src/protocols/secure_channel/PASESession.cpp:255: CHIP Error 0x00000032: Timeout
>>> 
```

Any hints?

shubhamdp, 2023-09-18:
Please take a look at [Using chip-tool on macOS, or CHIP Tool on iOS](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/darwin.md#using-chip-tool-on-macos-or-chip-tool-on-ios)

richardklingler, 2023-09-18:
Yepp...that was it..thanks (o;

Now if there were only examples for sensors (o;

closed 2023-09-18 by richardklingler