espressif/esp-matter #639: Voice commands from Google Home app

leonardbay, 2023-09-15:
I build the light device Matter sample on a WROVER kit (Matter over Wi-Fi), then commissioned it with an iPad running Apple Home and with the Google Home app. Both Apple's and Google's UI commands work on the device. Voice commands issued to Siri work on the device. However, voice commands (on / off, dimming) issued to the Google Home app (and to a Google Nest Hub Max) fail to execute, the voice assistant reports that the device "does not support that functionality." Is this a known issue? Are there additional settings necessary to enable voice commands to work with Matter devices?

dhrishi, 2023-09-18:
@leonardbay 

> However, voice commands (on / off, dimming) issued to the Google Home app (and to a Google Nest Hub Max) fail to execute, the voice assistant reports that the device "does not support that functionality."

Ideally, if the device type is supported with Google, they should just work fine.

Can you please open an issue with Google (check here: https://github.com/google-home/sample-apps-for-matter-android) with the details of your setup (Nest Hub, phone, versions etc.) and close this one.

leonardbay, 2023-09-18:
@dhrishi

>Ideally, if the device type is supported with Google, they should just work fine.

It works fine with voice commands with Siri, and now also with Alexa. With Google the UI buttons work, but voice commands do not (in the screenshot below, "Brad" is the Matter device, the other lights have cloud integration with Actions on Google):

[image]

Have you tested the ESP-IDF light sample with the Google Home app, and does it work with voice commands?

I am hesitant to post on  https://github.com/google-home/sample-apps-for-matter-android, because the device also commissions and works well with the GHSAFM app, and the app does not support voice commands.

leonardbay, 2023-09-25:
It turns out the Google Home app and Google Hub Max 2 fail to relay voice commands to the last Matter device commissioned into the Google Home account. Once another (commercially available) Matter device was commissioned, both the Google Home app and Google Hub Max 2 started successfully relaying voice commands to our Matter over Thread sample device, and failed to relay voice commands to the newly commissioned 3rd party device. Hopefully this behavior will eventually be fixed.