espressif/esp-matter #641: Retrieve The New Generated Setup Passcode and Discriminator

anuarrozman, 2023-09-18:
**Retrieve The New Generated Setup Passcode and Discriminator**
Hi, Ive generated the unique pairing key using the mfg_tool.py. I know we can get the value of discriminator and passcode set from the generated .csv file, but is there a way for me to retrieve it from within the device itself ? Like where does it store and can i read the address memory to get the information i want.

shubhamdp, 2023-09-18:
@anuarrozman Sorry if the https://github.com/espressif/esp-matter/issues/612#issuecomment-1717203901 wasn't clear, but you should not have the passcode in the device. I'd suggest please read through the specification section mentioned in the comment.

If you still want to do that please implement [CommissionableDataProvider interface](https://github.com/project-chip/connectedhomeip/blob/master/src/include/platform/CommissionableDataProvider.h), we have one [implementation](https://github.com/project-chip/connectedhomeip/blob/master/src/platform/ESP32/ESP32FactoryDataProvider.h#L31)(which do not read passcode). And please check [this](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#custom-providers) section in ESP Matter programming guide on how to use the custom providers.

anuarrozman, 2023-09-18:
Oh, im sorry. Thank you for answering.

closed 2023-09-18 by anuarrozman

dhrishi, 2023-09-18:
@anuarrozman There's nothing to be sorry about :-)