espressif/esp-matter #642: Implementing Soft AP before Matter start

anuarrozman, 2023-09-18:
**Implementing Soft AP before Matter start **
Hi, Ive a question regarding on network commissioning, which required device to have network connection before do Matter stuff. Is it possible to implement Soft AP function before initialize Matter ? Can I use softAP example from esp-idf ? From my understanding, Matter will initialize NVS flash and check network credentials. It would be great if there's example or documentation that i can follow. Thank you

shubhamdp, 2023-10-04:
@anuarrozman You can surely start the AP and initialize it. Then I assume you will connect to this AP and provide it with network credentials and then once device connects to wifi you can start matter. And then the device advertises on the IP network.

As of now there is no guide/example, but this looks straight forward.

anuarrozman, 2023-10-06:
Thank you @shubhamdp , Ill have a look at it.

dhrishi, 2023-10-09:
Closing as the question is answered. Feel free to re-open in case of any issues

closed 2023-10-09 by dhrishi