espressif/esp-matter #643: Change Accessory and Endpoint Name

law-ko, 2023-09-18:
Hello,

How can we change the accessory and endpoint name?

Thank you.

jadhavrohit924, 2023-09-18:
@law-ko Could you please provide us more details? What are you trying to do?

law-ko, 2023-09-18:
@jadhavrohit924 When we pair the Matter device with Apple Home app, it will show the default device name as `Matter Accessory` and endpoints as default name such as `Outlet` `Light`, can we change these?

jadhavrohit924, 2023-09-18:
When you open the details of the device in the Apple Home app, you can edit the name of the device.

law-ko, 2023-09-18:
@jadhavrohit924 Yes, but we want to apply special names suitable for the product itself. HAP can do this by code. Can Matter change this by code as well?

jadhavrohit924, 2023-09-18:
@law-ko Apple Home App fetches these details from the DCL. 

> Can Matter change this by code as well?

No

jonsmirl, 2023-09-18:
You have to complain to Apple about this, this is not something you can control. The brand name is available in the API for Apple to use, Apple just isn't choosing to use it.

law-ko, 2023-09-19:
> @law-ko It totally depends on the commissioner. Some may use the information from the basic information cluster.

Are we able to change the basic information cluster name?

dhrishi, 2023-09-21:
Yes. You can change the basic information cluster attributes through using the mfg_tool. Please check mfg_tool help to find `Device instance information options`

law-ko, 2023-09-23:
@dhrishi Can you specify a link to this? Looking upon the matter mfg_tools [documentation](https://github.com/espressif/esp-matter/tree/release/v1.1/tools/mfg_tool) I can't find it.

Thank you.

dhrishi, 2023-09-23:
These are the parameters that can be configured through the mfg_tool just like the ones mentioned in the documentation. Can you please check `./mfg_tool.py --help` for their names.

law-ko, 2023-09-23:
@dhrishi Upon checking it does not allow us to change the specific endpoint names. I assume it is only referring to the whole device name:
```
Few more Basic clusters options:
  --product-label PRODUCT_LABEL                   Product label
  --product-url PRODUCT_URL                       Product URL
 ```
 
 I am seeking to see if it is possible to change the endpoint name: E.g. `Light` -> `Lamp`. Although it can be adjusted by the consumers, but we want to have the endpoint matching our product design when shipping to customer hands.

dhrishi, 2023-09-23:
> Are we able to change the basic information cluster name

I thought you wanted to change the basic information cluster attributes (product name)

> I am seeking to see if it is possible to change the endpoint name: E.g. Light -> Lamp.

As already mentioned above, the default names are populated by the commissioners from DCL. After the product is certified, you can add the name you want to be seen in the Model details of your product in DCL.

If you have any suggestions around it, I request you to take this forward with the ecosystems.

dhrishi, 2023-09-27:
Closing this as the question is answered

closed 2023-09-27 by dhrishi