espressif/esp-matter #644: Tag stable release of esp-matter repo

DCSBL, 2023-09-18:
**Describe the question/query that you have**
During initial setup, we found that it is not clear which commit is the 'stable'/reliable version is to use. We see some issues during `./install.sh`, but we want to make sure it is not our fault.

It would be nice if this repo uses tags (or releases). Or at least a guideline of which branch we can use.

Thank you 🙏

dhrishi, 2023-09-18:
@DCSBL As of now, there aren't any tags created. Current stable branch is [release/v1.1](https://github.com/espressif/esp-matter/tree/release/v1.1) which corresponds to the latest released Matter version v1.1. Please use that. Main branch corresponds to the currently being developed Matter version which is v1.2.

DCSBL, 2023-09-18:
Thank you for your fast reply! We will continue to look into that version.

Would it be possible to make this information you just gave publicly available (or easier to find, if already online). It could saved us some time 😇

dhrishi, 2023-09-18:
@DCSBL Sure. We will include it in the top-level README

closed 2023-09-18 by dhrishi

dhrishi, 2023-10-14:
>Would it be possible to make this information you just gave publicly available (or easier to find, if already online). It could saved us some time 😇

@DCSBL This has now been mentioned in the top-level README: https://github.com/espressif/esp-matter#supported-matter-specification-versions

jeevithkumarnm, 2024-04-25:
Hi dhrishi,

Could you start tagging esp-matter repo for releases? Or at least mention which commit ID to use from landing page?
We are finding hard time to get all the build/development systems to same level given that all branches keep moving always, including the release/v1.1.