espressif/esp-matter #645: Missing GetPrimaryMACAddress(MutableByteSpan buf) in ConfigurationManagerImpl.h

jzwyssig, 2023-09-19:
In the file `esp-matter/connectedhomeip/connectedhomeip/src/platform/ESP32/ConfigurationManagerImpl.h` only the function `CHIP_ERROR GetPrimaryWiFiMACAddress(uint8_t * buf) override;` is defined and as the name suggests support for Ethernet is missing.

Please add the generic definition `CHIP_ERROR GetPrimaryMACAddress(MutableByteSpan buf) override;` and provide a default implementation for Ethernet. I solved the problem by adding this definition and overriding the function in my main application with the correct MAC address of the used Ethernet chip.

**Environment**
- ESP-Matter Commit Id: https://github.com/espressif/esp-matter/commit/0a1d89d7235a5dc632edd488727814b9690b6270 (up to date)
- ESP-IDF Commit Id: e088c3766ba440e72268b458a68f27b6e7d63986 (5.1.1 stable)
- SoC: ESP32-S3
- Host Machine OS: Linux
- Device Log: E (9141) chip[DIS]: Failed to get primary mac address of device. Generating a random one.

PSONALl, 2023-09-22:
@jzwyssig Yes you are right, I am implementing GetPrimaryMACAddress for ethernet. I will let you know once it is available

dhrishi, 2023-10-13:
Update -
@jzwyssig This is merged in upstream now: https://github.com/project-chip/connectedhomeip/pull/29415
I will close this issue once the connectedhomeip submodule here is updated with the fix.

jzwyssig, 2023-10-13:
That is perfect, thank you @dhrishi.

closed 2023-11-29 by jzwyssig