espressif/esp-matter #648: Ethernet Matter supported?

ahmed8812, 2023-09-20:
Hi,

Does Esp Matter supports Matter over Ethernet?

Thanks

PSONALl, 2023-09-20:
Yes, Matter over Ethernet is supported
It can be enabled through CONFIG_ENABLE_ETHERNET_TELEMETRY config option

PSONALl, 2023-09-20:
For more details check: https://github.com/espressif/connectedhomeip/blob/ecc0d63cf7eb91f4017bf8c264b53cf690420eb5/docs/guides/esp32/build_app_and_commission.md#commissioning-the-ethernet-device-esp32-ethernet-kit

dhrishi, 2023-09-21:
I will close this as the question is answered. Please give it a try and let us know in case of any issues

closed 2023-09-21 by dhrishi

ahmed8812, 2023-10-03:
Hi,

I have the esp32 ethernet kit now. Tried example ble mesh bridge.

These are details:

Esp-matter: main - Commit 1cacb8b2fcd42c465e38317d3a63d0169460b397 
esp-idf: v5.1.1 - Commit e088c3766ba440e72268b458a68f27b6e7d63986 
esp-matter example: ble mesh bridge

ESP32S3: Wifi Telemetry
Total sizes: 
Used static IRAM:   88606 bytes ( 273634 remain, 24.5% used)
      .text size:   87579 bytes
   .vectors size:    1027 bytes
Used stat D/IRAM:   98044 bytes ( 247812 remain, 28.3% used)
      .data size:   21244 bytes
      .bss  size:   76800 bytes
Used Flash size : 1409563 bytes
           .text: 1120467 bytes
         .rodata:  288840 bytes
Total image size: 1519413 bytes (.bin may be padded larger)

ESP32: Ethernet Kit, Wifi Telemetry
Total sizes:
Used static DRAM:  104560 bytes (  20020 remain, 83.9% used)
      .data size:   23328 bytes
      .bss  size:   81232 bytes
Used static IRAM:  123362 bytes (   7710 remain, 94.1% used)
      .text size:  122335 bytes
   .vectors size:    1027 bytes
Used Flash size : 1467927 bytes
           .text: 1182839 bytes
         .rodata:  284832 bytes
Total image size: 1614617 bytes (.bin may be padded larger)

Why does the esp32 consume so much RAM whereas esp32s3 using the same out-of-box example it consumes way less RAM?

I tried using Ethernet Telemetry but compile failed. I attached the compile log for convenience

Appreciate feedback thanks 
[Ethernet compile error.txt](https://github.com/espressif/esp-matter/files/12789416/Ethernet.compile.error.txt)

PSONALl, 2023-10-04:
@ahmed8812 The support for the ethernet connectivity, is not currently present in blemesh platform
I will add the support soon