espressif/esp-matter #649: Request an example of a contact_sensor in the examples folder

Steven23456, 2023-09-20:
**Is your feature request related to a problem? Please describe.**
Using the given examples, I cannot figure out how to build the app_main or app_driver for a contact_sensor using the GPIO pins of an ESP32

**Describe the solution you'd like**
Just a simple example added to the examples folder that illustrates how to utilize the contact_sensor device type

**Describe alternatives you've considered**
I considered the button type, however it seems to be a client, where I need the contact_sensor to be a server

**Additional context**
None

PSONALl, 2023-09-21:
For contact sensor device type, you need to implement a Boolean State cluster. You can then implement a driver implementation around the Boolean State cluster's StateValue attribute, using the following convention:

FALSE = open or no contact
TRUE = closed or contact

PSONALl, 2023-09-21:
1. This means that your `app_driver_attribute_update_callback` will change the state of your sensor device whenever the StateValue is updated. And, some callback say for example `app_driver_sensor_state_change_cb` which will change the state whenever there is a change in contact

Example:

```
// app_driver_attribute_update callback
esp_err_t app_driver_attribute_update(app_driver_handle_t driver_handle, uint16_t endpoint_id, uint32_t cluster_id,
                   uint32_t attribute_id, esp_matter_attr_val_t *val)
{
  esp_err_t err = ESP_OK;
  if (endpoint_id == contact_sensor_endpoint_id) {
      // Implement your own handle
      contact_sensor_driver_handle_t handle= (contact_sensor_driver_handle_t )driver_handle;
    if (cluster_id == BooleanState::Id) {
      if (attribute_id == BooleanState::Attributes::StateValue::Id) {
        err = sensor_device_set_state(handle, val); 
  return err;
}

// app_driver_sensor_state_change_cb callback

static void app_driver_sensor_state_change_cb(void *arg, void *data)
{   
    ESP_LOGI(TAG, "Toggle button pressed");
    uint16_t endpoint_id = light_endpoint_id;
    uint32_t cluster_id = BooleanState::Id;
    uint32_t attribute_id = BooleanState::Attributes::StateValue::Id;
    
    node_t *node = node::get();
    endpoint_t *endpoint = endpoint::get(node, endpoint_id);
    cluster_t *cluster = cluster::get(endpoint, cluster_id);
    attribute_t *attribute = attribute::get(cluster, attribute_id);
    
    esp_matter_attr_val_t val = esp_matter_invalid(NULL);
    attribute::get_val(attribute, &val);
    val.val.b = !val.val.b;
    attribute::update(endpoint_id, cluster_id, attribute_id, &val);
}

```

2. To implement app_main for a contact sensor device, you can do the following:

- Initialize the contact sensor driver. Register `app_driver_sensor_state_change_cb` during initialisation under `app_driver_contact_sensor_init()`. The `contact_sensor_state_change_cb` callback is called whenever the contact sensor state changes. 
- Create a Matter node and add the mandatory Root Node device type on endpoint 0.
- Create a contact sensor device type and configure it with the desired initial state and contact sensor state change callback.
- Create an endpoint for the contact sensor device type and register it with the Matter node.
Here is an example:
```
app_driver_handle_t contact_sensor_handle = app_driver_contact_sensor_init();

/* Create a Matter node and add the mandatory Root Node device type on endpoint 0 */
node::config_t node_config;
node_t *node = node::create(&node_config, app_attribute_update_cb, NULL);

contact_sensor::config_t contact_sensor_config;
contact_sensor_config.state = CONTACT_SENSOR_STATE_OPEN;

endpoint_t *endpoint = contact_sensor::create(node, &contact_sensor_config, ENDPOINT_FLAG_NONE, contact_sensor_handle);
```

Steven23456, 2023-09-22:
@PSONALl  I must have missed something in the documentation or something .. I can't figure out how these examples are built .. (short of looking at idf.py code) .. what causes the contact_sensor types / includes to be added to the projects?  I have no idea how to build the "app_driver_contact_sensor_init()" function ...

PSONALl, 2023-09-26:
@Steven23456  Please provide more information about your contact sensor device.
Please provide an image of the device, this will help us to understand the device's physical layout and connections.

Steven23456, 2023-09-26:
Just a simple contact wired to an ESP32 (GPIO 2 in this example) where that state of the contact (open/closed) is sent back to the matter hub when its state changes..
[image]

PSONALl, 2023-09-27:
You can initialize the driver by setting the GPIO input pin to 2 in `app_driver_contact_sensor_init`
and whenever the state changes send the callback to the application when the state changes

PSONALl, 2023-09-27:
You can implement a simple boolean contact sensor and register positive edge interrupt handler for the same like below

```
// Initialize the contact sensor driver
#define CONTACT_SENSOR_GPIO 2
app_driver_handle_t app_driver_contact_sensor_init() {
    gpio_config_t io_conf;
    io_conf.intr_type = GPIO_INTR_POSEDGE; // Positive edge-triggered interrupt 
    io_conf.pin_bit_mask = (1 << CONTACT_SENSOR_GPIO);
    io_conf.mode = GPIO_MODE_INPUT;
    io_conf.pull_up_en = GPIO_PULLUP_DISABLE;
    io_conf.pull_down_en = GPIO_PULLDOWN_DISABLE;
    gpio_config(&io_conf);

    // Register the positive edge-triggered interrupt callback
    gpio_isr_handler_add(CONTACT_SENSOR_GPIO, app_driver_contact_sensor_state_change, (void*)CONTACT_SENSOR_GPIO);
}
```
you can refer this example in esp-idf for more details: https://github.com/espressif/esp-idf/tree/master/examples/peripherals/gpio/generic_gpio

dhrishi, 2023-10-14:
Closing this as the reference is provided.

closed 2023-10-14 by dhrishi

FloBer431, 2024-02-05:
@Steven23456 Have you managed to get the contact sensor running? I am struggeling with the same issue and don't get it running, even with the provided help of @PSONALl

Steven23456, 2024-03-01:
> @Steven23456 Have you managed to get the contact sensor running? I am struggeling with the same issue and don't get it running, even with the provided help of @PSONALl 

I've not had time.