espressif/esp-matter #651: chip-tool after fresh install using Espressif SDK for Matter guide

DrewOJensen, 2023-09-22:
**Describe the bug**
After having some issues when trying to build a matter example from esp-matter, I decided that I should see if there were any updates. Lo and behold, In the guide, ESPIDF 5.1.1 was recommended. So I decided to wipe the folders and .espressif folder on my Ubuntu dev computer. A fresh re-install completed without any errors or warnings, I was able to build Matter examples from esp-matter and flash them without an issue. When I went to try and commission them using chip-tool I got the following error when I tried to use the terminal command "chip-tool"

`crazyharry@CrazyHarry:~/esp/esp-matter/connectedhomeip/connectedhomeip$ chip-tool
[1695349578.199856][74627:74627] CHIP:CTL: Init Storage failure: src/controller/ExamplePersistentStorage.cpp:70: Error 0x000000AD`

I'm not sure what this error is so if anyone can provide direction that would be greatly appreciated. I came to report here since I hadn't seen this issue in previous IDF & Matter releases when I attempted to use chip-tool.

**Environment**
- ESP-Matter Commit Id: git log shows the following
`commit cd325c6fa4203c45707ecb8b5351b7303347acb7 (HEAD -> main, origin/main, origin/HEAD)
Merge: 499eadd 6cbc342
Author: Hrishikesh Dhayagude <hrishi@espressif.com>
Date:   Thu Sep 21 14:25:47 2023 +0800`

- ESP-IDF Commit Id: git log shows the following
`commit e088c3766ba440e72268b458a68f27b6e7d63986 (HEAD, tag: v5.1.1)
Author: Aditya Patwardhan <aditya.patwardhan@espressif.com>
Date:   Mon Aug 21 14:24:40 2023 +0550

    versions: Update version to 5.1.1`

- SoC (eg: ESP32 or ESP32-C3): ESP32-S3 & ESP32-C6
- Device Logs (Please attach the log file): Not sure where to get these
- Host Machine OS: Ubuntu 22.04
- Commissioner app and versions if present: NA
- Commissioner's logs if present: NA

**Any additional details**
...

shubhamdp, 2023-09-25:
`Error 0x000000AD` means [open failed](https://github.com/project-chip/connectedhomeip/blob/master/src/lib/core/CHIPError.h#L1424). Can you try again removing the earlier chip-tool storage, `rm -rf /tmp/chip*`.

It could be a file system permission issue. You can use the `--storage-directory` option with chip-tool and specify a custom directory as well.
> [--storage-directory]:
  Directory to place chip-tool's storage files in.  Defaults to $TMPDIR, with fallback to /tmp

DrewOJensen, 2023-09-27:
@shubhamdp 

You were right about it being a permissions issue. For my Ubuntu 22.04 install, as soon as I used sudo to initiate chip-tool
`sudo chip-tool interactive start`
everything so far appears to be working, although I have yet to use any chip-tool commands other than the above since I won't be attempting any commissioning until tomorrow. 

Again, thanks @shubhamdp for the tip on it possibly being permissions.

closed 2023-09-27 by DrewOJensen