espressif/esp-matter #652: Commissioning Window Getting Closed

SYESHASWINI, 2023-09-22:
I’m trying to flash partition.bin generated by mfg_tool to the project. 

Using **Matter version: release/v1.0
          Idf version: ESP-IDF v4.4.3**

Followed the [doc](https://github.com/espressif/esp-matter/blob/release/v1.0/tools/mfg_tool/README.md) for generating the bin file and flashing it to the specific address.

When enabled this option in the Menuconfig closes the commissioning window
> 
> Enable ESP32 Factory Data Provider [Component config → CHIP Device Layer → Commissioning options → Use ESP32 Factory Data Provider]
> Enable config option [CONFIG_ENABLE_ESP32_FACTORY_DATA_PROVIDER](https://github.com/project-chip/connectedhomeip/blob/master/config/esp32/components/chip/Kconfig#L645) to use ESP32 specific implementation of CommissionableDataProvider and DeviceAttestationCredentialsProvider.

Attaching the Logs:
[Factory_cert.txt](https://github.com/espressif/esp-matter/files/12696835/Factory_cert.txt)

shripad621git, 2023-09-22:
1. From the logs I am able to figure out that you are using the generic_switch example.
2. The generic switch example is designed for demonstrating the use of fixed labels and already contains a pre generated factory_partiton binary for use.
3. If you just want to try out the working of factory_partiton you can follow the README of the generic switch example.You need to enable `CONFIG_ENABLE_ESP32_DEVICE_INFO_PROVIDER` in addition to `CONFIG_ENABLE_ESP32_FACTORY_DATA_PROVIDER` as mentioned in the README even if you want to flash your own partition for the generic_switch example.
4. In order to get more understanding on factory_parttiton ,you can refer [this](https://github.com/espressif/connectedhomeip/blob/4088a77f557e8571a39338fad51a1d8eb0131d79/docs/guides/esp32/factory_data.md) link.

SYESHASWINI, 2023-09-22:
@shripad621git I have also enabled **CONFIG_ENABLE_ESP32_DEVICE_INFO_PROVIDER** 
but the commissioning window is still closed.

shripad621git, 2023-09-29:
@SYESHASWINI ,I have tried out the generic_switch example by flashing the factory partition provided in the example as well as by creating a new `factory_partition `using `mfg_tool.py` and both were successful.Please share your` sdkconfig file `and `factory_partition binary command` you used to generate the partition.

dhrishi, 2023-10-13:
@SYESHASWINI It should ideally work fine. Closing this now. Please re-open if you see the issue, with the requested information

closed 2023-10-13 by dhrishi