espressif/esp-matter #653: Change initial Attribute values while Node creation

PhLuReh, 2023-09-22:
While creating a new Matter node (taken from the examples), some mandatory Clusters get created along with there Attributes. Currently, the only Attribute which can be configured to be initialized is the `root_node.basic_information.node_label`. I did it like so and this works.
```
    node::config_t node_config;
    snprintf(node_config.root_node.basic_information.node_label, sizeof(node_config.root_node.basic_information.node_label), "Cool Product");
    node_t *node = node::create(&node_config, app_attribute_update_cb, app_identification_cb);
```

To be more precise. The testcase [TC_BINFO_1_1](https://github.com/espressif/connectedhomeip/blob/v1.1.0.1/src/app/tests/suites/certification/Test_TC_BINFO_1_1.yaml) from the official test suite reads the AttributeList of the `Basic Information Cluster` and checks if all the attributes in the list `[
                      0,
                      1,
                      2,
                      3,
                      4,
                      5,
                      6,
                      7,
                      8,
                      9,
                      10,
                      19,
                      65528,
                      65529,
                      65531,
                      65532,
                      65533,
                  ]` are reported. And for example attribute `HardwareVersionString` with the id `0x0008` fails during creation of the Matter Node. See Log:
```
[0;31mE (980) esp_matter_attribute: Could not create attribute, string length out of bound[0m
```

I also digged into the code, which obviously has to fail because of using `NULL` and `0` length when [initializing the attribute](https://github.com/espressif/esp-matter/blob/11121a5b0a51838409c937162019b1c88900e00e/components/esp_matter/esp_matter_cluster.cpp#L201C34-L201C34). Resulting into a call of the following, which has to fail with parameter `length = 0`
```
attribute_t *create_hardware_version_string(cluster_t *cluster, char *value, uint16_t length)
{
    if (length < k_min_version_string_length || length > k_max_version_string_length) {
        ESP_LOGE(TAG, "Could not create attribute, string length out of bound");
        return NULL;
    }
    return esp_matter::attribute::create(cluster, BasicInformation::Attributes::HardwareVersionString::Id,
                                         ATTRIBUTE_FLAG_NONE, esp_matter_char_str(value, length));
}
```

**So** please provide a way how the attributes are intended to be set. Or should the be an improvement to extend the corresponding `config_t` structs?

PSONALl, 2023-09-22:
@PhLuReh Please use the latest main branch commit for the test, It has fixed this issue : [6aa7601](https://github.com/espressif/esp-matter/commit/6aa7601258b10eaa0cb3845c39f41109ac0bfbaf)

jonsmirl, 2023-09-24:
Isn't the HW Version string supposed to be set via mfg-tool?
~~~
Device instance information options:
  -v VENDOR_ID, --vendor-id VENDOR_ID             Vendor id
  --vendor-name VENDOR_NAME                       Vendor name
  -p PRODUCT_ID, --product-id PRODUCT_ID          Product id
  --product-name PRODUCT_NAME                     Product name
  --hw-ver HW_VER                                 Hardware version
  --hw-ver-str HW_VER_STR                         Hardware version string
  --mfg-date MFG_DATE                             Manufacturing date in format YYYY-MM-DD
  --serial-num SERIAL_NUM                         Serial number
  --enable-rotating-device-id                     Enable Rotating device id in the generated binaries
  --rd-id-uid RD_ID_UID                           128-bit unique identifier for generating rotating device identifier,
                                                  provide 32-byte hex string, e.g. "1234567890abcdef1234567890abcdef"
~~~

PhLuReh, 2023-09-25:
@PSONALl Now, I rebuilt my [docker containter](https://hub.docker.com/layers/espressif/esp-matter/latest_idf_v5.1.1/images/sha256-d83e157e0c6d9a2c5f0dc24dbfa2f44ad67ba0b61ce90651595c47d2bcc8b3b1?context=explore) and it shows IDF commit e088c3766ba440e72268b458a68f27b6e7d63986 and Matter SDK commit 6aa7601258b10eaa0cb3845c39f41109ac0bfbaf. I tried to build, but it fails with
```
In file included from ../../../../../../../opt/espressif/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/app/DefaultAttributePersistenceProvider.h:19,
                 from ../../../../../../../opt/espressif/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/app/DefaultAttributePersistenceProvider.cpp:17:
../../../../../../../opt/espressif/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/app/SafeAttributePersistenceProvider.h: In member function 'CHIP_ERROR chip::app::SafeAttributePersistenceProvider::WriteScalarValue(const chip::app::ConcreteAttributePath&, T&)':
../../../../../../../opt/espressif/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/app/SafeAttributePersistenceProvider.h:57:12: error: 'if constexpr' only available with '-std=c++17' or '-std=gnu++17' [-Werror=c++17-extensions]
   57 |         if constexpr (std::is_signed_v<T>)
      |            ^~~~~~~~~
../../../../../../../opt/espressif/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/app/SafeAttributePersistenceProvider.h:57:28: error: 'is_signed_v' is not a member of 'std'; did you mean 'is_signed'?
   57 |         if constexpr (std::is_signed_v<T>)
      |                            ^~~~~~~~~~~
      |                            is_signed
../../../../../../../opt/espressif/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/app/SafeAttributePersistenceProvider.h:57:41: error: expected primary-expression before '>' token
   57 |         if constexpr (std::is_signed_v<T>)
      |                                         ^
../../../../../../../opt/espressif/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/app/SafeAttributePersistenceProvider.h:57:42: error: expected primary-expression before ')' token
   57 |         if constexpr (std::is_signed_v<T>)
      |                                          ^
```

I think it is a version conflict between IDF and Matter SDK, but I thought this was the purpose of having a docker container. I wouldn't like to switch branch/tag away from `v5.1.1`.

Hoping for any advice...

PhLuReh, 2023-09-25:
I now did a `idf.py fullclean` but it couldn't compile without errors:
```
../../../../../../../opt/espressif/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/nimble/BLEManagerImpl.cpp: In member function 'CHIP_ERROR chip::DeviceLayer::Internal::BLEManagerImpl::ConfigureScanResponseData(chip::ByteSpan)':
../../../../../../../opt/espressif/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/nimble/BLEManagerImpl.cpp:1009:35: error: missing template arguments before '(' token
 1009 |     mScanResponse = chip::Optional(scanResponseSpan);
      |                                   ^
```

jonsmirl, 2023-09-25:
Add this to your CMakeLists.txt
~~~
idf_build_set_property(CXX_COMPILE_OPTIONS "-std=gnu++17;-Os;-DCHIP_HAVE_CONFIG_H;-fpermissive" APPEND)
~~~

PhLuReh, 2023-09-25:
@jonsmirl this seems to have solved the `chip::Optional` problem, but not the one from comment https://github.com/espressif/esp-matter/issues/653#issuecomment-1733348449

I tried to figure out, but didn't find the reason, why there is still a `-std=gnu++14` inside the compiler options:
```
[1218/1230] Building CXX object esp-idf/main/CMakeFiles/__idf_main.dir/app_main.cpp.objFAILED: esp-idf/main/CMakeFiles/__idf_main.dir/app_main.cpp.obj
/home/smart/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -DESP_PLATFORM -DIDF_VER=\"v5.1.1\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/home/smart/projects/smp_matter/build/config -I/home/smart/projects/smp_matter/main -I/opt/espressif/esp-idf/components/newlib/platform_include -I/opt/espressif/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/opt/espressif/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/opt/espressif/esp-idf/components/freertos/esp_additions/include/freertos -I/opt/espressif/esp-idf/components/freertos/esp_additions/include -I/opt/espressif/esp-idf/components/freertos/esp_additions/arch/xtensa/include -I/opt/espressif/esp-idf/components/esp_hw_support/include -I/opt/espressif/esp-idf/components/esp_hw_support/include/soc -I/opt/espressif/esp-idf/components/esp_hw_support/include/soc/esp32 -I/opt/espressif/esp-idf/components/esp_hw_support/port/esp32/. -I/opt/espressif/esp-idf/components/esp_hw_support/port/esp32/private_include -I/opt/espressif/esp-idf/components/heap/include -I/opt/espressif/esp-idf/components/log/include -I/opt/espressif/esp-idf/components/soc/include -I/opt/espressif/esp-idf/components/soc/esp32 -I/opt/espressif/esp-idf/components/soc/esp32/include -I/opt/espressif/esp-idf/components/hal/esp32/include -I/opt/espressif/esp-idf/components/hal/include -I/opt/espressif/esp-idf/components/hal/platform_port/include -I/opt/espressif/esp-idf/components/esp_rom/include -I/opt/espressif/esp-idf/components/esp_rom/include/esp32 -I/opt/espressif/esp-idf/components/esp_rom/esp32 -I/opt/espressif/esp-idf/components/esp_common/include -I/opt/espressif/esp-idf/components/esp_system/include -I/opt/espressif/esp-idf/components/esp_system/port/soc -I/opt/espressif/esp-idf/components/esp_system/port/include/private -I/opt/espressif/esp-idf/components/xtensa/include -I/opt/espressif/esp-idf/components/xtensa/esp32/include -I/opt/espressif/esp-idf/components/lwip/include -I/opt/espressif/esp-idf/components/lwip/include/apps -I/opt/espressif/esp-idf/components/lwip/include/apps/sntp -I/opt/espressif/esp-idf/components/lwip/lwip/src/include -I/opt/espressif/esp-idf/components/lwip/port/include -I/opt/espressif/esp-idf/components/lwip/port/freertos/include -I/opt/espressif/esp-idf/components/lwip/port/esp32xx/include -I/opt/espressif/esp-idf/components/lwip/port/esp32xx/include/arch -I/opt/espressif/esp-matter/device_hal/device/include -I/opt/espressif/esp-matter/device_hal/led_driver/include -I/opt/espressif/esp-idf/components/driver/include -I/opt/espressif/esp-idf/components/driver/deprecated -I/opt/espressif/esp-idf/components/driver/analog_comparator/include -I/opt/espressif/esp-idf/components/driver/dac/include -I/opt/espressif/esp-idf/components/driver/gpio/include -I/opt/espressif/esp-idf/components/driver/gptimer/include -I/opt/espressif/esp-idf/components/driver/i2c/include -I/opt/espressif/esp-idf/components/driver/i2s/include -I/opt/espressif/esp-idf/components/driver/ledc/include -I/opt/espressif/esp-idf/components/driver/mcpwm/include -I/opt/espressif/esp-idf/components/driver/parlio/include -I/opt/espressif/esp-idf/components/driver/pcnt/include -I/opt/espressif/esp-idf/components/driver/rmt/include -I/opt/espressif/esp-idf/components/driver/sdio_slave/include -I/opt/espressif/esp-idf/components/driver/sdmmc/include -I/opt/espressif/esp-idf/components/driver/sigma_delta/include -I/opt/espressif/esp-idf/components/driver/spi/include -I/opt/espressif/esp-idf/components/driver/temperature_sensor/include -I/opt/espressif/esp-idf/components/driver/touch_sensor/include -I/opt/espressif/esp-idf/components/driver/twai/include -I/opt/espressif/esp-idf/components/driver/uart/include -I/opt/espressif/esp-idf/components/driver/usb_serial_jtag/include -I/opt/espressif/esp-idf/components/driver/touch_sensor/esp32/include -I/opt/espressif/esp-idf/components/esp_pm/include -I/opt/espressif/esp-idf/components/esp_ringbuf/include -I/home/smart/projects/smp_matter/managed_components/espressif__button/include -I/opt/espressif/esp-idf/components/esp_adc/include -I/opt/espressif/esp-idf/components/esp_adc/interface -I/opt/espressif/esp-idf/components/esp_adc/esp32/include -I/opt/espressif/esp-idf/components/esp_adc/deprecated/include -I/opt/espressif/esp-matter/components/esp_matter -I/opt/espressif/esp-matter/connectedhomeip/connectedhomeip/zzz_generated/app-common -I/opt/espressif/esp-matter/connectedhomeip/connectedhomeip/third_party/nlfaultinjection/include -I/opt/espressif/esp-matter/connectedhomeip/connectedhomeip/src -I/opt/espressif/esp-matter/components/esp_matter/zap_common -I/opt/espressif/esp-matter/connectedhomeip/connectedhomeip/src/platform/ESP32 -I/opt/espressif/esp-matter/connectedhomeip/connectedhomeip/src/platform/OpenThread -I/opt/espressif/esp-matter/connectedhomeip/connectedhomeip/src/include -I/opt/espressif/esp-matter/connectedhomeip/connectedhomeip/src/lib -I/opt/esp [... cut ...]
      |                                          ^
[1220/1230] Building CXX object esp-idf/main/CMakeFiles/__idf_main.dir/app_task.cpp.obj/home/smart/projects/smp_matter/main/app_task.cpp: In function 'void app_loop(uint16_t)':
/home/smart/projects/smp_matter/main/app_task.cpp:46:10: warning: variable 'updateIdentification' set but not used [-Wunused-but-set-variable]
   46 |     bool updateIdentification = false;
      |          ^~~~~~~~~~~~~~~~~~~~
```

Is there any convenient way to find the source?

PhLuReh, 2023-09-25:
In lack of more knowledge on cmake, I tried to find some code which could be included by the following commands:
```
grep -Rnw '/opt/espressif/esp-matter' -e 'gnu++14'
grep -Rnw '/opt/espressif/esp-idf/' -e 'gnu++14'
```

but only found:
```
/opt/espressif/esp-matter/connectedhomeip/connectedhomeip/config/zephyr/zephyr-util.cmake
```

PSONALl, 2023-09-26:
@PhLuReh You should also add C++ 17 in CMakelists.txt under main
```
set_property(TARGET ${COMPONENT_LIB} PROPERTY CXX_STANDARD 17)
```
reference: https://github.com/espressif/esp-matter/blob/main/examples/light/main/CMakeLists.txt

dhrishi, 2023-09-30:
Please close the issue if resolved

PhLuReh, 2023-10-02:
Source is compiling and running now. But so far I couldn't run the test case again. I'm still waiting for reply from the CSA to get a working test bench. So I couldn't really answer the initial question, on the [TC_BINFO_1_1](https://github.com/espressif/connectedhomeip/blob/v1.1.0.1/src/app/tests/suites/certification/Test_TC_BINFO_1_1.yaml)) and thus wasn't closing the issue.

dhrishi, 2023-10-13:
@PhLuReh Do you have any update here?

PhLuReh, 2023-10-13:
Unfortunately not, sent three requests to different points at the CSA and no response so far...

dhrishi, 2023-11-06:
I am closing this now. You can re-open, if you receive an response that needs assistance

closed 2023-11-06 by dhrishi

PhLuReh, 2023-11-09:
I do have access to a Matter TH (v. Fall 2023) now and can confirm, that the TC-BINFO_1_1 runs successfully. Thanks for the updates... Tests were running on d162b1357ce2d10a7971c434105633da0c4f3f11.

tanishqmanuja, 2024-05-19:
This trick actually works, and the name even shows on alexa app !!!
```
snprintf(node_config.root_node.basic_information.node_label, sizeof(node_config.root_node.basic_information.node_label), "Cool Product");
```

Can we somehow do the same with mfg_tool instead of hardcoding in app_main.cpp ?