espressif/esp-matter #654: Wrong responses from ESP in TC_IDM_1_2

PhLuReh, 2023-09-22:
**Describe the bug**
Device was tested at the ATL using [TC_IDM_1_2](https://github.com/espressif/connectedhomeip/blob/v1.1.0.1/src/app/tests/suites/certification/Test_TC_IDM_1_2.yaml). Issuing the following commands showed some invalid behavior:

- `./chip-tool any command-by-id 0x0003 0 1 1 1` fails with timeout (and ESP logs “Failed to send command response: 3” see supplied log)
- `./chip-tool any command-by-id 0x0003 3 1 1 0` fails with unsupported cluster (0x0003 is IdentifyCluster), but TC expects unsupported command, where `3` should be unsupported.

**Environment**
- ESP-Matter Commit Id: 0a1d89d7235a5dc632edd488727814b9690b6270
- ESP-IDF Commit Id: e088c3766ba440e72268b458a68f27b6e7d63986
- SoC (eg: ESP32 or ESP32-C3): ESP32
- Device Logs (Please attach the log file): [log_TC_IDM_1_2.txt](https://github.com/espressif/esp-matter/files/12698348/log_TC_IDM_1_2.txt)
- Host Machine OS: Matter TH v_2.8.1_official
- Commissioner app and versions if present:
- Commissioner's logs if present: [TC-IDM-1_2_2023_09_20_16_16_36.log](https://github.com/espressif/esp-matter/files/12698365/TC-IDM-1_2_2023_09_20_16_16_36.log)

**Any additional details**
...

VaishaliAvhale, 2023-09-22:
Are you executing SVE fall release 2023 test cases?

PSONALl, 2023-09-22:
@PhLuReh which device type are you using?
This test case expects you use  specific cluster that is unsupported for your device in first case
In second case it expects to use specific command that is unsupported for a cluster that is supported for your device type
reference: https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/interactiondatamodel.adoc#test-procedure-1

PhLuReh, 2023-09-22:
> Are you executing SVE fall release 2023 test cases?

I can't really answer this question. The ATL uses TH v_2.8.1_official.

PhLuReh, 2023-09-22:
> @PhLuReh which device type are you using? This test case expects you use specific cluster that is unsupported for your device in first case In second case it expects to use specific command that is [...]

The device is a WindowCovering device. Cluster 0x3 is Identify which is supported... So you mean the test case was executed incorrectly?

Sorry, the link guides me to a 404...

VaishaliAvhale, 2023-09-22:
The TC_IDM-1.2 needs to be executed inside Docker because it is a Python script test case. Please pull the latest ESP-Matter changes and retest it.

PhLuReh, 2023-09-22:
@VaishaliAvhale I didn't run these tests on my own. The tests were run at an CSA-certified ATL. I think they have done the test routines as required by CSA. Or where do you get the information from, that the execution was done wrong?

According to ESP-Matter changes, I was using commit 0a1d89d7235a5dc632edd488727814b9690b6270 for the tests. Have there been changes? I mean, when are we able to pass tests with esp-matter, if there is currently no release for passing official certification tests and I need to build against `latest`? Or is there any?

VaishaliAvhale, 2023-09-22:
./chip-tool any command-by-id 0x0003 0 1 1 1
Above command is incorrect because we have to check unsupported behaviour which we will get on endpoint 2. The correct command provided below
./chip-tool any command-by-id 0x0003 0 1 1 2

./chip-tool any command-by-id 0x0003 3 1 1 0  >> This command should work please use latest esp-matter commit and try again.

dhrishi, 2023-09-30:
@PhLuReh Please close the issue if resolved. We were able to pass the testcase during the SVE execution

dhrishi, 2023-11-06:
Closing this now. Please feel free to re-open if you still see this

closed 2023-11-06 by dhrishi

PhLuReh, 2023-11-09:
The TC_IDM_1_2 seems to be gone in Fall 2023 release of the Matter TH. So I cannot finally say, weather the test case is succeeding or not.