espressif/esp-matter #655: Matter QRCode Label Generator

law-ko, 2023-09-23:
Hello,

There is a pretty Matter commissioner QRCode label provided in the development guide. Is there any tools that can be used to generate this label? (not to be confused with generating QRCode itself)
[image]

dhavalgujar, 2023-10-04:
Hi @law-ko,
We are working to make the tool available publicly.

It will ideally be available in the next week, I will provide a pointer to it here.

anuarrozman, 2023-10-19:
@dhavalgujar great news, im waiting for it too. Thank you in advance.

Pips801, 2023-12-15:
> Hi @law-ko,
> 
> We are working to make the tool available publicly.
> 
> 
> 
> It will ideally be available in the next week, I will provide a pointer to it here.

Hello, it is several months later. Has this tool been made available yet?

law-ko, 2024-02-26:
@dhavalgujar

> Hi @law-ko, We are working to make the tool available publicly.
> 
> It will ideally be available in the next week, I will provide a pointer to it here.

@dhavalgujar Any updates?