espressif/esp-matter #656: Failed to post event to CHIP Platform event queue

jonsmirl, 2023-09-24:
I can go make this queue larger, but the off-the-shelf light example hits this from current git.  So the default in the sample is not large enough.
~~~
I (291305) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000006's Attribute 0x00000000 is 1 **********
I (291315) chip[ZCL]: Endpoint 1 On/off already set to new value
I (291315) esp_matter_attribute: ********** W : Endpoint 0x0001's Cluster 0x00000008's Attribute 0x00000001 is 0 **********
I (291335) chip[EM]: >>> [E:30834i S:26609 M:21664930 (Ack:150559259)] (S) Msg RX from 2:AB0DE00000000000 [1443] --- Type 0001:01 (IM:StatusResponse)
I (291355) chip[IM]: Received status response, status is 0x00
I (291355) chip[EM]: <<< [E:30834i S:26609 M:150559260 (Ack:21664930)] (S) Msg TX to 2:AB0DE00000000000 [1443] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (291375) chip[IN]: (S) Sending msg 150559260 on secure session with LSID: 26609
I (292225) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000005's Attribute 0x00000006 is 0 **********
I (292225) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000005's Attribute 0x00000003 is 0 **********
I (292245) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000008's Attribute 0x00000000 is 156 **********
I (292255) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000008's Attribute 0x00000001 is 0 **********
I (292265) chip[EM]: <<< [E:30835i S:26609 M:150559261] (S) Msg TX to 2:AB0DE00000000000 [1443] --- Type 0001:05 (IM:ReportData)
I (292285) chip[IN]: (S) Sending msg 150559261 on secure session with LSID: 26609
I (292325) chip[EM]: >>> [E:30835i S:26609 M:21664931 (Ack:150559261)] (S) Msg RX from 2:AB0DE00000000000 [1443] --- Type 0001:01 (IM:StatusResponse)
I (292335) chip[IM]: Received status response, status is 0x00
I (292345) chip[EM]: <<< [E:30835i S:26609 M:150559262 (Ack:21664931)] (S) Msg TX to 2:AB0DE00000000000 [1443] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (292355) chip[IN]: (S) Sending msg 150559262 on secure session with LSID: 26609
I (297315) wifi:new:<5,2>, old:<5,0>, ap:<255,255>, sta:<5,2>, prof:1
I (302295) chip[EM]: <<< [E:30836i S:26609 M:150559263] (S) Msg TX to 2:AB0DE00000000000 [1443] --- Type 0001:05 (IM:ReportData)
I (302295) chip[IN]: (S) Sending msg 150559263 on secure session with LSID: 26609
I (302345) chip[EM]: >>> [E:30836i S:26609 M:21664932 (Ack:150559263)] (S) Msg RX from 2:AB0DE00000000000 [1443] --- Type 0001:01 (IM:StatusResponse)
I (302355) chip[IM]: Received status response, status is 0x00
I (302365) chip[EM]: <<< [E:30836i S:26609 M:150559264 (Ack:21664932)] (S) Msg TX to 2:AB0DE00000000000 [1443] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (302375) chip[IN]: (S) Sending msg 150559264 on secure session with LSID: 26609
E (306145) chip[DL]: Failed to post event to CHIP Platform event queue
E (306145) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
E (306155) chip[DL]: Failed to post event to CHIP Platform event queue
E (306155) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000

~~~

jonsmirl, 2023-10-05:
I am still getting this one from light bulb sample. I have PKT BUFFER in PSRAM.  
I am using default config file so the default queue is too small.
~~~
I (124268139) ROUTE_HOOK: prefix FD6F:48B9:FE1A:1:: lifetime 1800
I (124268139) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: 472DC37BCBB11443-AB0DE00000000019.
E (124268159) chip[DL]: Failed to post event to CHIP Platform event queue
E (124268159) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
E (124268179) chip[DL]: Failed to post event to CHIP Platform event queue
E (124268179) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
E (124268199) chip[DL]: Failed to post event to CHIP Platform event queue
E (124268199) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
E (124268219) chip[DL]: Failed to post event to CHIP Platform event queue
E (124268219) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
E (124268239) chip[DL]: Failed to post event to CHIP Platform event queue
E (124268239) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
E (124268259) chip[DL]: Failed to post event to CHIP Platform event queue
E (124268259) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
I (124268269) chip[DIS]: mDNS service published: _matter._tcp
I (124268279) app_main: Interface IP Address changed
I (124268279) chip[DIS]: Updating services using commissioning mode 0
E (124268299) chip[DL]: Failed to post event to CHIP Platform event queue
E (124268299) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
E (124268309) chip[IN]: Unable to schedule lambda: 1000000
I (124268319) chip[DIS]: CHIP minimal mDNS started advertising.
I (124268329) chip[DIS]: Advertise operational node AF67F99EAFC7087E-F6FC071EEC50DFC9
I (124268339) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: AF67F99EAFC7087E-F6FC071EEC50DFC9.E (124268349) chip[DL]: 
Failed to post event to CHIP Platform event queue
E (124268349) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
E (124268369) chip[DL]: Failed to post event to CHIP Platform event queue
E (124268369) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
E (124268389) chip[DL]: Failed to post event to CHIP Platform event queue
E (124268389) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
E (124268409) chip[DL]: Failed to post event to CHIP Platform event queue
E (124268409) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
E (124268429) chip[DL]: Failed to post event to CHIP Platform event queue
E (124268429) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
~~~

jonsmirl, 2023-10-07:
New variation, it adds:
~~~
E (13952179) chip[DL]: Failed to schedule work: 1000000
E (13952179) chip[SC]: Failed to Schedule the AfterWorkCallback on foreground thread: 1000000
~~~

~~~
 (13951209) chip[EM]: >>> [E:29410r S:0 M:117847726 (Ack:217475601)] (U) Msg RX from 0:E8AC413FD1EE32CB [0000] --- Type 0000:32 (SecureChannel:CASE_Sigma3)
I (13951229) chip[EM]: <<< [E:29410r S:0 M:217475602 (Ack:117847726)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (13951239) chip[IN]: (U) Sending msg 217475602 to IP address 'UDP:192.168.1.206%st1:54933'
I (13951249) chip[SC]: Received Sigma3 msg
E (13951249) chip[SC]: The device does not support GetClock_RealTimeMS() API: 6c.  Falling back to Last Known Good UTC Time
E (13951739) chip[DL]: Failed to post event to CHIP Platform event queue
E (13951739) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
E (13951749) chip[DL]: Failed to post event to CHIP Platform event queue
E (13951759) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
E (13951769) chip[DL]: Failed to post event to CHIP Platform event queue
E (13951769) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
E (13951789) chip[DL]: Failed to post event to CHIP Platform event queue
E (13951789) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
E (13951809) chip[DL]: Failed to post event to CHIP Platform event queue
E (13951809) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
E (13951829) chip[DL]: Failed to post event to CHIP Platform event queue
E (13951829) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
E (13952059) chip[DL]: Failed to post event to CHIP Platform event queue
E (13952059) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
E (13952079) chip[DL]: Failed to post event to CHIP Platform event queue
E (13952079) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
E (13952099) chip[DL]: Failed to post event to CHIP Platform event queue
E (13952099) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
E (13952119) chip[DL]: Failed to post event to CHIP Platform event queue
E (13952119) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
E (13952139) chip[DL]: Failed to post event to CHIP Platform event queue
E (13952139) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
E (13952149) chip[DL]: Failed to post event to CHIP Platform event queue
E (13952159) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
E (13952179) chip[DL]: Failed to post event to CHIP Platform event queue
E (13952179) chip[DL]: Failed to schedule work: 1000000
E (13952179) chip[SC]: Failed to Schedule the AfterWorkCallback on foreground thread: 1000000
E (13952199) chip[DL]: Long dispatch time: 929 ms, for event type 2
~~~

dhrishi, 2024-07-26:
Recently, after quite some testing around different use cases, we increased the size of the event queue to 40. Post that, we do not encounter the "Failed to post" errors. Can you please try with the code on `release/v1.3` branch.

jonsmirl, 2024-07-26:
I have been running fine with 40 for months now.

closed 2024-07-26 by jonsmirl