espressif/esp-matter #657: E (39013585) chip[CSL]: PacketBuffer: pool EMPTY. B

jonsmirl, 2023-09-24:
I left the off-the-shelf esp-matter light demo running over night on ESP32 with 4MB of PSRAM.  And it ran out of packet buffers.
This is unchanged light demo code with nothing but a wildcard subscription monitoring it for 12 hours and it failed.

Why aren't the packet buffers being allocated out of PSRAM? I have 3.9MB of free PSRAM.
The IRAM is getting fragmented to the point that the packet buffers can't be allocated.

~~~
I (39010845) chip[IN]: (S) Sending msg 266512572 on secure session with LSID: 11432
E (39013585) chip[CSL]: PacketBuffer: pool EMPTY. B
Heap summary for capabilities 0x00000800:
  At 0x3ffb2730 len 15448 free 4 allocated 14860 min_free 4
    largest_free_block 0 alloc_blocks 42 free_blocks 0 total_blocks 42
E (39013595) chip[DL]:   At 0x3ffaff10 len 240 free 4 allocated 28 min_free 4
    largest_free_block 0 alloc_blocks 2 free_blocks 0 total_blocks 2
Failed to post event to CHIP Platform event queue  At 0x3ffb6388 len 7288 free 4 allocated 6668 min_free 4
    largest_free_block 0 alloc_blocks 59 free_blocks 0 total_blocks 59

  At 0x3ffb9a20 len 16648 free 64 allocated 15312 min_free 4
    largest_free_block 20 alloc_blocks 134 free_blocks 3 total_blocks 137
E (39013635) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
  At 0x3ffd5710 len 43248 free 72 allocated 40924 min_free 4
    largest_free_block 24 alloc_blocks 362 free_blocks 3 total_blocks 365
  At 0x3ffe0440 len 15072 free 4 allocated 14552 min_free 4
    largest_free_block 0 alloc_blocks 25 free_blocks 0 total_blocks 25
  At 0x3ffe4350 len 113840 free 13412 allocated 98880 min_free 1524
    largest_free_block 7936 alloc_blocks 169 free_blocks 10 total_blocks 179
  At 0x4009ee84 len 4476 free 4072 allocated 20 min_free 4040
    largest_free_block 3840 alloc_blocks 1 free_blocks 1 total_blocks 2
  Totals:
    free 17636 allocated 191244 min_free 5588 largest_free_block 7936

~~~

dhrishi, 2023-09-30:
@jonsmirl Which branch of esp-matter does this use? `main` or `release/v1.1`. We have had long duration runs with `release/v1.1` devices (light example) commissioned into 2 ecosystems (with their subscriptions) running fine for more than 3 weeks.

jonsmirl, 2023-09-30:
I was using the main branch at commit cd325c6fa4203c45707ecb8b5351b7303347acb7

This is not a problem with leaving the device booted for three weeks. It is specifically a problem with leaving a wildcard subscription monitoring the device. The wildcard subscription makes the memory allocations constantly churn. After about 12 hours I can't allocate packet buffers anymore due to fragmentation.

For me, a solution would be a malloc which forces everything into PSRAM unless it is marked as having to go into IRAM. I have megabytes of unused PSRAM just sitting there. I have ALL of the alloc strategies already set to PSRAM.

For a more general solution -- I think you need to work on memory management until a long running subscription does not cause any allocs. For example if I leave a subscription running for a week there should be zero allocations done once the subscription is established.

jonsmirl, 2023-09-30:
I can switch over to v1.1 and try again.

shubhamdp, 2023-10-03:
Hi @jonsmirl Can you try with below patch, it uses the ESP32's heap area for allocating the system buffer pool.
```
curl https://github.com/project-chip/connectedhomeip/commit/8bd558c85bebb56ecb9d80c1d5993dab3bc25c20.patch | git am
```

jonsmirl, 2023-10-03:
You need quotes around the menu string.
bool "Use heap memory for system buffer pool"
~~~
[0/1] cd /home/jonsmirl/lowpan/build && /home/jonsmirl/.espressif/...-env IDF_ENV_FPGA= --output config /home/jonsmirl/lowpan/sdkconfig
/home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip/Kconfig:138: warning: style: quotes recommended around 'Use' in 'bool Use heap memory for system buffer pool'
Traceback (most recent call last):

~~~

jonsmirl, 2023-10-03:
I have this installed on my test nodes now. It is harder to test since we are looking for a negative result. I will run it for several days and see if we hit the old problems.

jonsmirl, 2023-10-03:
Moving those packet buffers to PSRAM has huge impact on free IRAM. I went from 15KB free and fragmented. To 78K free now.
~~~
Heap summary for capabilities 0x00000800:
  At 0x3fcb3134 len 32767 free 31943 allocated 84 min_free 31915
    largest_free_block 31744 alloc_blocks 1 free_blocks 1 total_blocks 2
  At 0x3fcae548 len 242120 free 144 allocated 239216 min_free 4
    largest_free_block 72 alloc_blocks 455 free_blocks 3 total_blocks 458
  At 0x3fce9710 len 22308 free 6332 allocated 14768 min_free 1036
    largest_free_block 4352 alloc_blocks 118 free_blocks 7 total_blocks 125
  At 0x3fcf0000 len 32768 free 32032 allocated 0 min_free 11584
    largest_free_block 31744 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x600fe014 len 8148 free 7768 allocated 0 min_free 7768
    largest_free_block 7680 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 78219 allocated 254068 min_free 52307 largest_free_block 31744
~~~

jonsmirl, 2023-10-05:
I'm at two days now with this turned on for PSRAM and I haven't had a Packet Buffer alloc failure. I am going to stop the test and change the code to periodically dump out PSRAM alloc statistics to ensure it is not leaking PSRAM.  Since i have megabytes of free PSRAM it could be months before I run out due to a leak.

However, I still hit this one all of the time: https://github.com/espressif/esp-matter/issues/656
You need to make that queue longer and update the example sdkconfig.

Also, if you bump the CHIP version up, there is a fix for this one:
~~~
E (84098219) chip[EM]: OnMessageReceived failed, err = 70

~~~

As for the C3 --- The code should block and wait for a buffer to free instead of failing when there is insufficient RAM. Failing just cascades the problem by triggering even more packets due to retries. Blocking will throttle packet generation.

shubhamdp, 2023-10-06:
@jonsmirl Thanks for the help in testing it. I have been running this patch on C3 with one wildcard subscription and did not see this problem.

Thanks for the heads up will bump the CHIP version.

jonsmirl, 2023-10-06:
These are my old bugs for same problem. those were on C3. Currently my test lights are ESP32.
https://github.com/espressif/esp-matter/issues/434
https://github.com/espressif/esp-matter/issues/555

This is a memory dump which was inserted into CHIP code right after the error message printed.
https://github.com/espressif/esp-matter/issues/555#issuecomment-1704435684

~~~
diff --git a/src/system/SystemPacketBuffer.cpp b/src/system/SystemPacketBuffer.cpp
index 209a80b821..589bcc0924 100644
--- a/src/system/SystemPacketBuffer.cpp
+++ b/src/system/SystemPacketBuffer.cpp
@@ -150,7 +150,10 @@ void PacketBufferHandle::InternalRightSize()
     PacketBuffer * newBuffer = reinterpret_cast<PacketBuffer *>(chip::Platform::MemoryAlloc(blockSize));
     if (newBuffer == nullptr)
     {
-        ChipLogError(chipSystemLayer, "PacketBuffer: pool EMPTY.");
+        ChipLogError(chipSystemLayer, "PacketBuffer: pool EMPTY. A");
+#ifdef ESP_PLATFORM        
+        heap_caps_print_heap_info(MALLOC_CAP_INTERNAL);
+#endif        
         return;
     }
 
@@ -536,7 +539,10 @@ PacketBufferHandle PacketBufferHandle::New(size_t aAvailableSize, uint16_t aRese
 
     if (lPacket == nullptr)
     {
-        ChipLogError(chipSystemLayer, "PacketBuffer: pool EMPTY.");
+        ChipLogError(chipSystemLayer, "PacketBuffer: pool EMPTY. B");
+#ifdef ESP_PLATFORM        
+        heap_caps_print_heap_info(MALLOC_CAP_INTERNAL);
+#endif        
         return PacketBufferHandle();
     }
 ~~~

jonsmirl, 2023-10-13:
Something is definitely broken. Look at this trace from the light demo, the unit is just sitting idle. Some transient thing is eating up huge amounts of memory. You can see from trace that there is no Matter traffic.

At 9585776  -- 61,196 bytes free
At 9605776 -- 47,180 bytes free
then it comes back to 61,196 free

Then it hits PacketBuffer: pool EMPTY!
only 11,108 free and there is not a large enough block to make a packet buffer.
and that chains into a MDNS failure.

Then at 9705956 -- back to 61,196 bytes free.

We are lucky in this trace that there were two heap dump quite close to each other.
  At 0x3ffb9a20 len 16648 free 700 allocated 14740 min_free 4
    largest_free_block 672 alloc_blocks 118 free_blocks 2 total_blocks 120
  At 0x3ffe4350 len 113840 free 56316 allocated 56156 min_free 216
    largest_free_block 53248 alloc_blocks 124 free_blocks 6 total_blocks 130
changed to:
  At 0x3ffb9a20 len 16648 free 68 allocated 15304 min_free 4
    largest_free_block 24 alloc_blocks 135 free_blocks 3 total_blocks 138
  At 0x3ffe4350 len 113840 free 7504 allocated 104712 min_free 216
    largest_free_block 1472 alloc_blocks 188 free_blocks 16 total_blocks 204

So what suddenly ate 49,160 bytes of memory? It has to be something to do with network buffers. 
This node didn't have PSRAM enabled, but it did have WIFI_DYNAMIC buffers turned on with a count of 32.
It is using the CHIP MDNS code.
[sdkconfig.txt](https://github.com/espressif/esp-matter/files/12897536/sdkconfig.txt)

My guess? Something to do with MDNS. My home network has about 100 devices on it and dozens of them are using MDNS. Maybe one of them is doing something to trigger a bad response out of the chip MDNS implementation.
[trace.txt](https://github.com/espressif/esp-matter/files/12897558/trace.txt)

~~~
I (9585776) app_main: Free internal 61196 biggest block 53248 total 57148
I (9602506) ROUTE_HOOK: Received RIO
I (9602506) ROUTE_HOOK: prefix FD85:96F1:7244:1:: lifetime 1800
I (9602706) ROUTE_HOOK: Received RIO
I (9602706) ROUTE_HOOK: prefix FD85:96F1:7244:1:: lifetime 1800
I (9602706) ROUTE_HOOK: Received RIO
I (9602716) ROUTE_HOOK: prefix FD85:96F1:7244:1:: lifetime 1800
I (9602726) ROUTE_HOOK: Received RIO
I (9602726) ROUTE_HOOK: prefix FD85:96F1:7244:1:: lifetime 1800
I (9602916) ROUTE_HOOK: Received RIO
I (9602916) ROUTE_HOOK: prefix FD85:96F1:7244:1:: lifetime 1800
I (9602916) ROUTE_HOOK: Received RIO
I (9602926) ROUTE_HOOK: prefix FD9D:2EFF:AAAA:1:: lifetime 1800
I (9605566) ROUTE_HOOK: Received RIO
I (9605576) ROUTE_HOOK: prefix FD85:96F1:7244:1:: lifetime 1800
I (9605776) ROUTE_HOOK: Received RIO
I (9605776) ROUTE_HOOK: prefix FD85:96F1:7244:1:: lifetime 1800
I (9605776) ROUTE_HOOK: Received RIO
I (9605786) ROUTE_HOOK: prefix FD85:96F1:7244:1:: lifetime 1800
I (9605796) ROUTE_HOOK: Received RIO
I (9605796) ROUTE_HOOK: prefix FD85:96F1:7244:1:: lifetime 1800
I (9605776) app_main: Free internal 47180 biggest block 34816 total 43132
Heap summary for capabilities 0x00000800:
  At 0x3ffb2730 len 15448 free 4 allocated 14864 min_free 4
    largest_free_block 0 alloc_blocks 41 free_blocks 0 total_blocks 41
  At 0x3ffaff10 len 240 free 4 allocated 28 min_free 4
    largest_free_block 0 alloc_blocks 2 free_blocks 0 total_blocks 2
  At 0x3ffb6388 len 7288 free 4 allocated 6660 min_free 4
    largest_free_block 0 alloc_blocks 61 free_blocks 0 total_blocks 61
  At 0x3ffb9a20 len 16648 free 700 allocated 14740 min_free 4
    largest_free_block 672 alloc_blocks 118 free_blocks 2 total_blocks 120
  At 0x3ffd5710 len 43248 free 88 allocated 40976 min_free 4
    largest_free_block 40 alloc_blocks 345 free_blocks 3 total_blocks 348
  At 0x3ffe0440 len 15072 free 32 allocated 14572 min_free 4
    largest_free_block 24 alloc_blocks 13 free_blocks 1 total_blocks 14
  At 0x3ffe4350 len 113840 free 56316 allocated 56156 min_free 216
    largest_free_block 53248 alloc_blocks 124 free_blocks 6 total_blocks 130
  At 0x4009ee9c len 4452 free 4048 allocated 20 min_free 4016
[... 35 lines cut ...]
I (9698766) chip[DIS]: mDNS broadcast full failed in 1 separate send attempts.
E (9698776) chip[DIS]: Failed to reply to query: b
I (9705956) app_main: Free internal 61196 biggest block 53248 total 57148
I (9725956) app_main: Free internal 61196 biggest block 53248 total 57148
I (9745956) app_main: Free internal 61196 biggest block 53248 total 57148
I (9748826) ROUTE_HOOK: Received RIO
I (9748836) ROUTE_HOOK: prefix FD85:96F1:7244:1:: lifetime 1800
I (9749036) ROUTE_HOOK: Received RIO
I (9749036) ROUTE_HOOK: prefix FD85:96F1:7244:1:: lifetime 1800
I (9749036) ROUTE_HOOK: Received RIO
I (9749046) ROUTE_HOOK: prefix FD85:96F1:7244:1:: lifetime 1800
I (9749236) ROUTE_HOOK: Received RIO
I (9749236) ROUTE_HOOK: prefix FD85:96F1:7244:1:: lifetime 1800
I (9749236) ROUTE_HOOK: Received RIO
I (9749246) ROUTE_HOOK: prefix FD85:96F1:7244:1:: lifetime 1800
I (9749646) ROUTE_HOOK: Received RIO
I (9749646) ROUTE_HOOK: prefix FD9D:2EFF:AAAA:1:: lifetime 1800
I (9765956) app_main: Free internal 61196 biggest block 53248 total 57148
I (9785956) app_main: Free internal 61196 biggest block 53248 total 57148
I (9800236) ROUTE_HOOK: Ignore invalid ICMP packet
I (9800446) ROUTE_HOOK: Received RIO
I (9800446) ROUTE_HOOK: prefix FD85:96F1:7244:1:: lifetime 1800
I (9800446) ROUTE_HOOK: Received RIO

~~~

jonsmirl, 2023-10-14:
Possible explanation for the above trace....

There is MDNS problem. The CHIP MDNS stack quickly retries before the previous operation times out. Those retries chew up all 32 of the dynamic packet buffers causing the system to run out of memory.  Later those operations timeout and the packet buffers are freed.

jonsmirl, 2024-05-28:
This was caused by a memory leak in older versions of esp-matter. Those leaks appear to be fixed now.

closed 2024-05-28 by jonsmirl