espressif/esp-matter #658: Retrieving QR Payload from Device

anuarrozman, 2023-09-25:
**Describe the question/query that you have**
Hi, is there a way for me to read or retrieve the QR Payload from the device ? If the device is flash with default setup payloads then it is fine, but if Ive flash the device with newly generated payloads using mfg_tool.py, is it possible to read or retrieve the QR Payload ? 

**Additional context**
.- I know that the correct onboarding codes and payloads are in the csv file created during generating the new payload using mfg_tool.py. Looking for alternative way to retrieve the QR payload from the device if it is stored in the device.

anuarrozman, 2023-09-25:
Hi, thank you. I've goes through chapter 5 of Matter specs. In `5.1.1 Onboarding Payload Contents` states that `Some or all of this content will be encoded into different formats, some human-readable (such as numeric string) and machine-readable (such as QR code and NFC) formats for printing or display on or integration into the device`. 

`Integration into the device` means that we can retrieve it?  I've found that the setup pin code and discriminator are hardcoded in [CHIPDeviceConfig.h](https://github.com/project-chip/connectedhomeip/blob/master/src/include/platform/CHIPDeviceConfig.h) with default value example, setup pin code (20202021) and discriminator (0xF00).

anuarrozman, 2023-09-26:
Hi @shubhamdp. When you said this "Then, you can use the APIs from [Onboarding Codes Utils](https://github.com/project-chip/connectedhomeip/blob/master/src/app/server/OnboardingCodesUtil.h) which uses the commissionable data provider and device instance info provider to generate the qrcode." 

Is there a way to generate qrcode without using the mfg tool (Manual setup payload generator) ?

shubhamdp, 2023-09-26:
> Retrieving QR Payload from Device

Does this mean you want a qrcode (or some sort of onboarding payload) in the firmware itself?

If Yes:
`CHIPDeviceConfig.h` has the test values and those gets used in [test/provisional implementation](https://github.com/project-chip/connectedhomeip/blob/master/src/include/platform/internal/GenericConfigurationManagerImpl.ipp#L68) of `CommissionableDataProvider`.

ESP32 platform has an [implementation](https://github.com/project-chip/connectedhomeip/blob/master/src/platform/ESP32/ESP32FactoryDataProvider.h#L47) which reads values from the NVS partition.

What you need to do is follow the [guide](https://github.com/espressif/esp-matter/tree/main/tools/mfg_tool#readme) and generate the factory partition. Before that you may need some sort of provision for pincode. mfg_tool do not put that value into nvs partition, and you can do that by some minor code changes. And implement the [GetSetupPasscode()](https://github.com/project-chip/connectedhomeip/blob/master/src/platform/ESP32/ESP32FactoryDataProvider.h#L53).

Then use the APIs from [Onboarding Codes Utils](https://github.com/project-chip/connectedhomeip/blob/master/src/app/server/OnboardingCodesUtil.h) to get the qrcode on the device.

If No and you just want the another qrcode generator tool on host then there's [setup payload generator in python](https://github.com/project-chip/connectedhomeip/tree/master/src/setup_payload/python)

anuarrozman, 2023-09-27:
Thank you @shubhamdp !!  I managed to retrieve the setup discriminator and some other information like vendor name. But, for `GetSetupPasscode()` you share, there's no details in the .cpp file. Sorry if I'm asking a dumb question but, what should i do to get the setupPasscode value ?

shubhamdp, 2023-09-30:
Can you check the `ESP32FactoryDataProvider.h`, those are there and returns `CHIP_ERROR_NOT_IMPLEMENTED`.

https://github.com/project-chip/connectedhomeip/blob/master/src/platform/ESP32/ESP32FactoryDataProvider.h#L53

you can implement it and return the desired value.

dhrishi, 2023-09-30:
Closing as this is answered

closed 2023-09-30 by dhrishi

anuarrozman, 2023-10-01:
Thanks @dhrishi , sorry i was away for 4 days. Thank you for your help.

anuarrozman, 2023-10-03:
Hi @shubhamdp , sorry but Ive tested using the GetSetupPasscode from the `ESP32FactoryDataProvider.h`. It returns empty. From what i can see, in the `ESP32FactoryDataProvider.cpp` there's no function for get setupPasscode.