espressif/esp-matter #659: Esp Matter supports ESP Mesh Wifi?

ahmed8812, 2023-09-25:
Hi,

Does the esp-matter supports of adding esp wifi-mesh?

Thanks

shubhamdp, 2023-09-25:
@ahmed8812 It can be supported using bridge device type. it is not supported out of the box and you will have implement a wifi-mesh bridge similar to the ones implemented in [esp-matter/examples](../tree/main/examples/).  (1. [blemesh_bridge](https://github.com/espressif/esp-matter/tree/main/examples/blemesh_bridge) 2. [esp-now_bridge_light](https://github.com/espressif/esp-matter/tree/main/examples/esp-now_bridge_light) 3. [zigbee_bridge](https://github.com/espressif/esp-matter/tree/main/examples/zigbee_bridge))

ahmed8812, 2023-09-25:
Thanks, I close it for now

closed 2023-09-25 by ahmed8812