espressif/esp-matter #660: BasicInfo/NodeLabel corrupted on reboot via subscription

jonsmirl, 2023-09-25:
Using unaltered light demo....

1) Use my phone app to write BasicInfo/NodeLabel. The phone app gets the change properly reported via the subscription.
2) Restart phone app, BasicInfo/NodeLabel is still correctly reported via subscription
3) Reboot device.
a) The log shows proper retrieval of the NodeLabel
~~~
I (147625) esp_matter_attribute: ********** R : Endpoint 0x0000's Cluster 0x00000028's Attribute 0x00000005 is Office/Nine **********
~~~
b) But that string is not returned by the subscription. Instead it getting NULL plus some binary data.

Does the subscription code not understand the new compact attribute storage format in flash?
Or... it is bug messing up my subscriptions? https://github.com/espressif/esp-matter/issues/646

shubhamdp, 2023-09-29:
https://github.com/espressif/esp-matter/blob/1cacb8b2fcd42c465e38317d3a63d0169460b397/components/esp_matter/private/esp_matter_nvs.cpp#L56-L59

Here's the culprit, we are corrupting the `val.val.a.t` by updating `val.val.a.s` before that. Ideally, `val.val.a.s` should be updated before `val.val.a.t`.

jonsmirl, 2023-10-03:
Is fix coming soon?

dhrishi, 2023-10-03:
@jonsmirl We have the fix available internally. Can you please confirm if this works for you?

jonsmirl, 2023-10-03:
Where is the fix?

shubhamdp, 2023-10-03:
Please try the below patch
```
curl https://github.com/shubhamdp/esp-matter/commit/28b37dea29476f1d6c27a02f02f130aaa31ecc1c.patch | git am
```

closed 2023-10-04 by espressif-bot